espressif/esp-zigbee-sdk #212: ESP_ZB_ZCL_CMD_ON_OFF_ON_WITH_TIMED_OFF_ID command not working

Suxsem, 2024-01-10:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### esp-zigbee-lib version.

0.9.5

### esp-zboss-lib version.

0.7.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I'm trying to develop a smart switch with an auto timer off feature.
My goal is to use the `On With Timed Off Command` in order to achieve this.

[image]

I have registered the onOff cluster:

```
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

```

and when I issue a "On command", the `zb_action_handler` callback is called with callback_id = ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID, but when I issue the "On With Timed Off Command" the callback is not called at all.

How am I supposed to implement the command? Thank you

### What is the actual behavior?

zb_action_handler not called when `On With Timed Off Command` is received.

### Steps to reproduce.

1. Setup a basic sketch with onOff cluster support.
2. Send a `On With Timed Off Command` to the device.
3. No callback is called

g-chevalier, 2024-01-10:
Hi @Suxsem,

I stumble upon the same problem as you and found a workaround.
However, I think this should be definitely better supported by the stack.

I give you my code just to help you meanwhile a better support is provided.
I went through the priviledged command path to get this particular command.

The code to setup stuff in esp_zb_task is:
```
/* on-off cluster create with standard cluster config*/
// -----------------------------------------------------
esp_zb_on_off_cluster_cfg_t on_off_cfg;
on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);

// Add required attribute in On Off cluster to support "On With Timed Off" commands
esp_zb_on_off_cluster_add_attr(esp_zb_on_off_cluster, ESP_ZB_ZCL_ATTR_ON_OFF_ON_TIME, &onoff_on_time);
esp_zb_on_off_cluster_add_attr(esp_zb_on_off_cluster, ESP_ZB_ZCL_ATTR_ON_OFF_OFF_WAIT_TIME, &onoff_of_wait_time);
// "On With Timed Off" commands are managed at low level in the SDK, and nothing appears at app level,
// so going through privilege_command to trap the "On With Timed Off" command (0x42)
esp_zb_zcl_add_privilege_command(HA_ESP_MAIN_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_ON_OFF, 0x42);
```

The code for zb_privileged_cmd_handler is bellow. It just print the received data as a POC, you'll have to create the appropriate struct to get data.
```
static esp_err_t zb_privileged_cmd_handler(const esp_zb_zcl_privilege_command_message_t *message) {
    // Check if cluster and command correspond to "On With Timed Off" command
    if (message->info.cluster == ESP_ZB_ZCL_CLUSTER_ID_ON_OFF && message->info.command.id == 0x42) {
        ESP_LOGI(TAG, "On With Timed Off");
        uint8_t *pt = message->data;
        // display 02 09 00 04 00 for {"ctrlbits":2,"offwaittime":4,"ontime":9} via send Z2M cmd
        //    &    00 32 00 00 00 for {"state": "on", "on_time":5, "off_wait_time":0} via Z2M MQTT (zigbee2mqtt/0x404ccafffe4091b8/set)
        //   0x32 == 5 X 10
        for (int i = 0; i < message->size; i++) {
            printf("%2.2x ", *pt++);
        }
        printf("\n");
    } else {
        ESP_LOGE(TAG, "bad command in zb_privileged_cmd_handler");
    }
    return ESP_OK;
}
```

For information, I have a pending issue on how to send such commands from the SDK, which seems currently unsupported: https://github.com/espressif/esp-zigbee-sdk/issues/209

Suxsem, 2024-01-10:
thank you @g-chevalier,
while waiting for a reply i implemented basically the same exact solution. Still it would be great if a list of supported commands would be provided.

xieqinan, 2024-01-23:
@Suxsem @g-chevalier ,

The esp-zigbee-sdk will support the on_with_time_off command in the subsequenct version, thanks for your patience.

closed 2024-02-23 by espressif-bot

xieqinan, 2024-02-23:
Hello,

The issue has been fixed in [esp-zigbee-sdk v1.1.2](https://github.com/espressif/esp-zigbee-sdk).