espressif/esp-zigbee-sdk #213: Crash after receiving attribute read response

pieterjanbuntinx, 2024-01-10:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.0

### esp-zigbee-lib version.

1.0.7

### esp-zboss-lib version.

1.0.7

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

The ESP32 receives the read attr resp and thereafter continues running. 

### What is the actual behavior?

The NRF dev board I am using has been factory reset and is open to join new networks. The device successfully joins the network and after the read attr req gets sent, a few moments later I receive the read attr resp. I can read out all the attributes in this resp but with whatever I do (also just not handling the zb_action_handler response), the ESP32 just crashes after.

### Steps to reproduce.

The code below is based on the esp_zigbee_gateway example. I am running the ZB stack natively on the C6 without RCP.

``` c
/* -------------- endpoint & cluster creation -------------*/

static void esp_zb_task(void* pvParameters) {

    // (...)

    /* create endpoint list */
    esp_zb_ep_list_t* esp_zb_ep_list = esp_zb_ep_list_create();

    /* create main endpoint */
    esp_zb_cluster_list_t* main_endpoint = esp_zb_zcl_cluster_list_create();

    /* create basic cluster */
    esp_zb_basic_cluster_cfg_t basic_cfg = {
        .zcl_version  = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE,
        .power_source = ZB_ZCL_BASIC_POWER_SOURCE_DC_SOURCE,
    };
    esp_zb_attribute_list_t* basic_cluster = esp_zb_basic_cluster_create(&basic_cfg);
    esp_zb_cluster_list_add_basic_cluster(main_endpoint, basic_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

    esp_zb_ep_list_add_ep(esp_zb_ep_list, main_endpoint, MAIN_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_TEST_DEVICE_ID);
    // (...)
}

/* ----- handling device join, active_ep_req, simple_desc_req & read_attr_req -------*/

void esp_zb_app_signal_handler(esp_zb_app_signal_t* signal_struct) {
    // (...)
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE:
        dev_annce_params = (esp_zb_zdo_signal_device_annce_params_t*)esp_zb_app_signal_get_params(p_sg_p);
        ESP_LOGI(TAG, "New device commissioned or rejoined (short: 0x%04hx)", dev_annce_params->device_short_addr);
        device_handleJoined(app, dev_annce_params->device_short_addr, dev_annce_params->ieee_addr);
        break;
    }
}

void device_handleJoined(app_t* app, uint16_t short_addr, esp_zb_ieee_addr_t ieee_addr) {
    // check if device already exists in memory
    esp_zb_zcl_addr_t addr = {
        .addr_type = ESP_ZB_ZCL_ADDR_TYPE_SHORT,
        .u =
            {
                .short_addr = short_addr,
            },
    };
    device_t* device = device_get(app, addr);

    if (device != NULL) {
        return;
    }

    device = device_add(app, ieee_addr, short_addr, DEVICE_TYPE_UNKNOWN);

    device_ctx_t* ctx1 = malloc(sizeof(device_ctx_t));
    ctx1->app          = app;
    memcpy(ctx1->ieee_addr, ieee_addr, sizeof(esp_zb_ieee_addr_t));
    ctx1->short_addr = short_addr;

    // get EP list
    esp_zb_zdo_active_ep_req_param_t active_ep_req = {};
    active_ep_req.addr_of_interest                 = short_addr;
    esp_zb_zdo_active_ep_req(&active_ep_req, active_ep_cb, ctx1);
}

static void active_ep_cb(esp_zb_zdp_status_t zdo_status, uint8_t ep_count, uint8_t* ep_id_list, void* user_ctx) {
    device_ctx_t* ctx = (device_ctx_t*)user_ctx;

    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        TRACE_INFO(DEVICE, "Active endpoint response: status(%d) and endpoint count(%d)\n", zdo_status, ep_count);

        for (int i = 0; i < ep_count; i++) {
            TRACE_INFO(DEVICE, "Endpoint ID List: %d\n", ep_id_list[i]);

            esp_zb_zdo_simple_desc_req_param_t simple_descr_req;
            simple_descr_req.addr_of_interest = ctx->short_addr;
            simple_descr_req.endpoint         = ep_id_list[i];

            device_ctx_t* new_ctx = malloc(sizeof(device_ctx_t));
            memcpy(new_ctx, ctx, sizeof(device_ctx_t));
            esp_zb_zdo_simple_desc_req(&simple_descr_req, simple_desc_req_cb, new_ctx);
        }
    }
    free(ctx);
}

static void simple_desc_req_cb(esp_zb_zdp_status_t zdo_status, esp_zb_af_simple_desc_1_1_t* simple_desc, void* user_ctx) {
    device_ctx_t* ctx = (device_ctx_t*)user_ctx;

    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        TRACE_INFO(DEVICE,
                   "Simple desc response: status(%d), device_id(%d), "
                   "app_version(%d), profile_id(0x%x), endpoint_ID(%d)\n",
                   zdo_status, simple_desc->app_device_id, simple_desc->app_device_version, simple_desc->app_profile_id, simple_desc->endpoint);

        for (int i = 0; i < (simple_desc->app_input_cluster_count + simple_desc->app_output_cluster_count); i++) {
            uint16_t cluster_id = *(simple_desc->app_cluster_list + i);
            TRACE_INFO(DEVICE, "Cluster ID list: 0x%x, index %i\n", cluster_id, i);

            switch (cluster_id) {
            case ESP_ZB_ZCL_CLUSTER_ID_BASIC: {
                // request attributes

                uint16_t attributes[8] = {
                    ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID,       ESP_ZB_ZCL_ATTR_BASIC_APPLICATION_VERSION_ID, ESP_ZB_ZCL_ATTR_BASIC_STACK_VERSION_ID, ESP_ZB_ZCL_ATTR_BASIC_HW_VERSION_ID,
                    ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID,    ESP_ZB_ZCL_ATTR_BASIC_DATE_CODE_ID,     ESP_ZB_ZCL_ATTR_BASIC_SW_BUILD_ID,
                };

                esp_zb_zcl_read_attr_cmd_t read_req = {
                    .address_mode = ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT,
                    .zcl_basic_cmd =
                        {
                            .dst_endpoint = simple_desc->endpoint,
                            .src_endpoint = MAIN_ENDPOINT,
                        },
                    .attr_number = 8,
                    .attr_field  = attributes,
                    .clusterID   = cluster_id,
                };
                memcpy(read_req.zcl_basic_cmd.dst_addr_u.addr_long, ctx->ieee_addr, sizeof(esp_zb_ieee_addr_t));

                esp_zb_zcl_read_attr_cmd_req(&read_req);
            } break;
            case ZB_ZCL_CLUSTER_ID_MODBUS: {

            } break;
            default:
                break;
            }
        }
    }

    free(ctx);
}

/* -------------- zb_action_handler -------------*/

static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void* message) {
    esp_err_t ret = ESP_OK;
    switch (callback_id) {
    case ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID:
        ret = zb_read_attr_resp_handler((esp_zb_zcl_cmd_read_attr_resp_message_t*)message);
        break;
    default:
        ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
        break;
    }
    return ret;
}

static esp_err_t zb_read_attr_resp_handler(const esp_zb_zcl_cmd_read_attr_resp_message_t* message) {
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)", message->info.status);

    printf("%s\n", __FUNCTION__);
    esp_zb_zcl_read_attr_resp_variable_t* next = message->variables;
    while (next != NULL) {
        ESP_LOGI(TAG, "Read attr response: dev(0x%x), ep(0x%x), cluster(0x%x), status(%d), attr(0x%x)", message->info.dst_address, message->info.dst_endpoint, message->info.cluster, next->status, next->attribute.id);
        next = next->next;
    }

    // app_handleReadAttributeResp(app, message);
    return ESP_OK;
}
```

xieqinan, 2024-01-11:
Hello @pieterjanbuntinx ,

It appears to be an unsupported data type that is causing the crash. Could you try using a single attribute to test which attribute reading triggers this issue?

pieterjanbuntinx, 2024-01-11:
Hi @xieqinan 

I have done some testing and the error only seems to occur with the attribute ESP_ZB_ZCL_ATTR_BASIC_SW_BUILD_ID. The strange thing is that I can only generate this error in the simple_desc callback. If I use for example the code below everything works fine. I can't see any differences with how I am doing it in the callback.
This code allows me to request arbitrary attributes from a cluster using MQTT commands.
The end device I am using is an NRF development board with custom firmware. For this specific attribute (and all other string attributes) I am using the Pascal string style as required (so 1 char for length + string without delimiter). If I just write a plain string to the other string attributes like for example ESP_ZB_ZCL_ATTR_BASIC_DATE_CODE_ID I get the exact same error (also an assertion error at zcl_common.c:345). I then also get the error with the code below. So maybe it has something to do with this?

``` c
static void handleReadAttribute(void* ctx, char* topic, char* data, size_t len) {
    app_t* app = (app_t*)ctx;

    esp_zb_ieee_addr_t ieee_addr = {0};
    uint8_t*           addr      = (uint8_t*)ieee_addr;
    uint16_t           cluster_id;
    uint16_t           endpoint_id;

    cJSON* json = NULL;

    char* fullTopic = app_getFullTopic(app, "devices/%02hhx%02hhx%02hhx%02hhx%02hhx%02hhx%02hhx%02hhx/read");
    int   num       = sscanf(topic, fullTopic, &addr[7], &addr[6], &addr[5], &addr[4], &addr[3], &addr[2], &addr[1], &addr[0]);
    free(fullTopic);

    if (num != 8) {
        return;
    }

    json = cJSON_ParseWithLength(data, len);
    if (json == NULL) {
        return;
    }

    cJSON* obj = cJSON_GetObjectItem(json, "cluster_id");
    if (obj == NULL || !cJSON_IsNumber(obj)) {
        goto end;
    }
    cluster_id = cJSON_GetNumberValue(obj);

    obj = cJSON_GetObjectItem(json, "endpoint_id");
    if (obj == NULL || !cJSON_IsNumber(obj)) {
        goto end;
    }
    endpoint_id = cJSON_GetNumberValue(obj);

    obj = cJSON_GetObjectItem(json, "attrs");
    if (obj == NULL || !cJSON_IsArray(obj)) {
        goto end;
    }

    size_t numAttrs = cJSON_GetArraySize(obj);
    if (numAttrs > 0) {
        esp_zb_zcl_read_attr_cmd_t read_req;
        uint16_t*                  attributes = calloc(sizeof(uint16_t), numAttrs);
        memset(attributes, 0, sizeof(uint16_t) * numAttrs);

        cJSON* iterator = NULL;
        int    i        = 0;
        cJSON_ArrayForEach(iterator, obj) {
            if (cJSON_IsNumber(iterator)) {
                attributes[i] = cJSON_GetNumberValue(iterator);
                i++;
            } else {
                TRACE_WARNING(DEVICE, "%s(%d): Invalid value in attribute array!\n", __FUNCTION__, __LINE__);
                free(attributes);
                goto end;
            }
        }

        read_req.address_mode               = ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT;
        read_req.attr_number                = numAttrs;
        read_req.attr_field                 = attributes;
        read_req.clusterID                  = cluster_id;
        read_req.zcl_basic_cmd.dst_endpoint = endpoint_id;

        switch (cluster_id) {
        case ESP_ZB_ZCL_CLUSTER_ID_BASIC:
        case ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT:
        case ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT:
        case ESP_ZB_ZCL_CLUSTER_ID_PRESSURE_MEASUREMENT:
        case ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT:
        case ESP_ZB_ZCL_CLUSTER_ID_FAN_CONTROL:
            read_req.zcl_basic_cmd.src_endpoint = MAIN_ENDPOINT;
            break;
        default:
            goto end;
        }

        read_req.zcl_basic_cmd.src_endpoint = MAIN_ENDPOINT;
        memcpy(read_req.zcl_basic_cmd.dst_addr_u.addr_long, ieee_addr, sizeof(esp_zb_ieee_addr_t));
        esp_zb_zcl_read_attr_cmd_req(&read_req);
        free(attributes);
    }
end:
    cJSON_Delete(json);
}
```

xieqinan, 2024-01-23:
@pieterjanbuntinx ,

The `ESP_ZB_ZCL_ATTR_BASIC_SW_BUILD_ID` is a char string data type with a data structure of 1 char for length followed by the string without a delimiter. When adding this attribute to the cluster, ensure that you follow this structure. Furthermore, could you please provide the read response frame from the sniffer when the device triggers the error?

pieterjanbuntinx, 2024-01-23:
Hi @xieqinan 

The problem seems to be resolved in the latest version of the sdk (1.0.9). I do get the read attr response twice now. Once with a status of `ESP_ZB_ZCL_STATUS_SUCCESS` and once with `ESP_ZB_ZCL_STATUS_UNSUP_ATTRIB`. 
Just for in the future, what do you mean with sniffer? Is this an usb zigbee dongle with a sniffer firmware flashed onto it or can I get the same with a setting in the SDKCONFIG? Is there perhaps a sniffer firmware I can flash to an ESP32H2/C6 board?

```
I (66676) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 4 from in queue
I (66676) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(0), attr(0x0)
I (66686) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(0), attr(0x1)
I (66696) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(0), attr(0x2)
I (66706) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(0), attr(0x3)
I (66716) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(0), attr(0x4)
I (66726) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(0), attr(0x5)
I (66736) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(0), attr(0x6)
I (66746) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(0), attr(0x4000)
DEBUG   [MQTT] publish successful, msg_id=0
DEBUG   [MQTT] publish successful, msg_id=0
DEBUG   [MQTT] publish successful, msg_id=0
DEBUG   [MQTT] publish successful, msg_id=0
DEBUG   [MQTT] publish successful, msg_id=0
DEBUG   [MQTT] publish successful, msg_id=0
DEBUG   [MQTT] publish successful, msg_id=0
DEBUG   [MQTT] publish successful, msg_id=0
I (66806) ZBOSS: nwk/nwk_main.c:2301   set input_blocked_by and process 3 from in queue
I (66816) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(0), attr(0x0)
I (66826) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(134), attr(0x1)
I (66836) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(134), attr(0x2)
I (66846) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(134), attr(0x3)
I (66856) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(134), attr(0x4)
I (66866) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(134), attr(0x5)
I (66876) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(134), attr(0x6)
I (66886) ESP_ZB_GATEWAY: Read attr response: dev(0x0), ep(0x1), cluster(0x0), status(134), attr(0x4000)
```

xieqinan, 2024-02-06:
@pieterjanbuntinx 

> Just for in the future, what do you mean with sniffer? Is this an usb zigbee dongle with a sniffer firmware flashed onto it or can I get the same with a setting in the SDKCONFIG? Is there perhaps a sniffer firmware I can flash to an ESP32H2/C6 board?

Yes, the sniffer is a Zigbee frame capture tool, which is very helpful for debugging purposes. The ESP32-C6 can be used as a sniffer dongle. You can refer to the [documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark) for more details.

xieqinan, 2024-03-11:
@pieterjanbuntinx ,

I suspect that the issue may be triggered by thread safety concerns. If the project includes other tasks, please utilize `esp_zb_lock_acquire()` and `esp_zb_lock_release()` to mitigate potential thread safety issues.

pieterjanbuntinx, 2024-03-11:
I was only able to create the problem in certain zb callbacks like the simple_desc callback. When I did the command in a MQTT callback, the error did not occur. Since SDK 1.0.9 I have not had these crashes anymore. But good to know about these functions, I do quite a few Zigbee commands on other threads and this could explain some occasional crashes.

closed 2024-03-11 by pieterjanbuntinx