espressif/esp-zigbee-sdk #214: Add standard cluster 0xEF00 that is not a custom cluster

mundevx, 2024-01-12:
### Question

How i can add a standard cluster like 0xEF00, that is below custom cluster range. 
If i use esp_zb_cluster_list_add_custom_cluster, it will throw error "ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!". 
Is theer any other function that i can use for cluster 0xEF00.

mundevx, 2024-01-13:
I just want to add cluster 0xEF00 that is not in your header file and it not comes under custom cluster. 
How i can do it in espressif zigbee sdk. I need help on this topic.

Suxsem, 2024-01-16:
Hi, I just checked out of curiosity the zigbee spec and 0xEF00 is an invalid cluster id, why do you need this specific id?

[image]

mundevx, 2024-01-17:
This is a Tuya cluster 0xEF00, which seems to be a kind of tunnel inside the tuya firmware. The tuya Cluster 0xEF00 is more or less one tunnel for Tuyas MQTT commands from there MCUs to there cloud MQTT servers. Thats why i want to use it.

Suxsem, 2024-01-17:
I see. So tuya is not zigbee compliant, so I'm not sure that esp zigbee sdk supports this scenario. Let's wait for a reply from espressif devs.

Suxsem, 2024-01-17:
Ok it's sketchy but I think it works:

```
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();

    uint8_t myValue = 45;
    esp_zb_zcl_attr_t myClusterAttr = {
        .access = ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,
        .id = 0x0020,
        .type = ESP_ZB_ZCL_ATTR_TYPE_U8,
        .manuf_code = 0,
        .data_p = &myValue
    };
    esp_zb_zcl_attr_t myClusterAttrList[1];
    myClusterAttrList[0] = myClusterAttr;

    esp_zb_zcl_cluster_t myCluster = {
        .cluster_id = 0xef00,
        .attr_count = 1,
        .attr_desc_list = myClusterAttrList,
        .role_mask = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        .manuf_code = 0,
        .cluster_init = NULL
    };
    esp_zb_cluster_list_t myClusterListElement = {
        .cluster = myCluster,
        .next = NULL
    };
    esp_zb_cluster_list_t* linkedListIterator = esp_zb_cluster_list;
    while (linkedListIterator->next != NULL) {
        linkedListIterator = linkedListIterator->next;
    }
    linkedListIterator->next = &myClusterListElement;

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    /* add created endpoint (cluster_list) to endpoint list */
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, BASE_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
```

obviously it must be adapter to your needs (commands instead of attribute, client instead of server, ecc....)

mundevx, 2024-01-17:
Thanks for reply Suxsem, i will try your code and let you know.

mundevx, 2024-01-22:
Hi Suzsem,
I tried this code, it crashes.

xieqinan, 2024-01-23:
@Suxsem ,

Thank you for your answer on this matter.

xieqinan, 2024-01-23:
@mundevx ,

The esp-zigbee-sdk only supports the standard clusters defined in the Zigbee cluster library spec. The cluster id 0xEF00 is invalid and not supported. It may only be used within the TUYA ecosystem.

mundevx, 2024-01-27:
> @mundevx ,
> 
> The esp-zigbee-sdk only supports the standard clusters defined in the Zigbee cluster library spec. The cluster id 0xEF00 is invalid and not supported. It may only be used within the TUYA ecosystem.

Hi xieqinan,

Then how i can use this cluster 0xEF00, is there any trick or solution that we can solve this issue?

chshu, 2024-02-01:
Per Zigbee spec, 0xEF00 is an invalid cluster, so it can't be supported.

closed 2024-02-01 by chshu

straga, 2024-07-15:
How add tuya cluster in 1.4 version ?

alex-isen, 2025-05-02:
Hello,
Is there still no way to add a custom tuya cluster with espressif sdk ?
I need to obtain the value of a tuya sensor on a ESP32-H2
Thanks

chshu, 2025-05-06:
@alex-isen you can add the custom cluster following this doc: https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_custom.html#zigbee-zcl-custom-cluster

xieqinan, 2025-05-07:
> E (25261) ESP_ZIGBEE_ZCL_CUSTOM_CLUSTER: Received TO_CLI custom command, but cannot find the custom client cluster (0xef00) on endpoint (1)

Please add the custom client cluster to endpoint 1 on the receiver side to resolve the issue.

straga, 2025-05-23:
```
// Handler callbacks for Tuya custom cluster
static esp_err_t tuya_check_value_handler(uint16_t attr_id, uint8_t endpoint, uint8_t *value) {
    // Allow all values by default
    return ESP_OK;
}

static void tuya_write_attr_handler(uint8_t endpoint, uint16_t attr_id, uint8_t *new_value, uint16_t manuf_code) {
    // Log write operations
    ESP_LOGI(CUSTOM_TAG, "Tuya cluster attr write: ep=%d, attr=0x%04x", endpoint, attr_id);
}

// Custom clusters initialization
esp_err_t custom_clusters_init(void) {
    esp_err_t err;
    // Use existing cluster list from core
    extern esp_zb_cluster_list_t *cluster_list;  // Declare external variable

    if (cluster_list == NULL) {  
        ESP_LOGE(CUSTOM_TAG, "Failed get cluster list");
        return ESP_ERR_NO_MEM;
    }

    // Create handlers for custom cluster
    esp_zb_zcl_custom_cluster_handlers_t handlers = {
        .cluster_id = CUSTOM_CLUSTER_CLI_ID,
        .cluster_role = ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE,
        .check_value_cb = tuya_check_value_handler,
        .write_attr_cb = tuya_write_attr_handler
    };

    // Register handlers
    err = esp_zb_zcl_custom_cluster_handlers_update(handlers);
    if (err != ESP_OK) {
        ESP_LOGE(CUSTOM_TAG, "Failed to register custom cluster handler: %s", esp_err_to_name(err));
        return err;
    }

    // Create custom cluster for Tuya devices (0xEF00)
    esp_zb_attribute_list_t *custom_cluster = esp_zb_zcl_attr_list_create(CUSTOM_CLUSTER_CLI_ID);
    if (!custom_cluster) {
        ESP_LOGE(CUSTOM_TAG, "Failed to create custom cluster");
        return ESP_ERR_NO_MEM;
    }

    // Add cluster to the list as client role since we're communicating with Tuya devices
    err = esp_zb_cluster_list_add_custom_cluster(cluster_list, custom_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
    if (err != ESP_OK) {
        ESP_LOGE(CUSTOM_TAG, "Failed to add cluster to list: %s", esp_err_to_name(err));
        free(custom_cluster); // Clean up
        return err;
    }

    ESP_LOGI(CUSTOM_TAG, "ZB_CUSTOM: Custom clusters initialized");
    return ESP_OK;
}
```