espressif/esp-zigbee-sdk #215: Sleepy end device doesn't honor KEEP_ALIVE interval, how to debug?

Suxsem, 2024-01-13:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/5.2

### esp-zigbee-lib version.

1.0.8

### esp-zboss-lib version.

1.0.8

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I implemented the light sleep example (modified for my needs) for ESP32-C6.

The KEEP_ALIVE interval is set to 7500ms, and after successful steering I set the long poll interval with

`            zb_zdo_pim_set_long_poll_interval(ED_KEEP_ALIVE);`

Still, sometimes (I think it's not random but I didn't figured out the specific cause) the device keeps waking up every 2 seconds or so, never entering the 7500ms long poll mode.
Rebooting the board (with the reset button on the dev board) solves the issues, for some time.

Only way to prevent this is to call 

`zb_zdo_pim_permit_turbo_poll(false);`

but doing so, some complex operations fails because the device doesn't enter in the fast poll mode for executing them.

My feeling is that someting prevents the device from exiting the fast poll mode and entering the long poll mode, any way to debug this or knowing why the scheduler schedules such a short wake up interval?

Thank you

### What is the actual behavior?

As described

### Steps to reproduce.

1. Step
2. Step
3. Step
...
As described

Suxsem, 2024-01-14:
I was trying to add debug logs to the sdk...

I added 

```
CONFIG_ESP_ZB_TRACE_ENABLE=y
CONFIG_ZB_TRACE_LEVEL=4
CONFIG_ZB_TRACE_MASK=0x1
```

to my sdkconfig and

`esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_VERBOSE, ESP_ZB_TRACE_SUBSYSTEM_COMMON);`

before `esp_zb_init`.

Unfortunately, the ESP32-C6 crashes 

```
I (23) boot: ESP-IDF v5.2-beta1-263-ge49823f10c 2nd stage bootloader
I (24) boot: compile time Jan 13 2024 22:07:20
I (24) boot: chip revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (74) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (81) boot:  3 factory          factory app      00 00 00020000 00177000
I (88) boot:  4 zb_storage       Unknown data     01 81 00197000 00004000
I (96) boot:  5 ota_0            OTA app          00 10 001a0000 00177000
I (103) boot:  6 zb_fct           Unknown data     01 81 00317000 00000400
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00020020 vaddr=420a0020 size=368f0h (223472) map
I (174) esp_image: segment 1: paddr=00056918 vaddr=40800000 size=09700h ( 38656) load
I (184) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=9fd50h (654672) map
I (317) esp_image: segment 3: paddr=000ffd78 vaddr=40809700 size=07b04h ( 31492) load
I (325) esp_image: segment 4: paddr=00107884 vaddr=40811210 size=01d8ch (  7564) load
I (328) esp_image: segment 5: paddr=00109618 vaddr=50000000 size=00058h (    88) load
I (335) boot: Loaded app from partition at offset 0x20000
I (338) boot: Disabling RNG early entropy source...
I (354) cpu_start: Unicore app
W (363) clk: esp_perip_clk_init() has not been implemented yet
I (370) cpu_start: Pro cpu start user code
I (370) cpu_start: cpu freq: 160000000 Hz
I (370) cpu_start: Application information:
I (373) cpu_start: Project name:     light_bulb
I (378) cpu_start: App version:      1
I (383) cpu_start: Compile time:     Jan 13 2024 22:17:09
I (389) cpu_start: ELF file SHA256:  64fe95464...
I (394) cpu_start: ESP-IDF:          v5.2-beta1-263-ge49823f10c
I (401) cpu_start: Min chip rev:     v0.0
I (405) cpu_start: Max chip rev:     v0.99 
I (410) cpu_start: Chip rev:         v0.0
I (415) heap_init: Initializing. RAM available for dynamic allocation:
I (422) heap_init: At 4081BD60 len 000608B0 (386 KiB): RAM
I (428) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (434) heap_init: At 50000058 len 00003F90 (15 KiB): RTCRAM
I (441) spi_flash: detected chip: generic
I (445) spi_flash: flash io: dio
I (449) sleep: Enable automatic switching of GPIO sleep configuration
I (457) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (465) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (473) sleep_sys_periph: Interrupt Matrix sleep retention initialization
I (480) sleep_sys_periph: HP System sleep retention initialization
I (487) sleep_sys_periph: TEE/APM sleep retention initialization
[... 114 lines cut ...]

40822ac0: 0x00000000 0x00000000 0x00000000 0x420d0b8f 0x00000146 0x0000004d 0x30303234 0x34613564
40822ae0: 0x00000179 0x4004e20c 0x00000800 0x33005e5c 0x00000000 0x00000064 0x00000000 0x42098a94
0x42098a94: rxdc_est_min_new at ??:?

40822b00: 0x00000100 0xfffffffa 0xfffffff5 0x00000012 0x00000001 0x40822c1c 0x00000114 0x00000107
40822b20: 0x4081c000 0x00000114 0x00000107 0x42098e06 0x408141fa 0xa5a5a5a5 0xa5a5a5a5 0x40822f2c
0x42098e06: pbus_rx_dco_cal_1step_new at ??:?

40822b40: 0x00000005 0xa5a5a5a5 0xa5a5a5a5 0x40822f44 0x00000020 0x00a5a5a5 0xa5a5a5a5 0x00000002
40822b60: 0x42010104 0x00000001 0x00000000 0x40822c1b 0x00000003 0x00000000 0x00000030 0x00802000
0x42010104: zb_nvram_app1_write_cb at esp_zigbee_core.c.obj:?

40822b80: 0x00000020 0x00000000 0x00000000 0x408011be 0x0000001a 0x42099466 0x40822c10 0x40811a14
0x408011be: regi2c_exit_critical at C:/Users/rgiordano/esp/esp-idf/components/esp_hw_support/regi2c_ctrl.c:58

0x42099466: rfrx_gain_index_new at ??:?

40822ba0: 0x00000364 0x00000001 0x00040000 0x00000200 0x40812d20 0x00000001 0x40822c20 0x40812d38
40822bc0: 0x0000000e 0x40822c30 0x40822c30 0x40822c30 0x00000010 0x00000000 0x0000007b 0x40822da8
```

Without `esp_zb_set_trace_level_mask` (i.e. only with the sdkconfig directives) the board doesn't crash but only very limited log is printed out.

zboss and zigbee sdk 1.0.8, esp-idf release/5.2

Suxsem, 2024-01-16:
Tried that without any luck, often the device wakes up every 2 seconds instead of 7.5.

That said, I have fixed the issue in another way: I don't really need a KEEP_ALIVE interval of 7.5s (half an hour is more then good for a battery powered device); what I really needed was to set the long POLL INTERVAL to 7.5s, in order to respond to commands from other devices (zigbee specs requires that a router holds an outgoing message for 7.68s before be allowed to discard it).

So I simply set ED_KEEP_ALIVE to 30 * 60 * 1000 and used zb_zdo_pim_set_long_poll_interval(7500).

Now the device correctly wake up every 7.5s to poll for incoming messages then goes to sleep.

I advice you to investigate why setting a low KEEP_ALIVE interval (like 7500ms as in my cases) causes the board to wake up every 2 seconds, but as I said I already resolved in another, better for my needs, way.

Still, I'm not able to use the zigbee debug tracing with 1.0.8 and idf 5.2, take a look at my second post.

kelin6, 2024-03-12:
@Suxsem 

* Based on the latest esp-zigbee-sdk version 1.2.0, modify the ED_KEEP_ALIVE in the zigbee light sleep example to 7500, which allows the device to wake up every 7.5 seconds.
[image]

* Please update the esp-zigbee-sdk version to 1.2.0. Additionally, the [Zigbee debug trace documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32h2/developing.html#enable-trace-logging) has been updated. Please refer to the latest documentation for using Zigbee debug trace.

xieqinan, 2024-04-02:
@Suxsem ，

Does the esp-zigbee-sdk v1.2.0 resolve your issue?

Also, do you have any further topics you'd like to discuss? If the issue has been resolved, please consider closing it.

closed 2024-04-02 by Suxsem

kuyeon, 2026-01-08:
I am using version 1.6.5, but changing Keep Alive does not seem to work.
I want the interval to be 5 minutes, but my end device wakes up every 5 seconds.

It seems that internally the poll interval is fixed at 5 seconds, which causes this behavior.

Does anyone know the reason for this?

xieqinan, 2026-01-12:
@kuyeon,

> Does anyone know the reason for this?

Yes, the reason the end device wakes up is that the stack caches fragmented node descriptor information of the destinations to determine whether an APS frame can be sent to a given destination. In the current implementation, this cache is checked 20 times over a 1000-second period. As a result, the device wakes up every 50 seconds.

Since the end device will sends a ZDO `node_desc_req` to obtain the node descriptor of its parent, this mechanism is triggered during that process. This behavior is reasonable.