espressif/esp-zigbee-sdk #217: Zigbee docs miss reference to ESP32-C6

WeeJeWel, 2024-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

n/a

### esp-zigbee-lib version.

n/a

### esp-zboss-lib version.

n/a

### Espressif SoC revision.

n/a

### What is the expected behavior?

The documentation at https://github.com/espressif/esp-zigbee-sdk/blob/566830c/docs/en/introduction.rst misses all references to ESP32-C6. The diagram therefore is not accurate anymore.

### What is the actual behavior?

Document that the ESP32-C6 can act as a Zigbee + Wi-Fi device.

### Steps to reproduce.

n/a

xieqinan, 2024-01-23:
@WeeJeWel ，

Thanks for your reminder, we will add a more detailed introduction to the document.

chshu, 2024-03-08:
The docs page has been enhanced to include esp32-c6.

closed 2024-03-08 by chshu

WeeJeWel, 2024-03-08:
Doesn't seem fixed though. Can you share a link to the updated article or commit?

chshu, 2024-03-08:
@WeeJeWel https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32c6/index.html, hope the link works for you.

WeeJeWel, 2024-03-08:
Nice, thanks. The issue was about it missing on https://github.com/espressif/esp-zigbee-sdk/blob/main/docs/en/introduction.rst.

Anyway, I'll leave it up to you what to do with it :-)