espressif/esp-zigbee-sdk #219: Cannot configure reports correctly

pieterjanbuntinx, 2024-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.0

### esp-zigbee-lib version.

1.0.8

### esp-zboss-lib version.

1.0.8

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

After sending the config_report_cmd_req I should be receiving a config_report_resp with the correct attribute_id.

### What is the actual behavior?

The attribute_id in any config_report_respons I receive is 0xFFFF. Reporting also does not seem to be working correctly after receiving the config_resp. My battery powered end devices do report things, but this is because they have a default reporting setup (I think). My custom end device does not report anything. When using Zigbee2mqtt to configure reporting, reporting works on both my custom device and my battery powered end devices (Sonoff SNZB-02) at the configured intervals or value changes. 

What could be the cause of this?

### Steps to reproduce.

First I am binding the specific clusters in the end device's endpoints to the local endpoints. These local endpoints have the same clusters as the end device's but configured as client. 

``` c
void device_setupReportingForClusters(device_t* device, uint16_t src_ep, uint16_t dst_ep, uint16_t* clusters, size_t num_clusters) {
    bind_ctx_t bind_ctx_temp = {
        .short_addr = device->short_addr,
        .dst_ep     = dst_ep,
        .src_ep     = src_ep,
    };

    for (int i = 0; i < num_clusters; i++) {
        uint16_t cluster = clusters[i];

        bind_ctx_t* bind_ctx = malloc(sizeof(bind_ctx_t));
        memcpy(bind_ctx, &bind_ctx_temp, sizeof(bind_ctx_t));
        bind_ctx->cluster_id = cluster;

        // bind clusters
        esp_zb_zdo_bind_req_param_t bind_req;

        // bind request for
        memcpy(&(bind_req.src_address), device->ieee_addr, sizeof(esp_zb_ieee_addr_t));
        bind_req.src_endp      = src_ep;
        bind_req.cluster_id    = cluster;
        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        esp_zb_get_long_address(bind_req.dst_address_u.addr_long);
        bind_req.dst_endp     = dst_ep;
        bind_req.req_dst_addr = device->short_addr;

        esp_zb_zdo_device_bind_req(&bind_req, bind_cb, bind_ctx);
    }
}
```

In the bind callback the reporting_config request is send.

``` c
static void bind_cb(esp_zb_zdp_status_t zdo_status, void* user_ctx) {
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        bind_ctx_t* ctx = (bind_ctx_t*)user_ctx;

        TRACE_INFO(DEVICE, "Bind response from address(0x%x), with status(%d)\n", ctx->short_addr, zdo_status);

        // configure reporting
        esp_zb_zcl_config_report_cmd_t report_cmd;
        report_cmd.zcl_basic_cmd.dst_addr_u.addr_short = ctx->short_addr;
        report_cmd.zcl_basic_cmd.dst_endpoint          = ctx->dst_ep;
        report_cmd.zcl_basic_cmd.src_endpoint          = ctx->src_ep;

        report_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        report_cmd.clusterID    = ctx->cluster_id;

        esp_zb_zcl_config_report_record_t* records = NULL;
        switch (ctx->cluster_id) {
        case ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT:
        case ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT:
        case ESP_ZB_ZCL_CLUSTER_ID_PRESSURE_MEASUREMENT:
        case ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT:
            records = malloc(sizeof(esp_zb_zcl_config_report_record_t));
            if (records == NULL) {
                TRACE_WARNING(DEVICE, "Failed to alloc mem for report config! src_ep: %i, dst_ep: %i, nwk_id: 0x%04x\n", ctx->src_ep, ctx->dst_ep, ctx->short_addr);
                goto end;
            }
            records[0].min_interval = 1;
            records[0].max_interval = 30;

            records[0].direction     = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
            report_cmd.record_number = 1;

            if (ctx->cluster_id == ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT) {
                records[0].attributeID       = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID;
                records[0].attrType          = ESP_ZB_ZCL_ATTR_TYPE_S16;
                int16_t report_change        = 0x0032; // 0.5 degrees
                records[0].reportable_change = &report_change;
            } else if (ctx->cluster_id == ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT) {
                records[0].attributeID       = ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_VALUE_ID;
                records[0].attrType          = ESP_ZB_ZCL_ATTR_TYPE_U16;
                uint16_t report_change       = 0;
                records[0].reportable_change = &report_change;
            } else if (ctx->cluster_id == ESP_ZB_ZCL_CLUSTER_ID_PRESSURE_MEASUREMENT) {
                records[0].attributeID       = ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_VALUE_ID;
                records[0].attrType          = ESP_ZB_ZCL_ATTR_TYPE_S16;
                int16_t report_change        = 0;
                records[0].reportable_change = &report_change;
            } else if (ctx->cluster_id == ESP_ZB_ZCL_CLUSTER_ID_CARBON_DIOXIDE_MEASUREMENT) {
                records[0].attributeID       = ESP_ZB_ZCL_ATTR_CARBON_DIOXIDE_MEASUREMENT_MEASURED_VALUE_ID;
                records[0].attrType          = ESP_ZB_ZCL_ATTR_TYPE_SINGLE;
                float report_change          = 0;
                records[0].reportable_change = &report_change;
            }
        default:
            break;
        }

        report_cmd.record_field = records;

        if (records != NULL) {
            TRACE_INFO(DEVICE, "Report config send! dev(0x%x) endpoint(0x%x) cluster_id(0x%x)\n", ctx->short_addr, ctx->dst_ep, ctx->cluster_id);
            esp_zb_zcl_config_report_cmd_req(&report_cmd);
            free(records);
        }
    }

end:
    free(user_ctx);
}
```

The zb_action_handler receives the report_config response and prints out what it receives:

``` c
static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void* message) {
    esp_err_t ret = ESP_OK;
    switch (callback_id) {
    case ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID:
        ret = zb_configure_report_resp_handler((esp_zb_zcl_cmd_config_report_resp_message_t*)message);
        break;
    default:
        ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
        break;
    }
    return ret;
}

static esp_err_t zb_configure_report_resp_handler(const esp_zb_zcl_cmd_config_report_resp_message_t* message) {
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    TRACE_INFO(APP, "%s: Configure report response: ep(0x%x), cluster(0x%x), status(%d), attr(0x%x)\n", __FUNCTION__, message->info.dst_endpoint, message->info.cluster, next->status, next->attribute_id);

    app_handleConfigReportResp(app, message);
    return ESP_OK;
}
```

You can find the console log below with the 0xFFFF attribute values:

```
EBUG   [MQTT] publish successful, msg_id=0
INFO    [APP] zb_configure_report_resp_handler: Configure report response: ep(0x1), cluster(0x402), status(0), attr(0xffff)
INFO    [APP] zb_configure_report_resp_handler: Configure report response: ep(0x1), cluster(0x402), status(0), attr(0xffff)
INFO    [APP] zb_configure_report_resp_handler: Configure report response: ep(0x1), cluster(0x402), status(0), attr(0xffff)
INFO    [APP] zb_configure_report_resp_handler: Configure report response: ep(0x1), cluster(0x402), status(0), attr(0xffff)
```

petergjoel, 2024-01-18:
I am also observing this issue. I've also tried to get the "subscribe" in the CLI example to do *some* reporting, but with no luck.

pieterjanbuntinx, 2024-01-22:
> I am also observing this issue. I've also tried to get the "subscribe" in the CLI example to do _some_ reporting, but with no luck.

I am also not able to get it working in the CLI example with the latest zigbee-sdk version

xieqinan, 2024-01-23:
@pieterjanbuntinx 

> The attribute_id in any config_report_respons I receive is 0xFFFF.

There is a comment that addresses the issue: `The attribute id of the configuring report response. Please note that when info.status does not equal ESP_ZB_ZCL_STATUS_SUCCESS, the attribute_id is reported; otherwise, it is an invalid value (0xFFFF).` You can find more details in [zcl_command](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L1423).

xieqinan, 2024-01-23:
I have double-checked the code you provided, and I believe it is correct. The only point that may need verification is whether you could try using `int16_t report_change = 0;` as the parameter for configuring the report.

pieterjanbuntinx, 2024-01-23:
> @pieterjanbuntinx
>
>
> There is a comment that addresses the issue: `The attribute id of the configuring report response. Please note that when info.status does not equal ESP_ZB_ZCL_STATUS_SUCCESS, the [...]

That is a strange way of displaying that something was successful, you would think it would be the other way around. I would have preferred getting the attribute id back on success because that way I can check if the report configuration for that specific attribute has been successful. Now that it only returns 0xFFFF, I have no way of knowing what attribute_id the response corresponds to.

pieterjanbuntinx, 2024-01-23:
Sadly this did not seem to help after some quick testing.

petergjoel, 2024-01-23:
@pieterjanbuntinx do you get sane values in the `direction` field of `esp_zb_zcl_config_report_resp_variable_s`?

Also, I seem to get NLME reports at intervals that fit with my report interval, but no value reporting.

pieterjanbuntinx, 2024-01-23:
@petergjoel For the sonoff sensor I get a few good values for the direction field followed by 1 strange value:
```
[DEVICE] handleDeviceInfo: Device 0x679e already exists in memory, ignoring...
INFO    [APP] zb_configure_report_resp_handler: Configure report response: ep(0x1), cluster(0x402), status(0), attr(0xffff)
INFO    [APP] app_handleConfigReportResp: Configure report response: ep(0x1), cluster(0x402), status(0), attr(0xffff) dir(0)
DEBUG   [MQTT] publish successful, msg_id=19752
INFO    [APP] app_handleConfigReportResp: all clusters have been configured for reporting!
INFO    [APP] app_handleConfigReportResp: Configure report response: ep(0x1), cluster(0x402), status(0), attr(0xffff) dir(0)
DEBUG   [MQTT] publish successful, msg_id=1508
INFO    [APP] app_handleConfigReportResp: all clusters have been configured for reporting!
INFO    [APP] app_handleConfigReportResp: Configure report response: ep(0x1), cluster(0x402), status(0), attr(0xffff) dir(0)
DEBUG   [MQTT] publish successful, msg_id=11537
INFO    [APP] app_handleConfigReportResp: all clusters have been configured for reporting!
INFO    [APP] app_handleConfigReportResp: Configure report response: ep(0x1), cluster(0x402), status(0), attr(0xffff) dir(5)
DEBUG   [MQTT] publish successful, msg_id=32582
INFO    [APP] app_handleConfigReportResp: all clusters have been configured for reporting!
```
I only send the report config request once but the sensor returns 4 responses.

For my custom board I only seem to be getting random values for dir (25, 14, 67...). I don't seem to be getting these NLME reports at the intervals I configured the reporting for.

pieterjanbuntinx, 2024-01-30:
@xieqinan 
Do you perhaps have any updates on this? Does reporting configuration work for you? 
Is there maybe anything else I could try to solve this?

xieqinan, 2024-02-05:
@pieterjanbuntinx  @petergjoel，

I have attached an example based on your code for configuring attribute reporting of the local endpoint. This example allows the local endpoint to send attribute reports to the coordinator. You can refer to it and compare the differences with your existing code.

[config_report.zip](https://github.com/espressif/esp-zigbee-sdk/files/14163665/config_report.zip)

xieqinan, 2024-04-02:
@petergjoel @pieterjanbuntinx ，

The esp-zigbee-sdk has provide the temperturer sensor example to configure the report, you can refer to [temperture example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor) which is helpeful for your request.

By the way, if the issue has been resolved, please close it. Welcome to open other issues for esp-zigbee-sdk.

petergjoel, 2024-04-02:
@xieqinan , I am trying to build a gateway (and I think @pieterjanbuntinx is doing the same). AFAIK, the gateway needs to request/configure the reporting of the edge device. The code you refer to only implements the edge device.

This file https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c seems to do something along the line. If the functionality matured with this example, the feature was implemented three weeks ago?

xieqinan, 2024-04-03:
@petergjoel,

> This file https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c seems to do something along the line. If the functionality matured with this example, the feature was implemented three weeks ago?

Yes, you can try to run the couple of examples.

xieqinan, 2024-06-17:
@pieterjanbuntinx 

Do you have any updates on this issue? If it has been fixed, please consider closing it.

pieterjanbuntinx, 2024-06-17:
I stopped using IAS Zones and just configured reporting for binary input and on off clusters. This works for my application. But this does not resolve this issue.

closed 2025-02-08 by chshu