espressif/esp-zigbee-sdk #220: Unable to get any scene add, store or recall callback from gateway

mundevx, 2024-01-17:
### Question

I am working to creeate a scene switch with device_id = ESP_ZB_HA_SCENE_SELECTOR_DEVICE_ID.  I created four end points and registered the device. After that i bind my scene switch endpoints with gateway device. 
On adding scenes in gateway app, i am not receiving any scene adding or any of callback from gateway. I am using callback : esp_zb_core_action_handler_register(zb_action_handler);

static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message)
{
    esp_err_t ret = ESP_OK;
    switch (callback_id) {
    case ESP_ZB_CORE_SCENES_STORE_SCENE_CB_ID:
        ret = zb_scene_store_handler((esp_zb_zcl_store_scene_message_t*)message);
        break;
	break;
    case ESP_ZB_CORE_SCENES_RECALL_SCENE_CB_ID:
        ret = zb_scene_recall_handler((esp_zb_zcl_recall_scene_message_t*)message);
        break;   
    case ESP_ZB_CORE_CMD_GET_SCENE_MEMBERSHIP_RESP_CB_ID:
        ret = zb_scene_get_membership_handler((esp_zb_zcl_scenes_get_scene_membership_resp_message_t*)message);
        break;
    case ESP_ZB_CORE_CMD_VIEW_SCENE_RESP_CB_ID:
        ret = zb_scene_view_handler((esp_zb_zcl_scenes_view_scene_resp_message_t*)message);
        break;
    case ESP_ZB_CORE_OTA_UPGRADE_VALUE_CB_ID:
        ret = zb_ota_upgrade_status_handler(*(esp_zb_zcl_ota_upgrade_value_message_t *)message);
        break;
    default:
        ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
        break;
    }
    return ret;
}

xieqinan, 2024-01-23:
Hello @mundevx ，

If the device receives the add scenes command, it will only trigger a log to indicate successful addition and will not invoke the callback. Could you use a sniffer to capture the frame from the gateway to ensure that the frame is correctly sent by the gateway?

chshu, 2024-03-08:
@mundevx The scene features are mainly maintained within the SDK, including hanlding the scene commands and scene table storage. So there is no callback for the scene commands.

closed 2024-03-08 by chshu

Haerteleric, 2024-12-18:
can you elaborate how the sdk handles that ? is this only for specific device types ?

Faceing the same problem with X amount Extended Color Lights EPs (0x010D)