espressif/esp-zigbee-sdk #222: Scene Response Issue

mundevx, 2024-01-17:
### Question

I am getting this issue at ESP_ZIGBEE_CORE when i add scene in gateway app. I have 1 light and 1 scene switch at other end. Both are end device.
Endpoint(0x1) not found in the group table for group ID(0x01)
or
Endpoint(0x2) not found in the group table for group ID(0x02)
or
Endpoint(0x3) not found in the group table for group ID(0x03)

Why i am getting this error?

xieqinan, 2024-01-23:
Hello @mundevx ,

I suggest using `esp_zb_zcl_groups_add_group_cmd_req()` to make endpoint (0x01) join the group (0x01).

mundevx, 2024-02-16:
> Hello @mundevx ,
> 
> I suggest using `esp_zb_zcl_groups_add_group_cmd_req()` to make endpoint (0x01) join the group (0x01
Now i added endpoint 0x1 to group 0x1, endpoint 0x2 to group 0x2, endpoint 0x3 to group 0x3, endpoint 0x4 to group 0x4 after connection establishment with tuya gateway as coordinator. But scenes and groups are created by Tuya app, and can't handle any overwrite of same scene id. Now i am getting  Endpoint(0x1) not found in the group table for group ID(0x05). Then i again created new scene on app on same scene id, and it showing Endpoint(0x1) not found in the group table for group ID(0x06) . 
As the app not keeping same group when i edit the scene.

What can i do now?

mundevx, 2024-02-16:
Why this log "Endpoint(0x1) not found in the group table for group ID(0x01)" is taken in ESP_ZIGBEE_CORE, keep it in response handler so we can utilize this error.

xieqinan, 2024-02-29:
@mundevx ,

I am a bit confused about the group behavior in the TuYa APP, as it allows multiple groups to contain the same endpoint, which seems to serve no meaningful purpose. So, we may need to discuss this further.

> Why this log "Endpoint(0x1) not found in the group table for group ID(0x01)" is taken in ESP_ZIGBEE_CORE, keep it in response handler so we can utilize this error.

Since checking the group ID is a necessary operation as defined in the Zigbee specification, even if we provide the response handler to the user, they should still perform this check. The logging provided by the ESP-Zigbee-SDK serves to notify the user of any errors that occur.

mundevx, 2024-02-29:
> @mundevx ,
>
> I am a bit confused about the group behavior in the TuYa APP, as it allows multiple groups to contain the same endpoint, which seems to serve no meaningful purpose. So, we may need to [...]

@xieqinan 
Yes, logging should be provided to the user of any error that occur. So please provide us response handler for this error to the user, so we can take action on it.

xieqinan, 2024-02-29:
@mundevx 

> Yes, logging should be provided to the user of any error that occur. So please provide us response handler for this error to the user, so we can take action on it.

It is not a problem to provide an error response handler, but we need a compliant reason for the handler.

Regarding this issue, if we provide the error response handler, what action do you plan to take? Are you planning to add the group ID to the group table according to the command requirement? If so, I believe it is more important to investigate why the client side of the scene cluster is sending an undefined group ID to the cluster server. The specification only indicates that if a device receives an undefined and non-zero group ID for adding a scene, the server should respond with the evaluated status to the client.

We need more detailed information about the action you took so that we can determine the necessity of the response handler.

mundevx, 2024-02-29:
Yes, i am working with two branded zigbee gateway, and i am getting same issue on both of them when creating scene on their app. for each scene created, it created a new group i.e. suppose, if i have 12 scenes switch build in espressif with 4 buttons and each button have 3 scenes like single click, double click, and long press. then zigbee gateway will create new group for every new scene created in app. And on logs, i am getting these warnings:
Endpoint(0x1) not found in the group table for group ID(0x01)
or
Endpoint(0x2) not found in the group table for group ID(0x02)
or
Endpoint(0x3) not found in the group table for group ID(0x03)

...........................................................................................................................

Endpoint(0x18) not found in the group table for group ID(0x18).

xieqinan, 2024-03-01:
@mundevx ，

> then zigbee gateway will create new group for every new scene created in app.

It appears that the behavior is not in line with the actual application requirements. Could you please check if there is a way to adjust the relationship between the endpoint with the scene cluster and the group?

closed 2024-07-26 by chshu