espressif/esp-zigbee-sdk #226: Basic cluster attributes not written to internal registers

mrsteakhouse, 2024-01-19:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### esp-zigbee-lib version.

1.0.8

### esp-zboss-lib version.

1.0.8

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

The manufacturer name and model identifier should be available with the configured values.
[image]

### What is the actual behavior?

The memory regions containing the basic cluster attributes is not written with the configured values.

### Steps to reproduce.

1. Setup the project as described in [this comment](https://github.com/espressif/esp-idf/issues/10662#issuecomment-1859165643).
1. Make sure to use the library versions 1.0.8 for zboss and the zigbee sdk.
2. Set any differnt manufacturer or model.
3. build, flash, pair.
4. The device still has the previous manufacturer name and model id.
5. Change the zboss and zigbee sdk versions to 1.0.5.
6. rebuild, flash, pair
7. now the configured attributes are available

### More Information.

Apparently writing the given memory regions works with version 1.0.5. When i downgrade the version it works as intended, but switching back to the latest version and changing the given attributes does not change the reported attributes in zigbee2mqtt.

xieqinan, 2024-02-05:
@mrsteakhouse ,

As far as I understand, the `manufacturer name and model identifier` attribute of the basic cluster are variables that can be updated during runtime, but they are not volatile. You can use the following code to verify whether the value has been written or updated:

```c
esp_zb_zcl_attr_t *attr = esp_zb_zcl_get_attribute(HA_ESP_LIGHT_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_BASIC, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID);
if (attr) {
    printf("type(%d), value: %s\n", attr->type, (uint8_t *)attr->data_p);
}
```

If the value is updated, it is possible that the remote device with the GUI has not updated the value accordingly.

mrsteakhouse, 2024-02-06:
Alright, thanks. After many resets the desired values showed up. It might be as well caching of the underlying zigbee coordinator, that i use, but have not tested that.

closed 2024-02-06 by mrsteakhouse