espressif/esp-zigbee-sdk #227: Power reduction of a router

BaWe155, 2024-01-19:
### Question

Hello,

I’m wondering if is possible to reduce power consume of a router or is reduction of Tx power the only option? 

Kind regards
Bas

chshu, 2024-01-25:
The router should be RX always on, so the RX power consumption is the base line for a router device. Reduce Tx power only helps to save some power for transmission.

chshu, 2024-01-25:
The router device can't sleep, otherwise it will cause connectivity issues. Only Sleepy end device could enable sleep mode.

Please stay tuned for our new SoCs release which could do better on power consumption.

BaWe155, 2024-01-25:
Thank you, do you have a report with Zigbee router power consume?

Grtz
Bas

chshu, 2024-01-25:
You may refer to the `4.6 Current Consumption` section from C6 datasheet here: https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf.

The router power baseline is 802.15.4 RX mode, and the peak is the Tx mode.

BaWe155, 2024-01-25:
Thank you for the info, I had seen it. Now I understand it better

closed 2024-02-01 by chshu

BaWe155, 2024-07-11:
Hello,

How to set a TX power to -15dBm?  now I use      esp_zb_set_tx_power(0);

xieqinan, 2024-07-22:
@BaWe155 ,

You  can use the `esp_zb_set_tx_power(-15)` function, it sets the TX power to -15 dBm directly. However, please note that the minimum TX power value is -15 dBm for the ESP32C6 and -24 dBm for the ESP32H2.

BaWe155, 2024-07-22:
@xieqinan 

Thank you.
Is it possible to reduce the system clock speed of the microcontroller? and running the zigbee stack?