espressif/esp-zigbee-sdk #228: Zigbee ZHA support with Arduino - ESP32-C6

ilker-aktuna, 2024-01-24:
### Question

I am trying to make a working zigbee end device using a ESP32-C6 Dev module and Arduino IDE.
I found the Arduino example from here:
https://github.com/P-R-O-C-H-Y/arduino-esp32/blob/Zigbee-examples/libraries/ESP32/examples/Zigbee/Zigbee_Light_Bulb/Zigbee_Light_Bulb.ino

But this did not try to join any network.
So I changed the "esp_zb_task" method with:

```
char modelid[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};

static void esp_zb_task(void *pvParameters)
{
    // initialize Zigbee stack with Zigbee end-device config 
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    // set the on-off light device config
    uint8_t test_attr, test_attr2;
 
    test_attr = 0;
    test_attr2 = 4;
    // basic cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr2);
    esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr2);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    // identify cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &test_attr);
    // group cluster create with fully customized
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
    esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &test_attr);
    // scenes cluster create with standard cluster + customized
    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(NULL);
    esp_zb_cluster_update_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_NAME_SUPPORT_ID, &test_attr);
    // on-off cluster create with standard cluster config
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
    // create cluster lists for this endpoint
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    // update basic cluster in the existed cluster list
    //esp_zb_cluster_list_update_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    // add created endpoint (cluster_list) to endpoint list
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

with this change , the device is trying a join on my hub. (Hubitat)
I can see the device being detected on hub but it is stuck at "initializing" phase.

I'm not sure why , log_i and log_e statements in the example code do not produce on serial monitor.
So I added some serial.println replacing them. These work and I see that the ESP_ZB_BDB_SIGNAL_STEERING is received , but I get these errors i after that:

Network steering was not successful (status: ESP_FAIL
ZDO signal: ZDO Device Unavailable

So what is missing in this code to join a hub ?

Also, why doesn't the log_i and log_e statements work for me ?

Greginkansas, 2024-01-27:
Its the hub the code joins to my setup with zigbee2mqtt just fine.

ilker-aktuna, 2024-01-27:
> zigbee2mqtt

there is no zigbee2mqtt in my setup. And I did not understand the relevance of your answer with my question.

I have a Hubitat hub which is acting as a zigbee coordinator.
I want to make a zigbee end device from ESP32-C6 
so the esp32-c6 should join my hub using zigbee.

Greginkansas, 2024-01-28:
The relevance is that it can connect to another system.  So I would take a look at your zigbee coordinator, also as part of your trouble shooting.

ilker-aktuna, 2024-01-28:
you don't understand.
I have a Hubitat hub which is pretty known device. And on this hub , I have more than 50 zigbee devices connected. They are from different brands. So Hubitat follows zigbee standards. There is no problem on Hubitat side.

Besides , you are not connecting to a zigbee hub as I require. 
Your scenario is very different from mine. And thus, it is not an answer.

I am waiting support from the project owners...

ilker-aktuna, 2024-01-29:
@TheKayneGame 
actually Hubitat does not care about device signature at first pairing. If the device is not known to hub it first assigns the device as a "default device" with no capabilities. Then the user can select an appropriate device along from the internal devices or add a custom device driver to hub.
I got many Chinese devices like this. They were identified as default device and then either me or some other developer wrote a  device driver to support it.
If the device is a simple one (like what I am trying to achieve at first) , then there's even no need for writing a device driver. I can just select a generic zigbee light bulb to control it.

In my experience with Esp32-c6 Arduino examples the device can not complete pairing. On the other hand other people had success with Espressif IDF examples slightly modifying them.
So, here the problem is with Arduino support on Zigbee.

I am still looking for Espressif developers' help. We need proper examples for pairing to a hub.
Even if they were working , there are only two examples (light bulb and ligth switch). There's no example for a RGB light bulb which we could easily test with the RGB led on the esp32-c6 dev kit.

P-R-O-C-H-Y, 2024-01-30:
@ilker-aktuna To answer about the part of the Arduino support of Zigbee. 
If you have any ESP-IDF example you can run it within Arduino just with small changes. We didn't created any wrapper library on top of the Zigbee SDK, we are just providing the libs, so they can be used in Arduino. Those 2 examples are just a slightly edited Zigbee SDK example, to not rely on other esp-idf libraries that we don't have available in Arduino.
So if you have sources to the examples, that worked with ESP-IDF and the Habitat, please share them and I am sure with not much effort they can be used within Arduino.

fillibar, 2024-01-31:
I replaced the esp_zb_task with the below code (from within issue [10662](https://github.com/espressif/esp-idf/issues/10662#issuecomment-1859165643)'s thread). This allowed it to work with the Hubitat when the ESP32-C6 was flashed using the IDF.

```
char modelid[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack with Zigbee end-device config */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    /* set the on-off light device config */
    uint8_t test_attr, test_attr2;
 
    test_attr = 0;
    test_attr2 = 4;
    /* basic cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr2);
    esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr2);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    /* identify cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &test_attr);
    /* group cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
    esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &test_attr);
    /* scenes cluster create with standard cluster + customized */
    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(NULL);
    esp_zb_cluster_update_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_NAME_SUPPORT_ID, &test_attr);
    /* on-off cluster create with standard cluster config*/
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
    /* create cluster lists for this endpoint */
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    /* update basic cluster in the existed cluster list */
    //esp_zb_cluster_list_update_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    /* add created endpoint (cluster_list) to endpoint list */
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

ilker-aktuna, 2024-01-31:
@P-R-O-C-H-Y 
as you see, @fillibar had success with IDF example on the same hub.
Arduino version does not work as I wrote.
I don't know why I am trying to (or I have to) pursuade you that Arduino examples are not working. Somebody has to work on Ardunio version if it will be supported. Otherwise, let's say Zigbee is not supported on Arduino.

Greginkansas, 2024-02-01:
I have 2 zigbee running now 1 on the garage door now running for 4 days with no crashes and 1 with get this a lux senser using a Adafruit driver.   So this one both sends and receives commands. 

/**
 *    Light with a temp cluster sends LUX start
 * 
 * The example demonstrates how to use ESP Zigbee stack to create a end device light bulb.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator.
 * 
 * Proper Zigbee mode must be selected in Tools->Zigbee mode 
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 * 
 * Please check the README.md for instructions and more detailed description.
 */

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "esp_zigbee_core.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"

#include "Wire.h"
#include "Adafruit_VEML7700.h"
Adafruit_VEML7700 veml = Adafruit_VEML7700();
#include "esp_check.h"
#include "esp_err.h"
#include "esp_log.h"
#define SERIAL_BAUD 115200

 float  LUX_READ;  
uint16_t temperature = 10;  // old line gc
float temp;
/////   gc test vars
uint16_t tempsendmin = 0, tempsendmaax = 2000;
bool temp_SW_state = false;
static const char *TAG = "ESP_GC";

#define LED_PIN RGB_BUILTIN

/* Default End Device config */
#define ESP_ZB_ZED_CONFIG()                                     \
    {                                                           \
        .esp_zb_role = ESP_ZB_DEVICE_TYPE_ED,                   \
        .install_code_policy = INSTALLCODE_POLICY_ENABLE,       \
        .nwk_cfg = {                                            \
            .zed_cfg = {                                        \
                .ed_timeout = ED_AGING_TIMEOUT,                 \
                .keep_alive = ED_KEEP_ALIVE,                    \
            },                                                  \
        },                                                      \
    }

#define ESP_ZB_DEFAULT_RADIO_CONFIG()                           \
    {                                                           \
        .radio_mode = RADIO_MODE_NATIVE,                        \
    }
#define ESP_ZB_DEFAULT_HOST_CONFIG()                            \
    {                                                           \
        .host_connection_mode = HOST_CONNECTION_MODE_NONE,      \
    }

/* Zigbee configuration */
#define INSTALLCODE_POLICY_ENABLE       false    /* enable the install code policy for security */
#define ED_AGING_TIMEOUT                ESP_ZB_ED_AGING_TIMEOUT_64MIN
#define ED_KEEP_ALIVE                   3000    /* 3000 millisecond */
#define HA_ESP_LIGHT_ENDPOINT           10   /* 10 esp light bulb device endpoint, used to process light controlling commands */
#define ESP_ZB_PRIMARY_CHANNEL_MASK     ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK  /* Zigbee primary channel mask use in the example */
[... 189 lines cut ...]
    Serial.begin(SERIAL_BAUD);
    // Init Zigbee
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
   Serial.println("Set up done");
    // Init RMT and leave light OFF
    neopixelWrite(LED_PIN,0,0,0);
Wire.begin(22,21);
    //Start Zigbee task
      if (!veml.begin()) {
    Serial.println("Sensor not found");
   // while (1);
  }

   xTaskCreate(update_lux, "update_lux_value", 4096, NULL, 2, NULL);
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

void loop() {
    //empty, zigbee running in task
}

ilker-aktuna, 2024-02-01:
@Greginkansas 
do you say that this example is working with a Zigbee hub ?
If yes, can you put this example as raw file to somewhere ?
Because when you put it inline to your message it gets formatted and can't be copied fully.

mitag, 2024-02-12:
@ilker-aktuna 
> I am trying to make a working zigbee end device using a ESP32-C6 Dev module and Arduino IDE.

I tried @P-R-O-C-H-Y's example last night with success, but I downloaded it from a different link.
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Zigbee/Zigbee_Light_Bulb

I used an ESP32-C6-DevKitC. After a reset, the DevKit was recognised by ZHA. I use a Zigbee 3.0 USB Dongle Plus (ZBDongle-E EFR32MG21) with Multiprotocol firmware as coordinator. The Silicon Labs Multiprotocol Add-on forwards the ZigBee communication to ZHA.

ilker-aktuna, 2024-02-12:
@mitag 
did you use the "Zigbee_Light_Bulb.ino" file from that link as is (without any modification) ?
If so, that file is not even trying to pair with a hub. How did you make it work ?
Maybe your dongle (and firmware) is starting pairing in reverse direction?

mitag, 2024-02-12:
To be honest, I don't remember if my other ZigBee devices automatically triggered pairing, as I set up Home Assisstant several times a year ago, first twice on Raspberrys and finally on an Intel Mini PC. When I set up the Bosch Smart Home Radiator Thermostats II last autumn, manual work was necessary.

That's why I didn't notice yesterday that I had to call up ZHA and initiate pairing from there. I'm sorry that this didn't seem worth mentioning.

ilker-aktuna, 2024-02-13:
so your experience is not relevant to what I'm trying to achieve ?

mitag, 2024-02-13:
I'm sorry, I've probably written too much again. Only the following statements are relevant:
I had to call up ZHA and initiate pairing from there.
I have not changed anything in Zigbee_Light_Bulb.ino.

superzanti, 2024-02-16:
I'm trying to achieve something similar with the ESP32-H2.
I have a zigstar coordinator connected to a zigbee2mqtt instance.

If I load in this [Zigbee_Light_Bulb.ino](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Zigbee/Zigbee_Light_Bulb) does the ESP automatically enter pairing after flashing?

So would all I need to do is set the coordinator to pairing mode and it should connect? What's the device recognized as?

ilker-aktuna, 2024-02-16:
@superzanti 
it did not pair with mu hub. Try and please let us know the result.

mitag, 2024-02-16:
> If I load in this [Zigbee_Light_Bulb.ino](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Zigbee/Zigbee_Light_Bulb) does the ESP automatically enter pairing after flashing?

Zigbee_Light_Bulb.ino is not able to initiate pairing on its own, you have to initiate it from the coordinator. This worked for me, as I described above. The pairing went perfectly, I was then able to switch the LED on and off and the status of the LED was also transmitted correctly by Zigbee_Light_Bulb.ino. I used an ESP32-C6, but it should work just as well with the ESP32-H2.

superzanti, 2024-02-16:
@mitag Generally when pairing a device, I have tos et my coordinator in pairing mode AND the device in pairing mode.

I can initiate from the coordinator no problem (using the zigbee2mqtt interface).

So I'm assuming, if yours pairs perfectly after initiating from the coordinator, your ESP is either already paired to the coordinator (by manually setting some configs when flashing) or it flashes and starts in pairing mode. Can you clarify your setup?

mitag, 2024-02-16:
What I do know is that after flashing or resetting the ESP-C6, it is ready to perform a pairing initiated by the coordinator. I don't have to set any parameters anywhere, just let the coordinator search for new devices.
After resetting the ESP-C6 again, the pairing remains in place. If I delete the ESP-C6 in the coordinator, the pairing only works again after a reset of the ESP-C6.

As the example code refers to the importance of selecting the Partitio scheme (Zigbee 4MB with spiffs), I assume that the data transmitted by the coordinator during pairing is saved by one of the integrated libraries in the [spiffs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/spiffs.html) in order to survive a reset.

`
/**
 * @brief This example demonstrates simple Zigbee light bulb.
 * 
 * The example demonstrates how to use ESP Zigbee stack to create a end device light bulb.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator.
 * 
 * Proper Zigbee mode must be selected in Tools->Zigbee mode 
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 * 
 * Please check the README.md for instructions and more detailed description.
 * 
 */

`

I selected the following in the Arduino IDE, in Tools, because it seemed logical to me.
Partition Scheme: "Zigbee 4MB with spiffs"
...
Zigbee Mode: "Zigbee ED (end device)"

ilker-aktuna, 2024-02-17:
@mitag 
what do you mean by "reset the ESP-C6"
how do you reset it ?

thanks.

P-R-O-C-H-Y, 2024-02-17:
If the end device was never connected to the coordinator, just restarting the ESP can help by reset button. If it was already connected to some coordinator and you want to create new connection, you need to erase flash memory (it may also help to erase it for the first time to be sure, that the zigbee partition will be also clear). 

You can check the Readme of the example, all is written there:

If the End device flashed with this example is not connecting to the coordinator, erase the flash before flashing it to the board. It is recommended to do this if you did changes to the coordinator. You can do the following:

In the Arduino IDE go to the Tools menu and set Erase All Flash Before Sketch Upload to Enabled
In the sketch uncomment function esp_zb_nvram_erase_at_start(true); located in esp_zb_task function.

mitag, 2024-02-19:
Check the [logs ](https://docs2.hubitat.com/user-interface/settings/zigbee-logs)on your Hubidat.

ilker-aktuna, 2024-02-19:
but why doesn't these log_i , log_w lines work ?

log_i("Start network steering");

log_w("Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));

log_i("Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",

P-R-O-C-H-Y, 2024-02-19:
> but why doesn't these log_i , log_w lines work ?
>
>
>
> log_i("Start network steering");
>
>
>
> log_w("Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
>
>
>
> log_i("Joined network [...]

You need to set Debug Level in Arduino IDE - Tools menu to "Verbose" or "Info" to see those in serial monitor.

ilker-aktuna, 2024-02-24:
Debugs seem to work after I set it on IDE and restart.

With the example code , currently this is what happens with Hubitat hubs:

On C8 hub it gets stuck at "initializing" phase
On C7 hub it pairs (fast enough) but it is identified as a zigbee power outlet. I change it to zigbee switch , but I could not control it yet. on/off does not work. 
Just once it worked out of my attempts around hundred.
Someone from Hubitat forum says that it works on a C3 hub and that they could control it. However, same person could not make it work on a C7 and a C8 hub.

I really want this to work. But I'm not sure how to troubleshoot.
Debugs seem to work , but not really meaningful. 
Would anyone be interested in looking at the debug outputs , I have ?

Hedda, 2024-04-08:
> On C8 hub it gets stuck at "initializing" phase
> On C7 hub it pairs (fast enough) but it is identified as a zigbee power outlet. I change it to zigbee switch , but I could not control it yet. on/off does not work.
> Just once it worked out of my attempts around hundred.

FYI, you will get those same type of pairing/joning issues (including device interviewing failing or not completing) if you either have sources of EMF/EMI/RMI interference too close to the Zigbee devices (especially close to the Zigbee Coordinator and closest Zigbee Router) or similar problem symptoms you do not have enough "known good" Zigbee Router devices as well as also similar problem symptoms when having bad Zigbee Router devices that do not pass along all communication messages. See example:

* https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

Anyway, regardless of your enviroment and Zigbee gateway solution you are using, with all Zigbee setups you first be sure to always activly take preventive actions to avoid EMF/EMI/RMI interference by using a long USB 2.0 extension cable (or USB 3.0 extension cable to a USB 2.0 port/hub) as well as adding known good Zigbee Router devices. If possible try to follow all the tips in these hardware-independent best practise guides: 

* https://community.home-assistant.io/t/zigbee-networks-how-to-guide-for-avoiding-interference-optimizing-using-zigbee-router-devices-repeaters-extenders-to-get-best-possible-range-and-coverage/515752
* https://docs2.hubitat.com/en/how-to/build-a-solid-zigbee-mesh

Another tips; for easier troubleshooting from the Zigbee gateway side can I suggest that you buy one CC2652P based Zigbee Coordinator radio adapter dongle (like example the ["Sonoff ZBDongle-P" by ITead](https://www.google.com/search?q=%22Sonoff+ZBDongle-P%22)), + flash that with the [latest latest community Z-Stack Zigbee Coordinator (NCP) firmware build from Koenkk's master branch](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator), and then install the latest version of Zigbee2MQTT to test with that as Zigbee Coordinator USB dongle? -> https://www.zigbee2mqtt.io

Zigbee2MQTT has a great community for troubleshooting (and developing of) third-party devices from the Zigbee gateway side:

* https://github.com/Koenkk/zigbee2mqtt/discussions
  * https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Optionally, for some chaos engineering you could also buy the Home Assistant's [SkyConnect USB stick ](https://www.home-assistant.io/skyconnect/)and test with Home Assistant's [ZHA integration ](https://www.home-assistant.io/integrations/zha) as at least those should rule out of the problem is only with one Zigbee gateway solution or with all.

Hedda, 2024-04-09:
With Zigbee much also depends on the device handling on the Zigbee Gateway side, and depending on their implementation the Zigbee Gateway developers may need to make changes to support new devices. Thus you get different results with different software.

> Besides, there is really very few documentation on Zigbee zupport of Esp32-c6. Only a switch and a light bulb. What if I want to make a temperature sensor ? Or a rgb bulb ?

Check out the recently added additional examples:

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/

ilker-aktuna, 2024-04-09:
@Hedda 

ok thank you. I'll try these. But really we have to understand what's missing that causes my hub not pair with this device...

Avenitos, 2024-04-26:
i try join in SLS and z2m and I didn't succeed, the announcement comes and that's it, the device does not respond to interview requests

> [ 44665][I][sketch_apr26a.ino:102] esp_zb_app_signal_handler(): ZDO signal: ZDO Leave (0x3), status: ESP_OK
[ 44676][I][sketch_apr26a.ino:98] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 66175][I][sketch_apr26a.ino:102] esp_zb_app_signal_handler(): ZDO signal: ZDO Leave (0x3), status: ESP_OK
[ 66186][I][sketch_apr26a.ino:98] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 87685][I][sketch_apr26a.ino:102] esp_zb_app_signal_handler(): ZDO signal: ZDO Leave (0x3), status: ESP_OK
[ 87696][I][sketch_apr26a.ino:98] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)

Hedda, 2024-05-31:
FYI, Espressif released "stable" version 3.0.0 Arduino ESP32 Core based on ESP-IDF v5.1.4 with support for ESP32-H2 and ESP32-C6

https://github.com/espressif/arduino-esp32/releases/tag/3.0.0

(Arduino ESP32 core v3.0.0 was first announced over 7-month ago but this updated version is now finally been marked as stable).

That also means from now can report upstream problems or feature requests related to ESP32-H2/ESP32-C6 as new issues here:

https://github.com/espressif/arduino-esp32/issues/

PS: Espressif has still not created a Arduino wrapper library on top of the Zigbee SDK, they are however at least providing libraries.

chshu, 2024-07-26:
Closing, please file separate issues if anything else need to follow up for this topic.

closed 2024-07-26 by chshu

Hedda, 2024-10-07:
FYI, looks like a Zigbee API/wrapper for Arduino on ESP32 ([arduino-esp32](https://github.com/espressif/arduino-esp32)) is being working on very activly right now and initial (basic) Zigbee library support has been merged into both `master` and `release/v3.1` branches, so initial support and a few examples will be included in upcoming 3.0.6 and 3.1.0-rc2 releases of arduino-esp32. See:

* https://github.com/espressif/arduino-esp32/issues/10135