espressif/esp-zigbee-sdk #229: CRASH !! esp_zb_zcl_scenes_recall_scene_cmd_req is crashing

mundevx, 2024-01-25:
### Question

here is my code

esp_zb_zcl_scenes_recall_scene_cmd_t cmd_req;
cmd_req.group_id = 0;//group_id;
cmd_req.scene_id = 1;
cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = on_off_light.short_addr;
cmd_req.zcl_basic_cmd.dst_endpoint = on_off_light.endpoint;
cmd_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
esp_zb_zcl_scenes_recall_scene_cmd_req(&cmd_req);

The code crashes!

Here are the successful logs:
W (21903) ESP_HA_ON_OFF_LIGHT: ESP_ZB_CORE_CMD_GET_GROUP_MEMBERSHIP_RESP_CB_ID
W (21903) ESP_HA_ON_OFF_LIGHT: zb_group_get_membership_handler
I (21913) ESP_HA_ON_OFF_LIGHT: Received report from group_id(1) group_count(1) capacity(31) 
I (21923) ESP_HA_ON_OFF_LIGHT:  address(0x0) src endpoint(1) to dst endpoint(1) cluster(0x4)
I (23043) ESP_HA_ON_OFF_LIGHT: **************Sending scene add request**************
I (23043) ESP_ZIGBEE_CORE: Add scene(0x1) for group(0x0) to scenes table
I (23053) ESP_HA_ON_OFF_LIGHT: Received message: endpoint(1), cluster(0x5), group_id(0x0), scene_id(1)
I (23073) ESP_HA_ON_OFF_LIGHT: Received message: endpoint(1), cluster(0x5), group_id(0x0), scene_id(1)
I (26073) ESP_HA_ON_OFF_LIGHT: **************Sending scene membership request**************
I (26073) ESP_ZIGBEE_CORE: Send scene membership response
W (26073) ESP_HA_ON_OFF_LIGHT: zb_scene_get_membership_handler
I (26083) ESP_HA_ON_OFF_LIGHT: Received report from group_id(0) scene_count(1) capacity(15)
I (26093) ESP_HA_ON_OFF_LIGHT:  address(0x0) src endpoint(1) to dst endpoint(1) cluster(0x5)
I (26093) ESP_HA_ON_OFF_LIGHT:  scn_id[0](1)
+-------+----------+----------+-----------------+-------------------------------------+
| Index | Group ID | Scene ID | Transition Time |   [Cluster ID | Extendsion Field]   |
+-------+----------+----------+-----------------+-------------------------------------+
|   00  |  0x0000  |   0x01   |      0001       | [ 0x0005 | 0x11 ] |
+-------+----------+----------+-----------------+-------------------------------------+

The scene id is generated, but when i call scene recall command, it crashes.
abort() was called at PC 0x420197bf on core 0
0x420197bf: zb_assert at ??:?

Core  0 register dump:
MEPC    : 0x4080063a  RA      : 0x4080691e  SP      : 0x40818e90  GP      : 0x4080ca00
0x4080063a: panic_abort at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/panic.c:452

0x4080691e: __ubsan_include at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/ubsan.c:313

TP      : 0x408091c4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
0x408091c4: vPortSetupTimer at C:/Espressif/frameworks/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/portable/port_systick.c:111

xieqinan, 2024-02-05:
@mundevx ，

I believe that the usage of `esp_zb_zcl_scenes_recall_scene_cmd_req()` is correct. Which process the crashing occur? sending or receiving. 

To determine which process triggers the crashing for the `esp_zb_zcl_scenes_recall_scene_cmd_req()` command, we need to analyze the trace log when the crash occurs. 

Could you please provide the trace log when the crash happens? To debug the trace log, you can refer to the [documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging) for guidance on how to interpret and analyze it effectively.

mundevx, 2024-02-05:
> @mundevx ，
>
> I believe that the usage of `esp_zb_zcl_scenes_recall_scene_cmd_req()` is correct. Which process the crashing occur? sending or receiving.
>
> To determine which process triggers the [...]

I got crash on sending.

chshu, 2024-07-26:
@mundevx Could you please check if the issue still exist in latest esp-zigbee-sdk release (1.4.1)? If yes, could you share us more information for debugging, thanks.

mundevx, 2024-07-27:
@chshu , this issue is resolved!

closed 2024-07-31 by chshu