espressif/esp-zigbee-sdk #230: Deep slepp wake up cause

mrsteakhouse, 2024-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### esp-zigbee-lib version.

1.0.9

### esp-zboss-lib version.

1.0.9

### Espressif SoC revision.

Esp32-c6 devkit c1

### What is the expected behavior?

After waking the device from deep sleep through an ext1 configured gpio, the pin that caused the wake up is available through the api.

### What is the actual behavior?

Regardless of the wake up source, the reset reason is always 0x15 (uart).

### Steps to reproduce.

1. Build and flash the zigbee deep sleep example.
2. Trigger the wake up through one of the gpios.
3. Reset reason 0x15 is given.

### More Information.

I'm using the the Esp32c6 devkit c1 with the cable connected to usb and not the uart port.

mrsteakhouse, 2024-01-28:
I have done some testing with the ESP32C6-DevKit-C1 and a regular ESP32C6 module.
From what i can tell the reset cause is always 0x15 (USB_UART_HPSYS). Here is a log from the Devkit. The behavior of a regular module is the same.
The first reset was manual with the reset button. The second was induced from GPIO7 pulled low.

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0x2c (SPI_FAST_FLASH_BOOT)
Saved PC:0x40806438
0x40806438: rv_utils_wait_for_intr at /home/user/esp/esp-idf/components/riscv/include/riscv/rv_utils.h:33
 (inlined by) esp_cpu_wait_for_intr at /home/user/esp/esp-idf/components/esp_hw_support/cpu.c:119

SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd50
load:0x4086e610,len:0x2f6c
load:0x40875728,len:0x189c
entry 0x4086c410
I (23) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (23) boot: compile time Jan 28 2024 15:37:17
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 80MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (35) boot.esp32c6: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 000e1000
I (79) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (86) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=42068020 size=0fb58h ( 64344) map
I (119) esp_image: segment 1: paddr=0001fb80 vaddr=40800000 size=00498h (  1176) load
I (121) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=6373ch (407356) map
I (208) esp_image: segment 3: paddr=00083764 vaddr=40800498 size=10370h ( 66416) load
I (224) esp_image: segment 4: paddr=00093adc vaddr=50000200 size=00078h (   120) load
I (228) boot: Loaded app from partition at offset 0x10000
I (229) boot: Disabling RNG early entropy source...
I (244) cpu_start: Unicore app
I (245) cpu_start: Pro cpu up.
W (254) clk: esp_perip_clk_init() has not been implemented yet
I (260) cpu_start: Pro cpu start user code
I (261) cpu_start: cpu freq: 160000000 Hz
I (261) cpu_start: Application information:
I (263) cpu_start: Project name:     sleepy_end_device
I (269) cpu_start: App version:      1
I (274) cpu_start: Compile time:     Jan 28 2024 15:37:11
I (280) cpu_start: ELF file SHA256:  4c095f68e6df5e92...
I (286) cpu_start: ESP-IDF:          v5.1.2
I (291) cpu_start: Min chip rev:     v0.0
I (295) cpu_start: Max chip rev:     v0.99 
I (300) cpu_start: Chip rev:         v0.0
I (305) heap_init: Initializing. RAM available for dynamic allocation:
I (312) heap_init: At 408194D0 len 00063140 (396 KiB): D/IRAM
I (318) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (325) heap_init: At 50000278 len 00003D60 (15 KiB): RTCRAM
I (332) spi_flash: detected chip: generic
I (336) spi_flash: flash io: dio
W (340) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (353) sleep: Configure to isolate all GPIO pins in sleep state
I (360) sleep: Enable automatic switching of GPIO sleep configuration
I (367) coexist: coex firmware version: b6d5e8c
I (373) coexist: coexist rom version 5b8dcfa
I (378) app_start: Starting scheduler on CPU0
I (382) main_task: Started on CPU0
I (382) main_task: Calling app_main()
I (392) ESP_ZB_DEEP_SLEEP: Not a deep sleep reset
I (392) ESP_ZB_DEEP_SLEEP: Enabling timer wakeup, 20s
[... 68 lines cut ...]
I (332) spi_flash: detected chip: generic
I (336) spi_flash: flash io: dio
W (340) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (353) sleep: Configure to isolate all GPIO pins in sleep state
I (360) sleep: Enable automatic switching of GPIO sleep configuration
I (367) coexist: coex firmware version: b6d5e8c
I (373) coexist: coexist rom version 5b8dcfa
I (378) app_start: Starting scheduler on CPU0
I (382) main_task: Started on CPU0
I (382) main_task: Calling app_main()
I (392) ESP_ZB_DEEP_SLEEP: Not a deep sleep reset
I (392) ESP_ZB_DEEP_SLEEP: Enabling timer wakeup, 20s

I (402) phy_init: phy_version 230,c773401,Oct 30 2023,15:07:16
I (442) phy: libbtbb version: 7243671, Oct 30 2023, 15:07:30
I (452) main_task: Returned from app_main()
I (462) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (462) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized
I (742) ESP_ZB_DEEP_SLEEP: Device started up in non factory-reset mode
I (742) ESP_ZB_DEEP_SLEEP: Device rebooted
I (762) ESP_ZB_DEEP_SLEEP: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
```

The logs were generated from the deep sleep example of this repo without any modification.
Is there something obvious i'm doing wrong?

igrr, 2024-01-28:
If you are using the USB_SERIAL_JTAG port (rather than UART), please try running `idf.py monitor` with `--no-reset` flag. (ref. https://github.com/espressif/esp-idf/issues/12899 — different chip, but probably the same underlying issue that idf.py monitor resets the device once the port re-appears.)

mrsteakhouse, 2024-01-28:
Thank you for pointing me to the issue. It was indeed a reset from the jtag adapter. However, using the jtag connection with the flag results in random resets and random stuff printed on the console, but this is probably due to my setup.
Using the uart connection everything worked as expected and i got the pin mask causing the wale up. Thanks!

closed 2024-01-28 by mrsteakhouse