espressif/esp-zigbee-sdk #232: How do i get the Temp / Moisture / pm 2.5 from a IKEA VINDSTYRKA?

semteXKG, 2024-01-28:
### Question

Hi.
I have the above mentioned HA Device and i thought "sweet, it supports zigbee, how hard can it be". turns out: extremely hard.
It appears that I'm struggling with the concepts on how the reading of attributes  should work. I started with the "gateway" example, could pair the device and search for endpoints (found two).

however when i try to discover attributes (or read them) i have to provide a local endpoint. which one? the gateway example does not have one... somewhere i read i have to mirror the remote device locally? but i don't even know what the remote device supports... and then there is this "bind" and "unbind" thing and a couple of methods who go straight to the closed source part where i have no idea what the heck is going on behind it....

the whole thing is super confusing and extremely frustrating to me... i have now spent 10 or more hours on the thing and i still have no idea what the heck i am doing...

Thank you very much and if someone wants to provide me with a sample code i would be so thankful!

sssionek, 2024-01-28:
Generally, a basic Coordinator example with ZED endpoint/cluster discovery (and maybe some attr reporting) would be quite appreciated. I also have been trying to set up a simple network and hook up my Sonoff temperature sensor to it - I can get the device associated, but couldn't get it to report parameters to me.
Here's my jerryrigged testing code, if you'd have any comments etc, i'd be glad to hear them.
```
#include <esp_log.h>
#include <string.h>
#include <esp_zigbee_core.h>
#include "ha/esp_zigbee_ha_standard.h"
#include "thread.h"

#define TAG "zb_thread"

static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Some bondage has happened!");
    }
}

static void node_cb(esp_zb_zdp_status_t zdo_status, uint16_t addr, esp_zb_af_node_desc_t *node_desc, void *user_ctx)
{
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Got node description!");
        ESP_LOGI(TAG, "Manufacturer desc %02X",node_desc->manufacturer_code);
    }
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
	esp_zb_app_signal_type_t sig_type = *(signal_struct->p_app_signal);
	esp_err_t err_status = signal_struct->esp_err_status;
	esp_zb_zdo_signal_device_annce_params_t *dev_annce_params = NULL;
	switch (sig_type) {
		case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
			ESP_LOGI(TAG,"Zigbee stack initialized");
			//once initialized - form a network
			esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
			break;
		case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
			ESP_LOGI(TAG ,"ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START");
		case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
			ESP_LOGI(TAG, "ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT");
			//reboot shows up after network formation - allow for joins
			if (err_status == ESP_OK) {
				ESP_LOGI(TAG, "Start network steering");
			} else {
				ESP_LOGE(TAG, "Failed to initialize Zigbee stack (status: %d)", err_status);
			}
			break;
		case ESP_ZB_BDB_SIGNAL_FORMATION:
			ESP_LOGI(TAG, "ESP_ZB_BDB_SIGNAL_FORMATION");
			if (err_status == ESP_OK) {
				esp_zb_ieee_addr_t extended_pan_id;
				esp_zb_get_extended_pan_id(extended_pan_id);
				ESP_LOGI(TAG, "Formed network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",
						extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
						extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
						esp_zb_get_pan_id(), esp_zb_get_current_channel(), esp_zb_get_short_address());
				esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
			}
			break;
		case ESP_ZB_BDB_SIGNAL_STEERING:
			ESP_LOGI(TAG, "ESP_ZB_BDB_SIGNAL_STEERING");
			break;
		case ESP_ZB_NWK_SIGNAL_DEVICE_ASSOCIATED:
			ESP_LOGI(TAG, "ESP_ZB_ZDO_SIGNAL_DEVICE_ASSOCIATED");
			break;
		case ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE:
			ESP_LOGI(TAG, "ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE");
			dev_annce_params = (esp_zb_zdo_signal_device_annce_params_t *)esp_zb_app_signal_get_params(signal_struct->p_app_signal);
			ESP_LOGI(TAG, "New device commissioned or rejoined (short: 0x%04hx)", dev_annce_params->device_short_addr);
[... 77 lines cut ...]
        .install_code_policy = false, //disable install code policy (????)
        .nwk_cfg.zczr_cfg = {
            .max_children = 5, //max amount of connected devices
        },
    };
    esp_zb_init(&zb_nwk_cfg);
    //basic cluster init, i guess?
    uint8_t test_attr = 0;
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr);
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_update_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, 1, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_TEST_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
    // network setup & loop
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(1l << 13); // Zigbee primary channel mask
    ESP_ERROR_CHECK(esp_zb_start(false));

    esp_zb_main_loop_iteration();
}
```

djorr5, 2024-01-29:
I second the "a basic Coordinator example with ZED endpoint/cluster discovery (and maybe some attr reporting) would be quite appreciated." comment.

Would be great to have a coordinator example that commercially available end devices can connect to and auto discover their end points/attributes.

xieqinan, 2024-02-05:
Hello @semteXKG ,

I appreciate your patience and support. However, having a basic understanding of Zigbee is necessary for working with the esp-zigbee-sdk.

For obtaining address information, when a remote device joins the coordinator, it announces its basic address. You can obtain the short or IEEE address in the `ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE` signal. For example:
```c
  case ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE:
      dev_annce_params = (esp_zb_zdo_signal_device_annce_params_t *)esp_zb_app_signal_get_params(p_sg_p);
      ESP_LOGI(TAG, "New device commissioned or rejoined (short: 0x%04hx)", dev_annce_params->device_short_addr);
      esp_zb_zdo_match_desc_req_param_t  cmd_req;
      cmd_req.dst_nwk_addr = dev_annce_params->device_short_addr;
      cmd_req.addr_of_interest = dev_annce_params->device_short_addr;
      esp_zb_zdo_find_on_off_light(&cmd_req, user_find_cb, NULL);
      break;
```

Regarding endpoint information, if you expect to get the `temp/moisture/pm2.5` value from the remote device, having client clusters for `temp/moisture/pm2.5` is necessary. The esp-zigbee-sdk provides `esp_zb_temperature_meas_cluster_create()` and `esp_zb_pm2_5_measurement_cluster_create()` for creating these related clusters. The source or local endpoint (ranging from 1 to 240) is required to fill when you add clusters to `endpoint_list`. The `dst_endpoint` can be obtained by `esp_zb_zdo_active_ep_req()`. Refer to this [example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L148).

Regarding the address mode, if you obtain the above information, the `ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT` mode is fine.

The parameters `cluserID`, `attr_number`, and `attr_field` can be referred to in the `2.5.1 Read Attributes Command` section of the Zigbee Cluster Library Specification revision 8.

xieqinan, 2024-02-05:
Hello @sssionek @djorr5,

The esp-zigbee-sdk provides a [customized example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_customized_devices) that may meet your requirements. It offers a way to make the light report its on/off attribute, which is similar to the temperature cluster.

Furthermore, we also have plans to add more examples or more simple APIs for users, such as `a basic Coordinator example with ZED endpoint/cluster discovery`. Before that, feel free to raise any problems or issues you encounter here, and I will do my best to resolve them.

semteXKG, 2024-02-05:
@xieqinan thank you very much for your great explanation! one additional question: if i probe a device i get information like "cluster ID 1066" which should be the pm2.5 sensor... but i don't know that.. and i did not find any lookup table in the spec mapping those IDs to human readable descriptions.

is there such a thing and did i just miss it? Also i bought a sonoff gateway which should make the whole probing and identifying zigbee devices thing less painfull.

xieqinan, 2024-02-05:
@semteXKG @djorr5 @sssionek ,

I have attached a raw example addressing the issue of "a basic Coordinator example with ZED endpoint/cluster discovery". You can refer to it, as it provides a local configuration for attribute reporting and a way to report the attribute to the coordinator.
[config_report.zip](https://github.com/espressif/esp-zigbee-sdk/files/14163858/config_report.zip)

djorr5, 2024-02-06:
@xieqinan , thanks for the prompt response and providing some more sample code. However it looks like this code doesn't do auto discovery of nodes, clusters or attributes for connected end devices. It more "assumes" that a newly connected device is going to be reporting Temp and Humidity. 

There was another issue that was opened, but since closed, that was requesting more information around this as well. [https://github.com/espressif/esp-zigbee-sdk/issues/55](url)

If I/we could get an example that would show a coordinator that allows joining of end devices/routers and allows discovery of their capabilities that would help me out tremendously.

xieqinan, 2024-02-06:
@djorr5 ,

The `esp_zb_zdo_active_scan_request()` function provides developers with a way to scan for devices nearby. Similarly, the `esp_zb_zdo_active_ep_req()` function allows developers to retrieve the active endpoints on a device within the network. Additionally, the `esp_zb_zcl_disc_attr_cmd_req()` function enables the discovery of attributes on an endpoint. These methods should be sufficient to meet your requirements.

semteXKG, 2024-02-07:
alright, i think i made progress but i'm still stuck. again, what i'm trying to do:
I wand to run the ESP32_C6 as coordinator, which opens a network and binds to a temperature sensor, receiving periodic updates.
this is now the code i wrote and binding is OK + command is executed, but i don't get any attribute updates in the callback.

Also the temp cluster is now added as client role.

Also: is there some documentation which describes how this whole cluster / coordinator / binding / ... thing works together? honestly i still have the feeling i'm guessing in most cases on what address to enter where (local, remote, coordinator, ...)
```
static const char *TAG = "ESP_ZB_GATEWAY";

typedef struct device_ {
    esp_zb_ieee_addr_t ieee_addr;
    uint8_t endpoint;
    uint16_t short_addr;
} device_t;

typedef struct bind_ctx_s {
    uint16_t short_addr;
    uint8_t dst_ep;
    uint8_t src_ep;
    uint16_t cluster_id;
} bind_ctx_t;

bind_ctx_t g_bind_ctx = {};
device_t g_device = {};
uint16_t g_clusters[1] = {
    ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT,
};

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_ERROR_CHECK(esp_zb_bdb_start_top_level_commissioning(mode_mask));
}

static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
    device_t* device = (device_t*) user_ctx;

    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Bound successfully!");
        esp_zb_zcl_report_attr_cmd_t reportArgs = {
            .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
            .zcl_basic_cmd.dst_addr_u.addr_short = device->short_addr,
            .zcl_basic_cmd.src_endpoint = MEASUREMENT_ENDPOINT,
            .zcl_basic_cmd.dst_endpoint=1,
            .cluster_role = ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE,
            .attributeID = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID,
            .clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT    
        }; 
        

        esp_err_t err_status = esp_zb_zcl_report_attr_cmd_req(&reportArgs);
        ESP_LOGI(TAG, "Command Returned: %s", esp_err_to_name(err_status));
    }
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    esp_zb_zdo_signal_device_annce_params_t *dev_annce_params = NULL;
    esp_zb_zdo_signal_macsplit_dev_boot_params_t *rcp_version = NULL;

    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Zigbee stack initialized");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_MACSPLIT_DEVICE_BOOT:
[... 99 lines cut ...]

    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_temperature_meas_cluster_cfg_t temp_cfg = {
        .measured_value = 0x954d,
        .min_value = 0x954d,
        .max_value = 0x7fff,
    };

    
    esp_zb_cluster_list_add_basic_cluster(cluster_list, esp_zb_basic_cluster_create((NULL)), ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
    esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster_create((NULL)), ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
    esp_zb_cluster_list_add_temperature_meas_cluster(cluster_list, esp_zb_temperature_meas_cluster_create(&(temp_cfg)),
                                                     ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

    esp_zb_ep_list_add_ep(ep_list, cluster_list, MEASUREMENT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_TEST_DEVICE_ID);
    esp_zb_device_register(ep_list);

    /* initiate Zigbee Stack start without zb_send_no_autostart_signal auto-start */
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

semteXKG, 2024-02-10:
alright, i got it! it's working :D thanks for your help 
[esp_zigbee_gateway_and_sensor_reader.zip](https://github.com/espressif/esp-zigbee-sdk/files/14230026/esp_zigbee_gateway_and_sensor_reader.zip)

closed 2024-02-10 by semteXKG