espressif/esp-zigbee-sdk #233: esp_zb_zcl_report_attr_cmd_t not working for humidity or pressure clusters

kylemwagner83, 2024-01-28:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### esp-zigbee-lib version.

0.9.0

### esp-zboss-lib version.

0.7.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Update and report ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_VALUE_ID and ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_VALUE_ID

### What is the actual behavior?

Reporting works fine for temperature, but not for humidity or pressure.
Nothing happens when esp_zb_zcl_report_attr_cmd_t is called for humidity/pressure, no errors are thrown, and the values are not reported. 

### Steps to reproduce.

Here is the esp_zb_task function, defining the clusters, attributes/default values, cluster/endpoint lists:

```
static void esp_zb_task(void *params)
{
    // Define cluster attributes
    char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};
    char modelid[] = {14, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'S', 'e', 'n', 's', 'o', 'r'};

    int tempValue = 2300;
    int tempMin = 0;
    int tempMax = 32000;

    int humidityValue = 4000;
    int humidityMin = 0;
    int humidityMax = 9999;
    int humidityTolerance = 1;

    int pressureValue = 100;
    int pressureMin = 0;
    int pressureMax = 32000;

    // Initialize Zigbee stack
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    // Create genBasic cluster/attribute list
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);

    // Create temperature cluster/attribute list
    esp_zb_attribute_list_t *esp_zb_temperature_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT);
    esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &tempValue);
    esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MIN_VALUE_ID, &tempMin);
    esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MAX_VALUE_ID, &tempMax);

[... 135 lines cut ...]
    static esp_zb_zcl_report_attr_cmd_t pressure_cmd_req = {
        .zcl_basic_cmd.src_endpoint = 1,
        .address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_PRESSURE_MEASUREMENT,
        .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        .attributeID = ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_VALUE_ID
    };
    state = esp_zb_zcl_report_attr_cmd_req(&pressure_cmd_req);

    /* Error check */
    if(state != ESP_ZB_ZCL_STATUS_SUCCESS)
    {
        ESP_LOGE(TAG, "Reporting pressure attribute failed!");
        return ESP_FAIL;
    }
    
    return ESP_OK;
}
```

### More Information.

The zb_update_temperature function works as expected, the ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID is updated, and the new value is properly reported to the coordinator.

The zb_update_humidity and zb_update_pressure don't work, it appears the attributes are updated in their respective clusters (I can see the updated values if I manually read the cluster), but are not automatically reported.

xieqinan, 2024-02-02:
@kylemwagner83,

I believe that the temperature, humidity, and pressure attributes are similar. The issue may arise from different configurations for attribute reporting. Could you please show me how to configure report to the Zigbee device in  the coordinator side? A Wireshark packet or code will be fine.

xieqinan, 2024-02-04:
@kylemwagner83,

Thank you for the detailed information. I believe that Zigbee2MQTT has already configured the attribute reporting. However, I am not sure whether the report information on the ESP32H2 side is correct. Could you please utilize the `esp_zb_zcl_find_reporting_info()` function to locate the attribute report information? This function can help us determine whether the humidity and pressure clusters can correctly send the attribute reports.

kylemwagner83, 2024-02-10:
Thank you for the advice, `esp_zb_zcl_find_reporting_info()` was not available in esp-zigbee-lib 0.9.0, so in idf_component.yml I updated both esp-zigbee-lib and esp-zboss-lib to 1.1.1 
This seems to have fixed the humidity and pressure reporting, so now all 3 attributes (temp, humidity, pressure) are updating in MQTT - if I set reporting intervals in Zigbee2MQTT.

I believe I'm still missing some reporting configuration to set report intervals in ESP-IDF code rather than in Zigbee2MQTT. I'm guessing this uses something like `esp_zb_zcl_update_reporting_info()`. Is this correct and are there any code examples for setting report intervals?

kylemwagner83, 2024-02-11:
Seems like there is still an issue with reporting, without manually setting report intervals in Zigbee2MQTT nothing is reported, `esp_zb_zcl_report_attr_cmd_req()` doesn't send anything when called. I'm not sure what information would be most useful from the `esp_zb_zcl_find_reporting_info()` command, I just printed what I can to console:

```
    esp_zb_zcl_attr_location_info_t humidity_attr_info = {
        HA_ESP_SENSOR_ENDPOINT,
        ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT,
        ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC, //
        ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_VALUE_ID,
    };

    esp_zb_zcl_reporting_info_t humidity_reporting_state = *esp_zb_zcl_find_reporting_info(humidity_attr_info);

    printf("Humidity report - direction: %i\n", humidity_reporting_state.direction);
    printf("Humidity report - endpoint: %i\n", humidity_reporting_state.ep);
    printf("Humidity report - cluster id: %i\n", humidity_reporting_state.cluster_id);
    printf("Humidity report - cluster role: %i\n", humidity_reporting_state.cluster_role);
    printf("Humidity report - attribute id: %i\n", humidity_reporting_state.attr_id);
    printf("Humidity report - flags: %i\n", humidity_reporting_state.flags);
    printf("Humidity report - run time: %lu\n", (unsigned long)humidity_reporting_state.run_time);
    printf("Humidity report - actual min interval: %i\n", humidity_reporting_state.u.send_info.min_interval);
    printf("Humidity report - actual max interval: %i\n", humidity_reporting_state.u.send_info.max_interval);
    printf("Humidity report - default min interval: %i\n", humidity_reporting_state.u.send_info.def_min_interval);
    printf("Humidity report - default max interval: %i\n", humidity_reporting_state.u.send_info.def_max_interval);
    printf("Humidity report - timeout: %i\n", humidity_reporting_state.u.recv_info.timeout);
    printf("Humidity report - dst short address: %i\n", humidity_reporting_state.dst.short_addr);
    printf("Humidity report - dst endpoint: %i\n", humidity_reporting_state.dst.endpoint);
    printf("Humidity report - dst profile id: %i\n", humidity_reporting_state.dst.profile_id);
    printf("Humidity report - manufacturer code: %i\n", humidity_reporting_state.manuf_code);
```

Here is the output:

```
Humidity report - direction: 0
Humidity report - endpoint: 1
Humidity report - cluster id: 1029
Humidity report - cluster role: 1
Humidity report - attribute id: 0
Humidity report - flags: 33
Humidity report - run time: 425
Humidity report - actual min interval: 5
Humidity report - actual max interval: 0
Humidity report - default min interval: 5
Humidity report - default max interval: 0
Humidity report - timeout: 5
Humidity report - dst short address: 0
Humidity report - dst endpoint: 0
Humidity report - dst profile id: 260
Humidity report - manufacturer code: 65535
```

When I try to configure the report using `esp_zb_zcl_config_report_cmd_req()`, it crashes (specifically when setting anything in the record_field), here is the code and output:

```
    esp_zb_zcl_config_report_record_t report_record;
    report_record.attributeID = ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_VALUE_ID;
    report_record.attrType = ESP_ZB_ZCL_ATTR_TYPE_U32;
    report_record.min_interval = 1;
    report_record.max_interval = 4;
    report_record.reportable_change = 0;

    esp_zb_zcl_config_report_cmd_t report_cmd;
    report_cmd.zcl_basic_cmd.dst_addr_u.addr_short=esp_zb_get_short_address();
    report_cmd.zcl_basic_cmd.dst_endpoint = 1;
    report_cmd.zcl_basic_cmd.src_endpoint = 1;
    report_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    report_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT;
    report_cmd.record_field = &report_record;

[... 77 lines cut ...]
4081d2a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d2c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d2e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d300: 0x40000000 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
0x40000000: _start in ROM

4081d320: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081d340: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081d360: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081d380: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081d3a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081d3c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081d3e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081d400: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081d420: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 9da32ebc678fbdb3

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a: software_reset_cpu in ROM
```

xieqinan, 2024-02-29:
@kylemwagner83 ，

Please use the `esp_zb_zcl_update_reporting_info()` function to update the report information, then use the `esp_zb_zcl_find_reporting_info()` function to verify whether it has been updated.

The attribute type of `ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_VALUE_ID` should be `ESP_ZB_ZCL_ATTR_TYPE_U16`. Could you update it and try again?

kylemwagner83, 2024-03-16:
I ended up restructuring my program using the HA_Temperature_Sensor/HA_Thermostat examples and the reporting seems to be working now. At least part of the problem seems to have been calling esp_zb_zcl_update_reporting_info() after the main loop iteration. Handling the reporting configuration as part of the cluster/attribute setup in esp_zb_task prior to starting the main loop works. Thanks!