espressif/esp-zigbee-sdk #234: How to increase the size of scene table?

mundevx, 2024-01-30:
### Question

I am getting this error:
ESP_ZIGBEE_CORE: Insufficient scenes table for scene(0x5), fail to store it

Then i read espressif documentation for zigbee:-
https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/zcl/esp_zigbee_zcl_command.html
There is written _The maximum number of scenes capable of being stored in the table is 10._

How i can increase this size for scene store table? and also in case of group.

mundevx, 2024-01-31:
Hi, i am still unable to find the file where i can increase size of scene table.

chshu, 2024-02-01:
The max scene table size is 16 in current 1.0.9 SDK, and not configurable. It should meet most use cases, could you let us know your use case and the table size you need?

mundevx, 2024-02-01:
Hi @chshu ,

I had a 12-scene 4-switch wireless scene switch.
For every switch had 3 scenes, Single click, Double Click, Long Press. 
I had 4 lights where espressif light code is running with 4 endpoints.

If i create a scene for single click from app for 4 lights, then it add 4 records in scene table. 
Below table is for scene created for 4 lights from app for SINGLE CLICK only.
............................................................................................................................................................
index     |     group_id    |     scene_id     |      endpoint         |
0            |          1           |          1            |      1                      |              
1            |          1           |          1            |      2                      |      
2            |          1           |          1            |      3                      |     
3            |          1           |          1            |      4                      | 
...............................................................................................................................................................
If i creates scenes on click modes Single, double and long press, entry in table extend upto 3x4(lights) = 12.
Now i have 4 switches in a 4-gang scene switch, so total scenes table entry is 3x4(lights)x4(switches) = 48. 
Total entries in scene table should be 48, but after 16 scene entries in a table, I am getting this error:
_**ESP_ZIGBEE_CORE: Insufficient scenes table for scene(0x5), fail to store it**_.

Suppose i am using 64 lights in a scene, then scenes size reached upto 3x64x4 = 768 table entries.

I my use case, i need 500 max size for scene table.

mundevx, 2024-02-05:
As i checked with other zigbee lights, that don't have espressif code, it is able to store scene more than 16 entries. i cheked it with my scene switch. 
How i can increase scene table size?

chshu, 2024-02-18:
@mundevx The scene table size is not configurable yet, we will make it dynamic allocation in the future releases.

closed 2025-08-29 by xieqinan

mundevx, 2025-08-30:
> Hi,
>
> The esp-zigbee-sdk v1.6.7
> has provide the esp_zb_zcl_scenes_table_set_size() to configure the
> capacity of scene_table, please update this version to test again.
>
> Since the issue has persisted for a while and has now been fixed, let’s
> close it. If you have any further questions, feel free to reopen it.

Ok Sure, I will test it and will back you shortly.