espressif/esp-zigbee-sdk #237: esp_zb_mfg_tool on ESP IDE (eclipse) on Windows 10.

RadekDymak, 2024-02-02:
### Question

Hi, I try to use esp_zb_mfg_tool on Windows 10 in ESP IDE(Version: 2.12.0)

I have esp_zb_mfg_tool  form esp-zigbee-sdk 1.0.9

I install   " pip install -r requirements.txt" correctly.

But when i run:  esp_zb_mfg_tool.py -i 966b9f3ef98ae6059708 -m CAFEBEEF50C0FFED -c 0x8000 -mn Espressif -mc 0x131B

then i get this error :

C:\Espressif\workspace\esp-zigbee-sdk-main\esp-zigbee-sdk-main\tools\mfg_tool>python esp_zb_mfg_tool.py -i 966b9f3ef98ae6059708 -m CAFEBEEF50C0FFED -c 0x8000 -mn Espressif -mc 0x131B
Traceback (most recent call last):
  File "esp_zb_mfg_tool.py", line 522, in <module>
    main()
  File "esp_zb_mfg_tool.py", line 518, in main
    generate(args)
  File "esp_zb_mfg_tool.py", line 445, in generate
    os.mknod(file_name)
AttributeError: module 'os' has no attribute 'mknod'

What is wrong?

xieqinan, 2024-02-04:
@RadekDymak ，

The `os.mknod` function is typically utilized for creating special files in Unix systems, a functionality not supported in Windows systems.

As a workaround for this issue, I recommend commenting out the following code:

```python
# if not os.path.exists(file_name):
#     os.mknod(file_name)
```

This will prevent the attempt to create special files using `os.mknod` in a Windows environment, avoiding potential errors.

xieqinan, 2024-02-04:
Additionally, if the script logs `module 'nvs_partition_gen' has no attribute 'set_target_filepath'`, please replace `import nvs_partition_gen` with `import esp_idf_nvs_partition_gen.nvs_partition_gen as nvs_partition_gen`.

RadekDymak, 2024-02-05:
Commenting lines above help me, but There is some new error:

unsupported operand type(s) for +: 'NoneType' and 'str'

Do you know, ho to solve this?

closed 2024-02-16 by RadekDymak