espressif/esp-zigbee-sdk #238: New Gateway based on OT_RCP fails

djorr5, 2024-02-03:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2

### esp-zigbee-lib version.

1.1

### esp-zboss-lib version.

1.1

### Espressif SoC revision.

ESP32S3 & ESP32H2

### What is the expected behavior?

To work

### What is the actual behavior?

I have the Thread Border Router Board and I have tried to setup the new ot_rcp and Zigbee gateway examples on it. It fails always at " Zigbee stack initialized"  and gives me a 
"abort() was called at PC 0x420128c6 on core 0
0x420128c6: zb_assert at ??:?"

This is unmodified code and I have updated everything. I flash the ot_rcp to the H2 chip first and then flash the zigbee gateway to the S3 chip second. 

I have also made sure to run " echo "CONFIG_OPENTHREAD_NCP_VENDOR_HOOK=y" >> sdkconfig.defaults" in powershell from the correct directory as that didn't seem to run whilst building. 

Assuming that it couldn't communicate via UART between the chips correctly I updated the UART pins via menuconfig for both chips. However I am unsure if I have this correct. Is it requesting the actual chip pin number or the GPIO number? For the S3 chip (after googling)  I chose Pin 43 for RCP TX and Pin 44 for RCP RX. And for the H2 I chose 23 for the RX Pin and 24 for the TX Pin. Are these correct. I would have assumed that I shouldn't need to modify these from their defaults when building? 

Or is there something else making it fail?

### Steps to reproduce.

1. Build ot_rcp from IDF framework v5.2 and flash to esp32h2 - after running set-target esp32h2
2. Build esp_zigbee_gateway from github and flash to esp32s3 - running set-target esp32s3
3. Monitor esp32s3
...

### More Information.

`
```c++

 I (5083) ESP_ZB_GATEWAY: Zigbee stack initialized

abort() was called at PC 0x420128c6 on core 0
0x420128c6: zb_assert at ??:?

Backtrace: 0x40375ae2:0x3fcb26a0 0x4037cd45:0x3fcb26c0 0x403845c1:0x3fcb26e0 0x420128c6:0x3fcb2750 0x4203cbeb:0x3fcb2770 0x4203cc13:0x3fcb2790 0x42039bd2:0x3fcb27b0 0x42039be3:0x3fcb27d0 0x42016d18:0x3fcb27f0 0x4202de03:0x3fcb2820 0x4200a513:0x3fcb2840 0x42009737:0x3fcb2860 0x4037d90d:0x3fcb2890
0x40375ae2: panic_abort at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/panic.c:472
0x4037cd45: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/port/esp_system_chip.c:93
0x403845c1: abort at C:/Espressif/frameworks/esp-idf-v5.2/components/newlib/abort.c:38
0x420128c6: zb_assert at ??:?
0x4203cbeb: error_unhandled at zb_error_indication.c.obj:?
0x4203cc13: zb_error_raise at ??:?
0x42039bd2: zb_macsplit_transport_do_retransmit at zb_macsplit_transport.c.obj:?
0x42039be3: zb_macsplit_transport_wait_for_ack_timeout at zb_macsplit_transport.c.obj:?
0x42016d18: zb_sched_loop_iteration at ??:?
0x4202de03: zboss_main_loop_iteration at ??:?
0x4200a513: esp_zb_main_loop_iteration at ??:?
0x42009737: esp_zb_task at C://GitHub/esp-idf/esp-zigbee-sdk/examples/esp_zigbee_gateway/main/esp_zigbee_gateway.c:219
0x4037d90d: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
`

xieqinan, 2024-02-04:
@djorr5 ,

The new feature of the gateway based on OT-RCP is only supported in the [esp-idf master](https://github.com/espressif/esp-idf) and [esp-idf release/v5.1](https://github.com/espressif/esp-idf/tree/release/v5.1) branches, and support for `esp-idf v5.2` will be added in the future. Documentation for the supported IDF version will be added immediately.

chshu, 2024-02-04:
Please use the esp-idf master branch ([Commit ad3e1e3](https://github.com/espressif/esp-idf/commit/ad3e1e3daf7037e58c650c735e1f6218ee886651) or later).

The note is also added to the example readme: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway.

Feel free to reopen if any further questions.

closed 2024-02-04 by chshu