espressif/esp-zigbee-sdk #239: Crash on HA_color_dimmable_light when pairing a remote.

edmeme, 2024-02-04:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1724-ga5b261f699

### esp-zigbee-lib version.

1.1.0

### esp-zboss-lib version.

1.1.0

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

Can connect a switch to the HA_color_dimmable_light example.

### What is the actual behavior?

When a switch is connected (testing with an IKEA Trådfri switch) the example crashes

[log.log](https://github.com/espressif/esp-zigbee-sdk/files/14158923/log.log)
[stack.txt](https://github.com/espressif/esp-zigbee-sdk/files/14158924/stack.txt)

 ```
Stack memory:
4081e130: 0x00000008 0x4081e168 0x4081e190 0x4080c7ba 0x00000000 0x0000001f 0x4081e1a0 0x4080e6b8
4081e150: 0x4081e194 0x4080e6d4 0x4081e190 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4081e170: 0x43502074 0x34783020 0x63313032 0x20646465 0x63206e6f 0x2065726f 0x00000030 0x42020000
...

ELF file SHA256: 14dff145f

Rebooting...
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400031b6
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460e0,len:0x17ac
load:0x4083cad0,len:0xe88
load:0x4083efd0,len:0x2c94
entry 0x4083cada
```

### Steps to reproduce.

0. System upgrade (archlinux)
`# pacman -Suy`
1. update IDF and the examples
` cd esp-idf ; git pull ; git submodule update --init --recursive ; ./install.sh`
`git describe --tags` now reports _v5.3-dev-1724-ga5b261f699_
`cd esp-zigbee-sdk ; git pull`
last commit is `a5b261f699808efdacd287adbded5b718dffd14e`
2. build and flash
```
source esp-idf/export.sh
cd esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_color_dimmable_light
idf.py set-target esp32h2
idf.py build
touch main/esp_zb_light.c # The first time I build I do not get a valid program, for some reason.
idf.py build
```
3. plug-in my board (ESP Thread Border Router/ESP Gateway v1.2)
`idf.py flash --port=/dev/ttyACM0`

4. press 5 times the link button in my remote to pair.
5. The following line appears in the log
`I (37373) ESP_ZB_COLOR_DIMM_LIGHT: ZDO signal: NWK Device Associated (0x12), status: ESP_OK`
Then the crash happens.

Full log and an attempt to identify all symbols in the stack trace are attached.
[log.log](https://github.com/espressif/esp-zigbee-sdk/files/14158923/log.log)
[stack.txt](https://github.com/espressif/esp-zigbee-sdk/files/14158924/stack.txt)

xieqinan, 2024-02-05:
Hello,

The issue appears to be triggered by an error short address.

In the `ESP_ZB_COLOR_DIMM_LIGHT` example, which is a centralized router, the short address should not be 0x0000.

The indication of `NWK Device Associated` suggests that it received an association request that should have been received by the coordinator rather than the router.

Firstly, you should confirm the remote device type. Is it a coordinator?

Secondly, you can check the joining process in a sniffer to determine what short address is assigned to the router by the coordinator.

Thirdly, what is the role of `ESP Thread Border Router/ESP Gateway v1.2` in your project?

xieqinan, 2024-03-14:
@edmeme ,

The compatibility of Zigbee economics is a contentious issue. We are unsure of the methods employed by devices like the `IKEA Trådfri switch` to prevent unauthorized devices from joining their network, such as install codes, link keys, manufacturer codes, and so on. Therefore, a packet sniffer is necessary for us to investigate and address this issue.

xieqinan, 2024-06-17:
@edmeme 

Do you have any updates on this issue?

closed 2024-07-26 by chshu