espressif/esp-zigbee-sdk #240: [REQUEST] Include the Zigbee RCP example (from esp-idf) in this esp-zigbee-sdk repo

Hedda, 2024-02-06:
### Is your feature request related to a problem?

Requesting you also include "Zigbee RCP" (Radio Co-Processor) example from esp-idf in this esp-zigbee-sdk repository as well:

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/esp_zigbee_rcp

That is currently only included under examples in esp-idf but not included under examples this esp-zigbee-sdk repository:

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples

If not include or move it here then at least create a placeholder with a readme that has a URL link to the upstream repository.

It can now be confusing which is the upstream example and which is the latest so not use an old example, this also goes for:

* https://github.com/espressif/esp-idf/tree/master/examples/zigbee/esp_zigbee_gateway
* https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample
  * https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample/HA_on_off_light
  * https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample/HA_on_off_switch

Suggest consolidating all Zigbee examples in only one repository and then just create placeholders with links in other repositories.

PS: Off-topic but suggest you also consider including a similar Co-Processor Designs clarification/explanation as OpenThread does:

https://openthread.io/platforms/co-processor

chshu, 2024-02-06:
@Hedda As you know, the new ESP Zigbee Gateway solution is now based on the openthread spinel protocol, the same [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) example could be used for both Thread Border Router and Zigbee Gateway.

We don't want to maintain duplicate examples in multiple repos, that's why there is no Thread RCP example in the [Thread BR repo](https://github.com/espressif/esp-thread-br/tree/main/examples ) as well. Users should be able to setup the example following the [readme](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway).

Want to note that, benefit from the IDF [components registry](https://components.espressif.com/), our users even don't need to download this esp-zigbee-sdk repo, they only need to download esp-idf and specify the zigbee lib component dependency in their application code ([an example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_gateway/main/idf_component.yml)).

teleger, 2024-02-21:
when I use zigbee_gateway project.

name of address: https://github.com/espressif/esp-zigbee-sdk 

zigbee_gateway 
managed_components/espressif_esp-zboss-lib/ CMakeLists.txt

if (CONFIG_ZB_RADIO_SPINEL_UART)
    idf_component_get_property(openthread_lib openthread COMPONENT_LIB)
    target_link_libraries(${COMPONENT_LIB} INTERFACE zboss_port_lib $<TARGET_FILE:${openthread_lib}> zboss_port_lib)
endif()
Why can't I find the openthread library, do I need to compile it beforehand?
thank you!

teleger, 2024-02-21:
so,now  i used gateway project demo should be
https: [//github.com/espressif/esp-zigbee-sdk]
is ok?

chshu, 2024-03-08:
@teleger Did you use some old IDF version where there is no openthread component under IDF? Please share the entire failure log if you still see the compile issue with IDF v5.1.3.

Please create separate issues for your follow up questions.

closed 2024-03-08 by chshu