espressif/esp-zigbee-sdk #241: Warning showing up in new OT-Gateway example

djorr5, 2024-02-06:
### Question

I built and flashed the new OT_RCP and Zigbee Gateway examples onto the Thread Border Router board. In the serial monitor on boot I noticed a warning and want to make sure there is no issue with it. It boots and forms a zigbee network ok, however this is what I see in the serial monitor:

```c
I (5514) ESP_ZB_GATEWAY: Zigbee stack initialized
I (5524) ESP_ZB_GATEWAY: Device started up in  factory-reset mode
I (5524) ESP_ZB_GATEWAY: Start network formation
W(5704) OPENTHREAD:[W] P-RadioSpinel-: Error processing result: NotImplemented
W(5704) OPENTHREAD:[W] P-RadioSpinel-: Error waiting response: NotImplemented
W (5724) ESP_ZB_GATEWAY: Network(0x0870) closed, devices joining not allowed.
I (5724) ESP_ZB_GATEWAY: Formed network successfully (Extended PAN ID: 74:4d:bd:ff:fe:60:fc:d3, PAN ID: 0x0870, Channel:13, Short Address: 0x0000)
I (6294) ESP_ZB_GATEWAY: Network(0x0870) is open for 180 seconds
I (6294) ESP_ZB_GATEWAY: Network steering started
``` 

Do I need to be worried about these 2 lines?
W(5704) OPENTHREAD:[W] P-RadioSpinel-: Error processing result: NotImplemented
W(5704) OPENTHREAD:[W] P-RadioSpinel-: Error waiting response: NotImplemented

gytxxsy, 2024-02-07:
I think this is an issue caused by an incorrect RCP firmware. For ot_rcp, the `OPENTHREAD_NCP_VENDOR_HOOK` should be enabled(refer to the [zigbee gateway README.md](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_gateway/README.md)).

[image]

You can enable the `OPENTHREAD_NCP_VENDOR_HOOK` for ot_rcp and try agin.

djorr5, 2024-02-07:
You were correct. I had to add it manually when I wasn't using the correct branch, and I forget to set it on when using the correct branch. Thanks for your help

closed 2024-02-12 by djorr5