espressif/esp-zigbee-sdk #242: Provide examples of group binding

hopsor, 2024-02-06:
### Is your feature request related to a problem?

I haven't been able to find clear documentation about binding with groups. Having examples for this would be amazing.

### Describe the solution you'd like.

I'm interested in building a device capable of controlling different light groups using multiple switches. More specifically using the binding feature of zigbee2mqtt.

### Describe alternatives you've considered.

I tried flashing my c6 devkit board with the ha_on_off_light example to see, if by any chance, this worked out of the box. Unfortunately I wasn't able to bind it properly with my Zigbee bulbs using zigbee2mqtt so I imagine this require additional implementation in the code.

Hedda, 2024-02-20:
+1 binding of a Zigbee group is very useful for controlling Zigbee groups that contain multiple Zigbee lights, switches, and fans:

https://www.zigbee2mqtt.io/guide/usage/groups.html

Zigbee2MQTT has support for Zigbee groups. By using Zigbee groups you can control multiple devices simultaneously with one command. It also supports the binding of devices or a group, allowing devices or groups to directly control each other without the intervention of Zigbee2MQTT or any home automation software.

https://www.zigbee2mqtt.io/information/binding.html

Home Assistant's ZHA integration also supports native Zigbee on-device grouping of multiple Zigbee lights, switches, and fans that enable controlling all entities for those devices in those groups with one command, but note that at least two entities must be added to a Zigbee group inside the ZHA integration before a group entity is created. There is also support for native on-device Zigbee binding and unbinding (i.e. bind a remote to a lightbulb or group):.

https://www.home-assistant.io/integrations/zha#binding-and-unbinding

xieqinan, 2024-02-28:
@hopsor @Hedda,

Based on your description, it seems that the group binding feature shares similarities with the functionality provided by the group cluster. The group cluster allows multiple endpoints to join a specific group, enabling ZCL commands sent via groupcast to be received by all endpoints within that group.

Could you please provide further clarification and details regarding the group binding feature? This information will be valuable for our implementation efforts.

xieqinan, 2025-02-08:
Hi, @hopsor 

I think the the [group user guide](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_groups.html) of esp-zigbee-sdk may help you resolve this issue, could you please refer to it to implement your feature?

closed 2025-02-08 by chshu