espressif/esp-zigbee-sdk #245: Crash with custom cluster and custom attributes with esp_zb_zcl_report_attr_cmd_req

MetaImi, 2024-02-09:
### Question

I'm trying to add a custom cluster with custom attributes to support wind speed and direction. Cluster creation works, however when I try to report the value to home assistant, it crashes. 

Reporting code: 
`  esp_zb_zcl_status_t state_co2 = esp_zb_zcl_set_attribute_val(MW_ENDPOINT, MW_WIND_CLUSTER_ID, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, MW_WIND_SPEED_ATTR_ID, &windSpeed, false);
  
         esp_zb_zcl_report_attr_cmd_t report_attr_cmd;

          report_attr_cmd.zcl_basic_cmd.dst_addr_u.addr_short = 0; // coordinator
          report_attr_cmd.zcl_basic_cmd.dst_endpoint = MW_ENDPOINT;
          report_attr_cmd.zcl_basic_cmd.src_endpoint = MW_ENDPOINT;
          report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
          report_attr_cmd.clusterID = MW_WIND_CLUSTER_ID;
          report_attr_cmd.attributeID = MW_WIND_SPEED_ATTR_ID;
          report_attr_cmd.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
          esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd);`

Cluster creation:
```#define MW_ENDPOINT 1

#define MW_WIND_SPEED_ATTR_ID         0x0000
#define MW_WIND_DIRECTION_ATTR_ID     0x0001
#define MW_WIND_CLUSTER_ID            0xFFF2

    //Custom cluster
    const uint8_t attr_type = ESP_ZB_ZCL_ATTR_TYPE_U16;
    const uint8_t attr_access = ESP_ZB_ZCL_ATTR_MANUF_SPEC | ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING;

    esp_zb_attribute_list_t *windCluster = esp_zb_zcl_attr_list_create(MW_WIND_CLUSTER_ID);
    esp_zb_custom_cluster_add_custom_attr(windCluster, MW_WIND_SPEED_ATTR_ID, attr_type, attr_access, &windSpeed);
    esp_zb_custom_cluster_add_custom_attr(windCluster, MW_WIND_DIRECTION_ATTR_ID, attr_type, attr_access, &windDirection);```

MetaImi, 2024-02-09:
Somehow I managed to get more info without actually crashing and rebooting. 
I added an error check, and got this: 

`10:53:08.404 -> [0;31mE (36352) ESP_ZIGBEE_COMMAND: esp_zb_zcl_report_attr_cmd_req(220): This attribute: 0x1 is not existed!
10:53:08.404 -> [0m
10:53:08.404 -> ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x420002be
10:53:08.455 -> file: "/Users/metaimi/Documents/Arduino/Zigbee/Weather/MetaWeather.ino" line 276
10:53:08.455 -> func: void update_wind(void*)
10:53:08.455 -> expression: err`

So for some reason the attribute is not registered to the cluster, and the operation is not supported. Question is why is it not registered?

xieqinan, 2024-03-14:
@MetaImi ，

Could you show me the completed code for creating the endpoint list? Have the functions `esp_zb_ep_list_add_ep()` and `esp_zb_device_register()` been utilized to register the `windCluster` cluster?

xieqinan, 2024-03-18:
Hi, 
The esp-zigbee-sdk now includes an example demonstrating the usage of ZCL reporting, which is related with this issue. You can refer to the [temperature and thermostat example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor).

xieqinan, 2024-04-15:
@MetaImi ,

Do you have any updates for this issue?

closed 2024-07-26 by chshu