espressif/esp-zigbee-sdk #246: Random packets that aren't processed by OT_RCP Gateway example.

djorr5, 2024-02-09:
### Question

Just want to know what the following means and if I need to worry about it? I have bound and can report the state of a switch back to the gateway example being either "true" or "false" when I turn it on or off. However when I press the switch on the power point I get some random packets that are not processed. Can you tell me what they might relate to?

```c
I (121584) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x6), attribute(0x0), type(0x10), value(true)
I (121764) ZBOSS: zdo/af_rx.c:463   APS pkt 6 for ep 1 is not recognized - drop
I (121964) ZBOSS: zdo/af_rx.c:463   APS pkt 7 for ep 1 is not recognized - drop
I (122164) ZBOSS: zdo/af_rx.c:463   APS pkt 8 for ep 1 is not recognized - drop
I (122364) ZBOSS: zdo/af_rx.c:463   APS pkt 9 for ep 1 is not recognized - drop
I (123054) ZBOSS: zdo/af_rx.c:463   APS pkt 11 for ep 1 is not recognized - drop
I (123054) ESP_ZB_GATEWAY: Received report attribute callback
I (123054) ESP_ZB_GATEWAY: Received report attribute response message
I (123064) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x6), attribute(0x0), type(0x10), value(false)
I (123254) ZBOSS: zdo/af_rx.c:463   APS pkt 13 for ep 1 is not recognized - drop
I (123454) ZBOSS: zdo/af_rx.c:463   APS pkt 21 for ep 1 is not recognized - drop
I (123654) ZBOSS: zdo/af_rx.c:463   APS pkt 25 for ep 1 is not recognized - drop
I (123854) ZBOSS: zdo/af_rx.c:463   APS pkt 14 for ep 1 is not recognized - drop
I (124164) ZBOSS: zdo/af_rx.c:463   APS pkt 10 for ep 1 is not recognized - drop
I (124174) ESP_ZB_GATEWAY: Received report attribute callback
I (124174) ESP_ZB_GATEWAY: Received report attribute response message
I (124184) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x6), attribute(0x0), type(0x10), value(true)
I (124364) ZBOSS: zdo/af_rx.c:463   APS pkt 16 for ep 1 is not recognized - drop
I (124564) ZBOSS: zdo/af_rx.c:463   APS pkt 18 for ep 1 is not recognized - drop
I (124774) ZBOSS: zdo/af_rx.c:463   APS pkt 20 for ep 1 is not recognized - drop
I (124974) ZBOSS: zdo/af_rx.c:463   APS pkt 2 for ep 1 is not recognized - drop
I (153234) ESP_ZB_GATEWAY: Received report attribute callback
I (153234) ESP_ZB_GATEWAY: Received report attribute response message
I (153244) ESP_ZB_GATEWAY: Read attribute response: status(0), cluster(0x6), attribute(0x0), type(0x10), value(true)
```

xieqinan, 2024-03-14:
Hello @djorr5 ,

Could you please attempt to call `esp_zb_zdo_simple_desc_req()` in response to the previous simple_desc request? In other words, `active_ep_cb()` triggers the first `esp_zb_zdo_simple_desc_req()`, and then the second `esp_zb_zdo_simple_desc_req()` is triggered upon receiving the simple_desc response.

xieqinan, 2024-04-15:
@djorr5 ,

Do you have some updates for this issue?

djorr5, 2024-04-15:
Hey, I never got around to testing this. I only needed to control 1 end
point, so I modified my code for that and haven't needed it since. I will
test this out and let you know.

syam-ari, 2024-04-16:
The same issue we also observed. In our case, we have multiple endpoints and each endpoint contains multiple clusters, and we fixed this issue by creating separate tasks and message queue to send an event to trigger the **esp_zb_zdo_simple_desc_req()** once get the response for previous **esp_zb_zdo_simple_desc_req()** call. These function calls trigger from separate tasks.

djorr5, 2024-04-16:
@syam-ari Appreciate the response. Are you in a position to share the code snippet?

closed 2024-05-21 by djorr5