espressif/esp-zigbee-sdk #247: Extend Zigbee gateway to include BLE gateway

totkeks, 2024-02-12:
### Is your feature request related to a problem?

I would like to build my own Zigbee and BLE gateway to control devices inside my home.

### Describe the solution you'd like.

Using a single device (not necessarily single SoC/chip) like the Espressif Thread Border Router used in [this example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway) to act as both Zigbee and BLE gateway at the same time.

Building on the example above, I imagine that the S3 could be capable of handling both protocols due to having two cores. Using the H2 RCP for Zigbee and the S3 for BLE.

Depending on the available resources the connection to the home network is then either done using the S3 Wifi or the additional ethernet HAT, depending on the feasibility of simultaneously using BLE and WiFi on the S3.

### Describe alternatives you've considered.

- Buying USB dongles for Zigbee and BLE and connecting those to an SBC like the RPI-4 or other similar ones.
- Using separate ESP (development) boards, either two or three, to achieve a similar result and allowing for two separate RCPs (when using three)

chshu, 2024-02-18:
@totkeks It's feasible to implement both BLE gateway and Zigbee gateway on the ESP Thread Border Router/Zigbee Gateway board:
* Wi-Fi, BLE and Zigbee software stacks all run on S3
* Zigbee RCP (ot_rcp) runs on H2

Wi-Fi and BLE packets share the S3 RF path with coexist feature enabled: https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s3/api-guides/coexist.html#supported-coexistence-scenario-for-esp32-s3, Zigbee packets are forwarded between S3 and H2 via Uart, and actually sent/received over H2 RF.

teleger, 2024-02-20:
when I use zigbee_gateway project.

managed_components/espressif_esp-zboss-lib/ CMakeLists.txt

    if (CONFIG_ZB_RADIO_SPINEL_UART)
        idf_component_get_property(openthread_lib openthread COMPONENT_LIB)
        target_link_libraries(${COMPONENT_LIB} INTERFACE zboss_port_lib $<TARGET_FILE:${openthread_lib}> zboss_port_lib)
    endif()

Why can't I find the openthread library, do I need to compile it beforehand?

closed 2024-03-08 by chshu