espressif/esp-zigbee-sdk #248: ESP_ZIGBEE_CORE  Logs

mundevx, 2024-02-16:
### Question

Hi, 

I got some errors that i have to handle, and displayed on ESP_ZIGBEE_CORE. As i got 'Endpoint(0x2) not found in the group table for group ID(0x13)' and i want to handle this and add group id 0x13 to endpoint 0x2.
How i can have access to ESP_ZIGBEE_CORE logs?

xieqinan, 2024-02-20:
@mundevx ,

The error log originates from the description in the specification.

[image]

One way to handle the relationship between groups and endpoints is to use the `esp_zb_zcl_groups_add_group_cmd_req()` function.

mundevx, 2024-02-21:
Ok,
I have four endpoints and i added four groups={1,2,3,4} using "esp_zb_zcl_groups_add_group_cmd_req()".
Now from gateway, when i add scene, then gateway send response on my device with group=10, scene=10.
Then i got same error "'Endpoint(0x1) not found in the group table for group ID(0x0A)".

I dont know what group id will be generated by gateway, so i cant pre add the unknown groups.

xieqinan, 2024-02-28:
Hello @mundevx ,

> Now from gateway, when i add scene, then gateway send response on my device with group=10, scene=10.
Then i got same error "'Endpoint(0x1) not found in the group table for group ID(0x0A)".

This behavior is strange; the gateway should respond with the related group ID and scene ID you've set. Could you please provide the usage of `esp_zb_zcl_groups_add_group_cmd_req()` and include the response log?

closed 2024-07-26 by chshu