espressif/esp-zigbee-sdk #251: ColorControl COLOR_MODE attribute not updated by color commands

qcrist, 2024-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### esp-zigbee-lib version.

1.1.1

### esp-zboss-lib version.

1.1.1

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

If a light device is in ColorMode=0x1(XY) and receives a MoveToColorTemperature request, the ColorMode attribute should be updated to 0x2.

### What is the actual behavior?

The device crashes

### Steps to reproduce.

1) Create a light cluster which supports XY and CT
```
    uint16_t color_capabilities = 0x18;
    uint16_t zero = 0;
    uint16_t mireds = 500;
    uint16_t mireds_max = ESP_ZB_ZCL_COLOR_CONTROL_COLOR_TEMP_PHYSICAL_MAX_MIREDS_DEFAULT_VALUE;
    uint16_t mireds_min = ESP_ZB_ZCL_COLOR_CONTROL_COLOR_TEMP_PHYSICAL_MIN_MIREDS_DEFAULT_VALUE;
    uint8_t color_mode = 1;

    esp_zb_attribute_list_t* esp_zb_color_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL);
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_CAPABILITIES_ID, &color_capabilities));
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_X_ID, &zero));
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_Y_ID, &zero));
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_TEMPERATURE_ID, &mireds));
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_MODE_ID, &color_mode));
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_OPTIONS_ID, &zero));
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_TEMP_PHYSICAL_MIN_MIREDS_ID, &mireds_min));
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_TEMP_PHYSICAL_MAX_MIREDS_ID, &mireds_max));
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_COLOR_MODE_ID, &color_mode));

```
2) send a MoveToColorTemperature request
3) device crashes, as it is in XY color mode

### More Information.

Example full source
```
#include <esp_check.h>
#include <driver/gpio.h>
#include "nvs_flash.h"

#include "ha/esp_zigbee_ha_standard.h"

#if !defined CONFIG_ZB_ZCZR
#error Define ZB_ZCZR in idf.py menuconfig to compile light (Router) source code.
#endif

[... 153 lines cut ...]
    ESP_ERROR_CHECK(esp_zb_color_control_cluster_add_attr(esp_zb_color_cluster, ESP_ZB_ZCL_ATTR_COLOR_CONTROL_ENHANCED_COLOR_MODE_ID, &color_mode));

    esp_zb_cluster_list_t* c_list = esp_zb_zcl_cluster_list_create();
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_basic_cluster(c_list, esp_zb_basic_cluster_create(&basic_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(c_list, esp_zb_identify_cluster_create(&identity_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_groups_cluster(c_list, esp_zb_groups_cluster_create(&groups_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_scenes_cluster(c_list, esp_zb_scenes_cluster_create(&scenes_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_on_off_cluster(c_list, esp_zb_on_off_cluster_create(&on_off_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_level_cluster(c_list, esp_zb_level_cluster_create(&level_cfg), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_color_control_cluster(c_list, esp_zb_color_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));

    esp_zb_ep_list_t* esp_zb_ep_list = esp_zb_ep_list_create();
/* add created endpoint (cluster_list) to endpoint list */
    esp_zb_ep_list_add_ep(esp_zb_ep_list, c_list, 1, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    ESP_ERROR_CHECK(esp_zb_start(false));

    esp_zb_main_loop_iteration();
    return 0;
}

```

The opposite example with ColorMode set to 2, then a XY color change also crashes the device.

xieqinan, 2024-03-01:
@qcrist ,

I have attached an example demonstrating this issue, where the feature works correctly. Could you please download and test it?
[HA_color_dimmable_example.zip](https://github.com/espressif/esp-zigbee-sdk/files/14459969/HA_color_dimmable_example.zip)

qcrist, 2024-03-02:
Hmm, your example does not crash for me.

If I add the hue attribute to my previous example it also does not crash. 

Why does the hue attribute have to be defined for the device to properly handle the MoveToColorTemperature request?

xieqinan, 2024-03-04:
@qcrist,

It is confirmed that there is a bug in the color mode change process within the esp-zigbee-sdk. We will address this issue in the next released version. You can utilize the workaround temporarily for your project.

closed 2024-03-08 by espressif-bot

xieqinan, 2024-03-08:
Hello @qcrist 
This issue has been addressed in version 1.2.0. Could you please update and test it?

qcrist, 2024-03-08:
Appears to be working now, thanks!