espressif/esp-zigbee-sdk #252: Waterleak wit ESP32-H2

vedatbotuk, 2024-02-20:
### Question

I try to create a metering cluster for water leak detection on surface.
With following code created a cluster. This works. I can read status attribute.

```
    esp_zb_metering_cluster_cfg_t metering_cluster_cfg = {
        .metering_device_type = ESP_ZB_ZCL_METERING_WATER_METERING,
        .status = ESP_ZB_ZCL_METERING_ALARM_CHECK_METER,
    };
    esp_zb_attribute_list_t *esp_zb_metering_cluster = esp_zb_metering_cluster_create(&metering_cluster_cfg);
    esp_zb_cluster_list_add_metering_cluster(esp_zb_cluster_list, esp_zb_metering_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
```

If a waterleak detected -> I want to update attribute. Unfortunately it does not work. For that I use following code.
```
    uint8_t leak = ESP_ZB_ZCL_METERING_ALARM_LEAK_DETECT;
    esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(SENSOR_DEVICE_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_METERING, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_METERING_STATUS_DEFAULT_VALUE, &leak, false);
```

Can anyone help me how to update  status meter attributes?

xieqinan, 2024-02-20:
@vedatbotuk ,

> esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(SENSOR_DEVICE_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_METERING, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_METERING_STATUS_DEFAULT_VALUE, &leak, false);

suggestion:
```
esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(SENSOR_DEVICE_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_METERING, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ZB_ZCL_ATTR_METERING_CURRENT_SUMMATION_DELIVERED_ID, &leak, false);
```

Could you please print the value of `state`, which its meaning can be referred to in [esp_zb_zcl_status_t](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_common.h#L174)?

vedatbotuk, 2024-02-20:
It looks ok. So there is no error here.
```
            esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(SENSOR_DEVICE_ENDPOINT, ESP_ZB_ZCL_CLUSTER_ID_METERING, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_METERING_STATUS_DEFAULT_VALUE, &leak, false);
            /* Check for error */
            if(state != ESP_ZB_ZCL_STATUS_SUCCESS) {
                ESP_EARLY_LOGI(TAG, "Leak attribute failed!");
                return;
```

vedatbotuk, 2024-02-20:
I will test it. With my code it shows nothing.

vedatbotuk, 2024-02-20:
I just get
```ESP_ZB_ZCL_STATUS_SUCCESS               = 0x00U```

vedatbotuk, 2024-02-23:
With following snippet works.
```esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(endpoint, ESP_ZB_ZCL_CLUSTER_ID_METERING, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_METERING_STATUS_ID, &leak, false);```

closed 2024-02-23 by vedatbotuk