espressif/esp-zigbee-sdk #253: [REQUEST] "ESP Zigbee NCP serial proxy SDK" project similar to "ESP Thread Boarder Router SDK" based on same hardware

Hedda, 2024-02-20:
### Is your feature request related to a problem?

@lhespress (and @chshu) not sure if should post a request here or in official repo at https://github.com/espressif/esp-zigbee-sdk

Feature request for "ESP Zigbee NCP serial proxy SDK" project similar to "Gateway Example" (Zigbee gateway with RCP firmware):

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway

So please consider making and maintaining an "ESP Zigbee NCP serial proxy SDK" project design to be used with [zigpy-espzb](https://github.com/lhespress/zigpy-espzb)

That is, a software solution that would make the two hardware designs from the "[ESP Thread Boarder Router SDK](https://github.com/espressif/esp-thread-br)" project usable as a network-attached remote Zigbee Coordinator NCP adapter, and as such instead offer a UART-to-IP serial streaming tunneling proxy relay solution (i.e. a remote Serial Port over TCP/IP) that works the same way as [TubesZB’s Zigbee Gateways](https://github.com/tube0013/tube_gateways) and [ZigStar’s Zigbee Gateways](https://github.com/mercenaruss/zigstar_gateways) solutions, meaning that it would present Espressif Zigbee NCP with ZNSP (Zigbee NCP Serial Protocol) interface as a remote serial communication port that can be used with Home Assistant's ZHA integration (which depends on zigpy):

https://www.home-assistant.io/integrations/zha

That is, just offering a pass-through tunneling of the UART port using a "ser2net" type Serial-to-IP proxy solution for relay connection:

* ESP32-C6/ESP32-H2 with Zigbee NCP firmware <-> ESP32-S3 with Serial-to-IP Proxy Server <-> Home Assistant's ZHA integration.

Preferably announcing the proxy service on the network via Zeroconf for automatic network discovery which is supported inside ZHA:

https://www.home-assistant.io/integrations/zha#discovery-via-usb-or-zeroconf

Otherwise, the end-user has to manually configure IP address and TCP port for the socket so Zeroconf makes it very user-friendly:

https://www.home-assistant.io/integrations/zha#zigate-or-sonoff-zbbridge-devices

FYI, Zeroconf (Zero-configuration networking) in the ZHA integration in turn depends on the Zeroconf integration:

https://www.home-assistant.io/integrations/zeroconf/

PS: TubesZB Zigbee Gateways and ZigStar Zigbee Gateways solutions as just examples as there are many ESP32 based solutions for this:

* https://tubeszb.com/
  * https://github.com/tube0013/tube_gateways

* https://zig-star.com/projects/
  * https://github.com/mercenaruss/zigstar_gateways

* https://shop.codm.de/automation/zigbee/40/zigbee-cc2652p2-tcp-ethernet-coordinator

* https://gio-dot.github.io/Z-Bee-Duo/

* https://smlight.tech/
  * https://github.com/smlight-dev/

* https://github.com/cyijun/OpenZ3Gateway

* https://shop68536829.taobao.com/

chshu, 2024-02-21:
@Hedda Thanks for the suggestion, we are indeed working on a esp-zigbee-host example, which will run on a ESP Wi-Fi SoC and access the ESP NCP via UART. Although we still recommend our customer to use the Host+RCP mode if they use ESP Wi-Fi SoC as the host.

Generally, for the Zigbee Gateway solution, we will have two options for our customers:
- ESP Wi-Fi + Zigbee SoCs: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway, It's the Host + RCP (Radio Co-Processor) design, application and Zigbee Stack run on an ESP32 series Wi-Fi SoC (ESP32, ESP32-C, ESP32-S, etc), with an 802.15.4 SoC (ESP32-H2, ESP32-C6) loaded with Zigbee RCP, the two communicate via UART
- ESP Zigbee + 3rd party host
It's the Host + NCP (Network Co-Processor) design, Zigbee Stack runs on ESP 802.15.4 SoC, and user application runs on a 3rd party SoC. The host could access NCP via the ESP Zigbee serial protocol (https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/application.html#esp-ncp-guide). We will provide a esp-zigbee-host example as a reference for customer.

teleger, 2024-02-21:
> @Hedda Thanks for the suggestion, we are indeed working on a esp-zigbee-host example, which will run on a ESP Wi-Fi SoC and access the ESP NCP via UART. Although we still recommend our customer to [...]

when I use zigbee_gateway project.

name of address: https://github.com/espressif/esp-zigbee-sdk

zigbee_gateway
managed_components/espressif_esp-zboss-lib/ CMakeLists.txt

if (CONFIG_ZB_RADIO_SPINEL_UART)
idf_component_get_property(openthread_lib openthread COMPONENT_LIB)
target_link_libraries(
&lt;TARGET_FILE:${openthread_lib}> zboss_port_lib)
endif()
Why can't I find the openthread library, do I need to compile it beforehand?
thank you!

xieqinan, 2024-02-28:
@teleger ，

The openthread is located within the [esp-idf components](https://github.com/espressif/esp-idf/tree/master/components/openthread). Could you please double-check the path to your esp-idf installation?

Hedda, 2024-02-28:
@teleger that is off-topic for this specific request.

teleger, 2024-03-05:
> @teleger ，
> 
> The openthread is located within the [esp-idf components](https://github.com/espressif/esp-idf/tree/master/components/openthread). Could you please double-check the path to your esp-idf installation?

I make sure the openthread library exists in my installation path. But so far, my problem, has not been completely solved, I do not know how to do

MartinXBcn, 2024-07-17:
> @Hedda Thanks for the suggestion, we are indeed working on a esp-zigbee-host example, which will run on a ESP Wi-Fi SoC and access the ESP NCP via UART. Although we still recommend our customer to [...]

Can you give an update of the status of the development of "esp-zigbee-host example, which will run on a ESP Wi-Fi SoC and access the ESP NCP via UART"?
Question for my understanding: 
Why do you recommend the "RCP-solution" instead of the "NCP-solution"? 
I would have thought that in case of the "NCP-solution" the workload for the host-ESP32-S3 is less compared to the "RCP-solution", i.e. more processor-capacity available on the host-ESP32-S3 for other user-tasks. Am I wrong?

Hedda, 2024-07-18:
> Why do you recommend the "RCP-solution" instead of the "NCP-solution"?

That might no longer be the recommation as the project by @lhespress that would be first to make use of it currently only suppot NCP, see:

https://github.com/lhespress/zigpy-espzb

lhespress, 2024-07-22:
> Can you give an update of the status of the development of "esp-zigbee-host example, which will run on a ESP Wi-Fi SoC and access the ESP NCP via UART"?

We have provide a [esp-zigbee-host](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_host) example as a reference.

 > Why do you recommend the "RCP-solution" instead of the "NCP-solution"?

This is recommend when ESP Wi-Fi + Zigbee SoCs, of course, you also can use  "NCP-solution" on ESP Wi-Fi SoCs.

MartinXBcn, 2024-07-22:
> We have provide a [esp-zigbee-host](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_host) example as a reference.
>
>
> This is recommend when ESP Wi-Fi + Zigbee SoCs, of [...]

I have seen the example "esp-zigbee-host". 
What I am missing in the example - as I do not have direct access to the ESP32-H2-MINI-1 but it is connected to the ESP32-S3-WROOM-1 via UART/SPI - is to upload the firmware of ESP32-H2 via the ESP32-S3 like it is realized in the example "esp-zigbee-gateway", i.e. the built NCP image will be automatically packed into the esp-zigbee-host-firmware and uploaded to ESP32-S3 and from the ESP32-S3 to ESP32-H2.
I started to add the upload-feature of "esp-zigbee-gateway" to "esp-zigbee-host", upload works, but I was struggling with other problems. That is why I asked if there is already a ready-to-run-example. Or is there a separate example only showing the feature of uploading a firmware from ESP32-S3 to ESP32-H2?

chshu, 2024-07-26:
> Or is there a separate example only showing the feature of uploading a firmware from ESP32-S3 to ESP32-H2?

The ESP RCP update is a standalone feature supported by the [esp_rcp_update](https://components.espressif.com/components/espressif/esp_rcp_update) component, it used in several examples:
- [Basic Zigbee Gateway](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway)
- [Thread Border Router](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router)
- [Rainmaker Zigbee Gateway](https://github.com/espressif/esp-rainmaker/tree/master/examples/zigbee_gateway)

You may refer to these examples and integrate the feature to your project.

If you are using both ESP SoCs as Wi-Fi and Zigbee, the Gatway + RCP mode is recommended. The Host + NCP is only suggested when using ESP NCP with 3rd-party host SoC. The difference between RCP and NCP modes:
- RCP Mode: Zigbee Stack and application all run on host SoC, only 15.4 Mac layer frame transmit and receive actions are communicated between the two SoCs
- NCP mode: Zigbee stack runs on NCP, ZCL and Applications run on host, the protocols between Host and NCP is more complicate than RCP mode, more code needs to be implemented for each ZCL cluster behavior (commands/events/callbacks), all these are naturally supported if the application and stack run on the same SoC in RCP mode.

Hedda, 2024-10-16:
@lhespress Do you have any more thoughts on these name change suggestions? -> https://github.com/lhespress/zigpy-espzb/issues/11

Again, please see all related comments in your pull request for zigpy-cli here -> https://github.com/zigpy/zigpy-cli/pull/45