espressif/esp-zigbee-sdk #254: esp_zb_mfg_tool.py and changing manufacturer_code

RadekDymak, 2024-02-21:
### Question

Hi I create zigbee device based on examples of esp_zigbee_customized_devices. Device work fine wit Zigbee Home Assistant.
But I want to change "manufacturer_code" of zigbee device (ZHA shows value 4660).

I use esp_zb_mfg_tool.py with input in multi_devices_config.csv:
1,NULL,NULL,0x07FFF800,Nexelec,0x1249

It create 1.bin file. I write this file to zb_fct section (0xf5000) according file partitions.csv of my project file.
writing was complete:
[image]

But when I pair device to ZHA i show that manufacturer code is the same (without any change)
What is wrong. Is the any way how to make it work. Can I change anything insource code of mycode based of examples esp_zigbee_customized_devices?

xieqinan, 2024-02-28:
Hello @RadekDymak,

I believe the flash manufacturer code process is functioning correctly.

Could you please double-check the version of the ESP-Zigbee-SDK? It's important to use the latest version for this feature.

Could you also share the `Node Description Response` packet from the device to ZHA? This information is necessary to analyze the issue.

RadekDymak, 2024-02-28:
Hello @xieqinan 

Here is Node Description Response 

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=1613, server_mask=11264, maximum_outgoing_transfer_size=1613, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405",
        "0x040d"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Nexelec",
  "model": "Air Quality Sensor Nexelec",
  "class": "zigpy.device.Device"
}

I use esp-zigbee-sdk version 1.1.2

but I create bin file from esp_zb_mfg_tool.py (from version 1.0.9)
because version 1.1.2 give me error bellow:
[image]

I don't know what is wrong

RadekDymak, 2024-02-29:
Manufacturer and model was changed manually throgh my code : 
	esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster,ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, manufacturer_name);
	esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster,ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, model_identifier);

but manufacturer code is still same as in message above:  manufacturer_code=4660

I allready try to remove device from ZHA and rejoin. But still same problem.

> > because version 1.1.2 give me error bellow:
> 
> it might be related to the esp-idf version. Could you update esp-idf to the master branch and try again?

I update ESP-IDF to 5.2 (for ESP32-C6) but still same result

xieqinan, 2024-03-01:
@RadekDymak ,

Was the `manufacturer_code=4660` set by you previously?

> I update ESP-IDF to 5.2 (for ESP32-C6) but still same result

Please use the latest esp-idf branch or v5.1.3 to test this issue

RadekDymak, 2024-03-01:
> Was the `manufacturer_code=4660` set by you previously?
This was default value. I don't change it or set it.
I need to change from this value 4660 to 0x1249 (which i set to multi_devices_config.csv). But I don't know how.

> > I update ESP-IDF to 5.2 (for ESP32-C6) but still same result
> 
> Please use the latest esp-idf branch or v5.1.3 to test this issue

Ok I will try v5.1.3

RadekDymak, 2024-03-01:
> > I update ESP-IDF to 5.2 (for ESP32-C6) but still same result
> 
> Please use the latest esp-idf branch or v5.1.3 to test this issue

Same result with v5.1.3

xieqinan, 2024-03-04:
@RadekDymak 
The error stems from the changes made in the [script](https://github.com/espressif/esp-zigbee-sdk/blob/5f70b2f0466b03c4c05559d7d879a2edb3cba697/tools/mfg_tool/esp_zb_mfg_tool.py#L32) . I believe that using the master branch of the esp-idf will resolve the issue. It appears to be related to the esp-idf version, as the esp-zigbee-sdk library is compiled using the latest master branch of the esp-idf. This may lead to errors in other idf versions. However, using esp-zigbee-lib v1.0.9 with your current esp-idf version should still generate the correct manufacturer_code.

Regarding the issue, I will try to reproduce it and provide you with a response as soon as possible.

RadekDymak, 2024-03-04:
> @RadekDymak The error stems from the changes made in the [script](https://github.com/espressif/esp-zigbee-sdk/blob/5f70b2f0466b03c4c05559d7d879a2edb3cba697/tools/mfg_tool/esp_zb_mfg_tool.py#L32) . I [...]

I can create bin file using esp_zb_mfg_tool.py in esp-zigbee-lib v1.0.9. I must only comment lines 444 and 445 of the script (because using Windows?)
    if not os.path.exists(file_name):
        os.mknod(file_name)

But after flashing this bin, manufacturer code is same.

Ok please try to reproduce error, and please try to change manufacturer code shown in ZHA and let me know how to correctly change it.

 Thank you

RadekDymak, 2024-03-07:
Do you have some new information ?

kelin6, 2024-03-11:
@RadekDymak  Pease try to add follow code:

[image]

[image]

We will address the issue in the next release version.

RadekDymak, 2024-03-14:
> @RadekDymak Please try to add follow code:
> 
> ![image](https://private-user-images.githubusercontent.com/39639038/311680238-9e82b6c4-2413-4261-a265-1b66c4a794af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAzOTIwODksIm5iZiI6MTcxMDM5MTc4OSwicGF0aCI6Ii8zOTYzOTAzOC8zMTE2ODAyMzgtOWU4MmI2YzQtMjQxMy00MjYxLWEyNjUtMWI2NmM0YTc5NGFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE0VDA0NDk0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU2NjRlYzBiYjIyNjk0MDljNWYyMDU0YjY1YTIxZWU0OTRiNTFmMWIzNmRmOTkyMmE3ZWViOTA0OTNhNGY0ZGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tPUk6FLaGHgqd39VRhcwjcZdUXWVVaG_nSFGeyW7lHo)
> 
> ![image](https://private-user-images.githubusercontent.com/39639038/311680378-42d959b0-9955-4528-bbd7-7fa23e6b74a4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAzOTIwODksIm5iZiI6MTcxMDM5MTc4OSwicGF0aCI6Ii8zOTYzOTAzOC8zMTE2ODAzNzgtNDJkOTU5YjAtOTk1NS00NTI4LWJiZDctN2ZhMjNlNmI3NGE0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE0VDA0NDk0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThkZDYwNGFjYzUyMjNjOThlMzU2ZGU0ODA1ZTNkYjNlMjFmZGQyOTRlYThjZGMxNGQ0NjQzMTc4MzAwMDkyZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OBBC_UTj5WaGNF7mocYa2ebd3r5onHDIaIQMZoGEouc)
> 
> We will address the issue in the next release version.

Thank you a lot, it works. It will be fine, if you correct it in next release version. I olose this issue.

closed 2024-03-14 by RadekDymak