espressif/esp-zigbee-sdk #256: The periodic reporting system is not working correctly.

diazmanuel, 2024-02-21:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-588-gc1c843f5e2

### esp-zigbee-lib version.

1.1.0

### esp-zboss-lib version.

1.1.0

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

Hello, I am trying to set up the reports for a device using the functions added in the 1.0.3 zigbee sdk version ( #137). After creating the cluster structure, I use the mentioned functions to configure the default values for the reports and the reporting delta. The goal was to configure certain attributes so that they only report when changes occur.

### What is the actual behavior?

I am unable to set them to report periodically at long intervals, for example, every 15 hours. According to the documentation of the Zigbee 3.0 standard, the fields for maximum and minimum reporting intervals are of type uint16, indicating that it is possible to set times up to 18 hours (65536 seconds maximum).Currently, if I try to set the reports every 18 hours, they are generated randomly without maintaining the 18-hour interval or modifying their value by a range greater than that established in the delta.
I'm leaving a screenshot corresponding to the configuration of the reports from the Zigbee 3.0 standard documentation.

[image]

### Steps to reproduce.

1. Created a custom cluster and added a reportable attribute.
2. also set the device to low power consumption, using light sleep (since my project utilizes it, it might be related to the clock; additionally, I use an external clock as recommended).
3. After creating the endpoint, I configured the default reports using the functions added in version 1.0.3 of the SDK.
4. Check that the reports are not generated randomly, as is the case in my situation.

### More Information.

Below is an example of the code I'm using. What I'm trying to achieve is to set the default reports for a custom attribute within a custom cluster so that it is not necessary to call the configure report functions from a third party after binding. In other words, I wish that if the configure report function is not called, it takes the values that I set below.

```
    esp_zb_device_register(esp_zb_ep_list);

    esp_zb_zcl_attr_location_info_t filter;
    esp_zb_zcl_reporting_info_t *report;

    filter.attr_id=CUSTOM_ON_OFF_CLUSTER_ATTR_STATE_ID;
    filter.cluster_id=CUSTOM_ON_OFF_CLUSTER_ID;
    filter.endpoint_id=ZIGBEE_BOILER_ENDPOINT;
    filter.cluster_role=ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
    filter.manuf_code=ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC;
    report=esp_zb_zcl_find_reporting_info(filter);
    if(report!=NULL){
        report->u.send_info.def_max_interval=0xFFFE;
        report->u.send_info.def_min_interval=0x00;
        report->u.send_info.max_interval=0xFFFE;
        report->u.send_info.min_interval=0x00;
        report->u.send_info.delta.u32=1;
        if(esp_zb_zcl_update_reporting_info(report)!=ESP_OK) ESP_LOGE(TAG, "error conf update");
    }else ESP_LOGE(TAG, "error filter");
```

The problem is that the reports are generated without cause, meaning the reports are generated even if the maximum set interval has not been reached, and I verified that their value does not change, so I discard out the possibility that they are being generated because their value has altered.If I set short times, for example, 30 seconds, it works correctly

xieqinan, 2024-02-28:
@diazmanuel ，

Noted, I will double-check this issue according to your description.

closed 2024-03-08 by espressif-bot

xieqinan, 2024-03-08:
Hello @diazmanuel ,

This issue has been addressed in version 1.2.0. Could you please update and test it?

diazmanuel, 2024-03-08:
> Hello @diazmanuel ,
> 
> This issue has been addressed in version 1.2.0. Could you please update and test it?

Hello, thank you, we will try it right away. You could also check the issue #244 It's been a month and I still haven't had a response, maybe they couldn't locate the problem or they just missed it. thank you