espressif/esp-zigbee-sdk #257: Keep attribute report persistent across end device reboot.

djorr5, 2024-02-22:
### Question

General question, does anyone know a way to keep an end device reporting its attributes persistent across reboots? Using the Gateway example I can bind an end device cluster with an attribute reporting back to me every 30 seconds successfully. However if I power cycle the end device the reporting doesn't start back up. It still appears to be bound, as I can control it via ON_OFF command and I do get an attribute response that way. But the timed reporting fails. 
If I power cycle the gateway, the scheduled reports start back up as normal so sounds like the report config command needs to be sent back to the end device if it ever drops power?

xieqinan, 2024-02-28:
Hello @djorr5,

I believe that the reported data can persist in the ESP-Zigbee-SDK. The end device will store the reported data in the NVRAM and read it when the device reboots. You can utilize the [customized_device example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_customized_devices) to verify this feature.

To resolve this issue, you can call `esp_zb_zcl_find_reporting_info()` to retrieve the reporting information. When the device reboots, we can verify the reported information to determine the initialization reason for this issue.

djorr5, 2024-02-28:
Thanks for following up @xieqinan . I decided to test out another end device from a different manufacture and it appears to handle power cycles a lot better. It seems it was just the end devices implementation.

closed 2024-02-28 by djorr5