espressif/esp-zigbee-sdk #258: Custom Cluster write Callback

qcrist, 2024-02-22:
### Question

Is it currently possible to implement custom cluster write callbacks?

I see some chat about this in #22, but the recommended solution `esp_zb_add_custom_cluster_command_cb`, does not seem to exist anymore?

I see the `ZB_ZCL_GENERAL_INIT_WRITE_ATTR_RESP` macro in the zboss-api, but I have no idea what to pass for the _cmd_ptr parameter.

xieqinan, 2024-02-28:
@qcrist ,

The esp-zigbee-sdk v1.1.2 has already support the write callback for custom cluster.  You can refer to [zcl_command](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L2499)

qcrist, 2024-02-29:
Great, thanks!

closed 2024-02-29 by qcrist