espressif/esp-zigbee-sdk #259: Application Support Sub Layer API: meaning of tx_time field in esp_zb_apsde_data_confirm_s struct

jackDonati, 2024-02-22:
### Question

Can anyone please explain the functionality of the [tx_time ](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/aps/esp_zigbee_aps.h#L58) field in esp_zb_apsde_data_confirm_s struct form esp_zigbee_aps.h ? 

The documentation only specify that it is a reserved field, but it's not clear if such timestamp refers to the transmission time of the acknowledgement, expressed in the clock domain of the destination node addressed by the data request which has been confirmed by the ack, or otherwise it refers to the actual TX time of the data request whose data confirm has been received and handled by the data confirm callback, expressed in the local clock.

I need to understand this aspect since I'm trying to implement a clock synchronization protocol for wireless nodes based on MAC timestamping. It seems that the RX time of a data indication is currently available in the indication handler callback, but it is not clear if the TX timestamp of a data request can be retrived in some way.

Thank you in advance.

xieqinan, 2024-02-28:
Hello @jackDonati,

The detailed description of the TxTime of APSDE-confirm is introduced in section 2.2.4.1.2 of the Zigbee spec:

[image]

From the Zigbee stack, it is reserved and always zero. If it is necessary for you, we can provide a fix for the TxTime.

jackDonati, 2024-02-28:
> Hello @jackDonati,
>
> The detailed description of the TxTime of APSDE-confirm is introduced in section 2.2.4.1.2 of the Zigbee spec: [...]

Hi @xieqinan. It would be very usefull for users like me who are trying to implement custom high-precision clock synchronization protocols to have read-access to the TxTime value of a APSDE-request in the subsequent APSDE-confirm callback.
I still don't understand if the [RxTime](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/aps/esp_zigbee_aps.h#L81) value of APSDE-indication (section 2.2.4.1.3 of the Zigbee spec) is already read-accessible from the user code in the destination device, or instead is always zero like the TxTime in APSDE-confirm (but eventually I can check by myself).

It would be great if you can provide this fix for the TxTime (and for the RxTime too if not already implemented, as already mentioned).
Let me know if it is necessary for me to report all this considerations in a new feature request.

Thank you very much for your support!

xieqinan, 2024-02-29:
@jackDonati ,

No problem, we will provide the new feature support ASAP. Thanks for your patience.

jackDonati, 2024-02-29:
> @jackDonati ,
> 
> No problem, we will provide the new feature support ASAP. Thanks for your patience.

Ok @xieqinan , thank you so much!

closed 2024-02-29 by jackDonati