espressif/esp-zigbee-sdk #260: Having the possibility to handle zcl command within the esp_zb_core_action_handler and Raw command for custom command within esp_zb_raw_command_handler in the same execution

Hatim27, 2024-02-22:
### Is your feature request related to a problem?

Yes, when I enable both the core action handler and the raw action handler simultaneously, the core action handler becomes useless. Therefore, we will need to manually manage even the commands of the implemented clusters commands.

### Describe the solution you'd like.

I'm interested in having the capability to utilize both core actions for ZCL standard commands and raw actions for custom commands.

### Additional context.

I'm transitioning from Silicon Labs to Espressif for Zigbee development. In Simplicity Studio, we have the flexibility to handle both raw commands and ZCL commands in a similar manner, with the primary difference being the handler.

xieqinan, 2024-02-28:
@Hatim27 ，

> I'm interested in having the capability to utilize both core actions for ZCL standard commands and raw actions for custom commands.

Of course, they are not conflicting handlers. I can introduce the basic flow for the two commands. When a ZCL command is indicated by the APS layer, the stack will check if the `raw_command_handler()` is registered first. If it is, the `raw_command_handler()` will be triggered with the raw command and then callback to the application. If the `raw_command_handler()` returns false, the stack will parse the ZCL command header and deliver it to the `zb_core_action_handler()` using the parsed result. So they can be executed in a sequence.

Hatim27, 2024-02-29:
Thanks a lot it's working correctly

closed 2024-02-29 by Hatim27