espressif/esp-zigbee-sdk #263: Deep Sleep Sample not joining new Network on C6

susch19, 2024-02-25:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-166-ge0944287f3, v5.3-dev-2087-g5f42493573

### esp-zigbee-lib version.

1.1.0, 1.1.2

### esp-zboss-lib version.

1.1.0, 1.1.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

The C6 connects to the network, the same way it works in light sleep or any other sample.

### What is the actual behavior?

The Network steering is never successful. Doesn't matter if i try to connect it to another C6 coordinator, a test environment with a CC2531 as a coordinator or a productive environment with a SONOFF Zigbee 3.0 USB Dongle Plus as a coordinator.
If i Flash the Light Sleep Sample, let it connect and then reflash Deep Sleep, it connects, bounds and can receive the on off toggle command.

### Steps to reproduce.

1. Build and Flash the HA_on_off_switch from the main ESP-IDF Repo
2. Build and Flash the Deep Sleep Example
3. Network steering will never succeed

### More Information.

Log from the Coordinator:
```
I (445) main_task: Returned from app_main()
I (455) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (455) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
I (455) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (465) ESP_ZB_ON_OFF_SWITCH: Start network formation
I (465) ESP_ZB_ON_OFF_SWITCH: Formed network successfully (Extended PAN ID: 40:4c:ca:ff:fe:47:e6:54, PAN ID: 0x8dba, Channel:12)
I (945) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (945) ESP_ZB_ON_OFF_SWITCH: Network steering started
I (183415) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
```

Log from the C6 deep sleep device:
```
I (459) main_task: Returned from app_main()
I (469) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (469) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized
I (469) ESP_ZB_DEEP_SLEEP: Device started up in  factory-reset mode
I (479) ESP_ZB_DEEP_SLEEP: Start network steering
I (629) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (1809) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (2979) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (4159) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (5329) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (6509) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (7679) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (8859) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (10029) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (11209) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (12379) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (13559) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (14729) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (15909) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (17079) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (18259) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (19429) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (20609) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
I (21779) ESP_ZB_DEEP_SLEEP: Network steering was not successful (status: -1)
```

Logs when connecting via other example first from Coordinator:
```
I (455) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (455) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
I (465) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (465) ESP_ZB_ON_OFF_SWITCH: Start network formation
I (475) ESP_ZB_ON_OFF_SWITCH: Formed network successfully (Extended PAN ID: 40:4c:ca:ff:fe:47:e6:54, PAN ID: 0x8dba, Channel:12)
I (945) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (945) ESP_ZB_ON_OFF_SWITCH: Network steering started
I (3625) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (4115) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (4765) ESP_ZB_ON_OFF_SWITCH: New device commissioned or rejoined (short: 0x7b3e)
I (5065) ESP_ZB_ON_OFF_SWITCH: Found light
I (5065) ESP_ZB_ON_OFF_SWITCH: Try to bind On/Off
I (5065) ESP_ZB_ON_OFF_SWITCH: Bound successfully!
I (5065) ESP_ZB_ON_OFF_SWITCH: The light originating from address(0x7b3e) on endpoint(10)
I (5715) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (6215) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (13945) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command
I (18135) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command
I (48755) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (48765) ESP_ZB_ON_OFF_SWITCH: New device commissioned or rejoined (short: 0x7b3e)
I (49485) ESP_ZB_ON_OFF_SWITCH: Found light
I (49485) ESP_ZB_ON_OFF_SWITCH: Try to bind On/Off
I (49485) ESP_ZB_ON_OFF_SWITCH: Bound successfully!
I (49485) ESP_ZB_ON_OFF_SWITCH: The light originating from address(0x7b3e) on endpoint(1)
I (53345) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command
```

Logs when connecting via other example first from C6 End Device:
```
I (409) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
I (449) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
I (459) main_task: Returned from app_main()
I (469) ESP_ZB_DEEP_SLEEP: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (469) ESP_ZB_DEEP_SLEEP: Zigbee stack initialized
I (3299) ESP_ZB_DEEP_SLEEP: Device started up in non factory-reset mode
I (3299) ESP_ZB_DEEP_SLEEP: Device rebooted
I (12379) ESP_ZB_DEEP_SLEEP: Received message: endpoint(1), cluster(0x6), attribute(0x0), data size(1)
I (12379) ESP_ZB_DEEP_SLEEP: Light sets to On
```

xieqinan, 2024-02-28:
@susch19，

From the log provided by you, I believe that the non-joining behavior may be triggered by the log `I (183415) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK`. If you use the latest ESP-Zigbee-SDK example, it will print `Network (0xxxxx) closed, devices joining not allowed.`

susch19, 2024-02-28:
@xieqinan 
What you describe would be a problem on the coordinator side. But that is not the case with what I mention here.
The ESP32 can join any network (Zigbee2Mqtt with 2 different coordinators testet and also ESP32-C6 as coordinator), if I don't use the deep sleep example. As soon as I flash the deep sleep sample, the ESP32 never joins any open network, no matter how long I wait for, no matter how often I reboot the ESP32 Coordinator, so the network gets opened again. 
So for me it must have something to do with the deep sleep sample.

And as I said, it is only this one, all other samples like light sleep works as expected and can join the network, can be interviewed and configured. In the case of the light sleep sample, even with sleeping times in between. 
That's also what I find confusing, since the deep sleep sample is completely awake for the joining procedure and therefore should have even less Issues than the light sleep sample. In my understanding, the deep sleep shouldn't have any influence on how zigbee behaves

kelin6, 2024-03-04:
@susch19  We are based on the following environment：
IDF: v5.2-166-ge0944287f3 （）
ESP-Zigbee-SDK: 1.1.2
Coordinator: [esp_zigbee_HA_sample/HA_on_off_switch](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch)
ESP32C6 Deep_sleep: [esp_zigbee_sleep/deep_sleep]( https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/deep_sleep)

ESP32C6 Deep_sleep log: 
[image]
[image]

It works correctly, and I observed that your coordinator is operating on channel 12, while the deep sleep example is working on channel 13. Is there a problem?

[image]

So you can change the line [channel mask](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/deep_sleep/main/esp_zb_sleepy_end_device.h#L22)  to `#define ESP_ZB_PRIMARY_CHANNEL_MASK     ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK  /* Zigbee primary channel mask use in the example */` and try again.

susch19, 2024-03-04:
@kelin6 
You're right, when I use ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK it joins without problems, that's why light sleep always worked.
So I guess the Problem was, that I started with the Sample from Version 5.2 where this code was still present
[#define ESP_ZB_PRIMARY_CHANNEL_MASK     ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK  /* Zigbee primary channel mask use in the example */](https://github.com/espressif/esp-idf/blob/e0944287f33b348f9ccb7026bd32517d0f6b3e56/examples/zigbee/light_sample/HA_on_off_switch/main/esp_zb_switch.h#L44)

I've never erased the flash on the Coordinator, just flashed the new firmware. And since the Coordinator ignores the `esp_zb_set_primary_network_channel_set` when a network is already formed (Which it was in this case), I got stuck on Channel 12, because it was the one where those two ESP32 found each other first. 
It didn't even occur to me, to check for channels, but now I've learned, that the Channel can't be overridden, when a network is already formed and that the Coordinator supports automatic Channel Selection, which I also didn't now.

Thanks again, I'll close it now

closed 2024-03-04 by susch19