espressif/esp-zigbee-sdk #264: Start network steering inside non factory-reset mode

susch19, 2024-02-25:
### Question

In all examples there is this code to join a network:
``` c
 case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGI(TAG, "Device rebooted");
            }
```

but when the end device is already in the network, so the case "Device rebooted" is hit, then `Joined network successfully` will never be printed. So the esp deep sleep example won't ever enter deep sleep after the first time, since the deep sleep trigger is inside the successful join method. 
The comment of another issue seems to not have this problem: https://github.com/espressif/esp-zigbee-sdk/issues/183#issuecomment-1859530893

So is this a bug with the ESP32-C6 or was the code in the example modified? I'm just confused, because when we always have to call the commissioning of the network steering, than no samples should have the wrong code. But if we are not supposed to call the commissioning, when a network is already joined, then why do my log lines differ from the ones posted? Also the broken deep sleep sample seems to be an indication, that it shouldn't be that way

I only can test on a ESP32-C6 and tried with different combinations of zboss, zigbee lib and idf versions, all show the same behavior.
### IDF versions
v5.2-166-ge0944287f3, v5.3-dev-2087-g5f42493573

### esp-zigbee-lib version.
1.1.0, 1.1.2

### esp-zboss-lib version.
1.1.0, 1.1.2

xieqinan, 2024-02-28:
@susch19 ，

Could you please utilize the code below to meet your requirement? If it works, the example will be updated accordingly.
```
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                /* Start the one-shot timer */
                const int before_deep_sleep_time_sec = 5;
                ESP_LOGI(TAG, "Start one-shot timer for %ds to enter the deep sleep", before_deep_sleep_time_sec);
                ESP_ERROR_CHECK(esp_timer_start_once(s_oneshot_timer, before_deep_sleep_time_sec * 1000000));
                ESP_LOGI(TAG, "Device rebooted");
            }

        } else {
            /* commissioning failed */
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %d)", err_status);
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;
```

xieqinan, 2024-02-29:
@susch19 

Thanks for your explanation.

First, the code you provided is incorrect because the steering commissioning is meant to find a network for factory-reset devices only. It is not necessary for non-factory-reset devices. In other words, non-factory-reset devices do not need to trigger the steering process.

```
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            ESP_LOGI(TAG, "Start network steering");
            esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
        } else {
            /* commissioning failed */
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %d)", err_status);
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;
        
```

I'm still unclear about this issue. What behavior are you anticipating from the device?

- The device should perform the steering commission every time it reboots.
- The example needs to be adjusted; certain pre-configuration functions should not be placed after the device joins the network.

susch19, 2024-02-29:
@xieqinan 
Exactly. That's what I said, that I think that this code wouldn't be correct. I thought it just might clear up a few things.

I actually don't know which behavior I should expect, that's why this is marked as a question and not an issue. But this would mean, that the linked logs from https://github.com/espressif/esp-zigbee-sdk/issues/183#issuecomment-1859530893 are incorrect and the second network steering after wakeup shouldn't be there? 

If this is the case, than yes the deep sleep sample would be incorrect. But than my question would change into: Where do I put `zb_zdo_pim_set_long_poll_interval`, so that it works even after rebooting? Because it only ever worked, if I put it into the successful network steering signal and it was the initial join. If I put it somewhere else, it seems to be ignored. 
Or is it only possible to use the Poll Control cluster for having a poll intervall, which still works after restarting? Because this is not mentioned in the documentation, only that the set method is preferred, when Poll Control cluster is not used.

susch19, 2024-03-01:
@xieqinan 
Thank you, this matches with other startup procedures I found and answers the question.
So the final verdict is, that the deep sleep sample has to be fixed for deep sleep to always trigger. Also the log output of the deep sleep sample is wrong, since network steering is called after timer wakeup, even though it is not a factory reset. The wrong log output from the linked comment and the deep sleep example resulted in this confusion. 

Should I open an issue for this and link this question or should we change this question into an issue?

closed 2024-03-09 by susch19