espressif/esp-zigbee-sdk #265: End device and Coordinator with multiple End Points

HOSTEER, 2024-02-27:
### Question

How do I bind multiple End points between a End Device and a Coordinator ?
I have multiple switches on the End Device side and multiple lights on the Coordinator side. 
For each switches and lights I made an End Point.
I saw the function "esp_zb_zdo_find_on_off_light(&find_req, user_find_cb, NULL);" to bind a switch to a light, but it's only binding one End Point.

I declared my End Points this way:

for(int i=0;i<8;i++){
        esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT+i, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
        ESP_LOGI(TAG, "ep No. %d",i);
}

xieqinan, 2024-02-28:
Hello, @HOSTEER ,

> I saw the function "esp_zb_zdo_find_on_off_light(&find_req, user_find_cb, NULL);" to bind a switch to a light, but it's only binding one End Point.

The `esp_zb_zdo_find_on_off_light()` function is used to locate a specific light device rather than for binding purposes. You can achieve your goal by utilizing the `esp_zb_zdo_device_bind_req()` function.

HOSTEER, 2024-02-28:
Hello @xieqinan,

Thank you for your answer. 
I put the function from your advise in the `esp_zb_app_signal_handler()`.

[image]

I still have an issue, if the way I wrote the code is correct for the endpoint binding, should I turn on the light this way ? 
I'm trying to turn on one light each time I press the button.

[image]

Reports are missing when I try to turn on the remaining End points

End Device : 

[image]

Coordinator : 

[image]

HOSTEER, 2024-02-29:
I'm in fact starting from the [customized_device example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_customized_devices). 

I understood that binding is not mandatory to turn on and off the lights.

I'm expecting to see each lights from HA_ESP_LIGHT_ENDPOINT + 0 to HA_ESP_LIGHT_ENDPOINT + 7, turning on one after the other. But I'm getting this pattern of 1 and then 0, did I missunderstood how the toggle work ?

Client : 
[image]

Server : 
[image]

This the code I used in the button callback : 

```c
void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    static uint8_t endpoint_increment = 0;
    switch (button_func_pair->func) {
    case SWITCH_ONOFF_TOGGLE_CONTROL: {
        esp_zb_zcl_on_off_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = 0x0000;
        cmd_req.zcl_basic_cmd.dst_endpoint = HA_ESP_LIGHT_ENDPOINT + endpoint_increment % 8;
        cmd_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT + endpoint_increment % 8;        
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_ON_ID;
        ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command to address(0x%x) endpoint(%d)", 0x0000, cmd_req.zcl_basic_cmd.dst_endpoint);
        esp_zb_zcl_on_off_cmd_req(&cmd_req);
        endpoint_increment++;
    } break;
    default:
        break;
    }
}
``` 

And this the code I used in the attribute handler callback : 

```c
static esp_err_t zb_attribute_handler(const esp_zb_zcl_set_attr_value_message_t *message)
{
    esp_err_t ret = ESP_OK;
    bool light_state = 0;
    uint16_t identify_time = 0;
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)",
                        message->info.status);
    ESP_LOGI(TAG, "Received message: endpoint(%d), cluster(0x%x), attribute(0x%x), data size(%d)", message->info.dst_endpoint, message->info.cluster,
             message->attribute.id, message->attribute.data.size);
    for(int i=0;i<8;i++){
        if(message->info.dst_endpoint == (HA_ESP_LIGHT_ENDPOINT + i)){
            if (message->info.cluster == ESP_ZB_ZCL_CLUSTER_ID_ON_OFF) {
                if (message->attribute.id == ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID && message->attribute.data.type == ESP_ZB_ZCL_ATTR_TYPE_BOOL) {
                    light_state = message->attribute.data.value ? *(bool *)message->attribute.data.value : light_state;
                    ESP_LOGI(TAG, "Light sets to %d, endpoint %d", light_state, (HA_ESP_LIGHT_ENDPOINT + i));
                    //light_driver_set_power(light_state);
                }
            }
            if (message->info.cluster == ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY) {
                if (message->attribute.id == ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID && message->attribute.data.type == ESP_ZB_ZCL_ATTR_TYPE_U16) {
                    identify_time = *(uint16_t *)message->attribute.data.value;
                    ESP_LOGI(TAG, "Identify time set to %d, endpoint %d", identify_time, (HA_ESP_LIGHT_ENDPOINT + i));
                }
            }
        }
    }
    return ret;
}
```

HOSTEER, 2024-03-28:
Nerver guessed what was the issue, I'm not trying to light up multiple lights with multiple switches anymore. I hope an example will show up for people trying to do so

closed 2024-03-28 by HOSTEER