espressif/esp-zigbee-sdk #266: How address translation table size is determined?

Zeboo14, 2024-02-27:
### Question

Need to know the capacity in terms of number a devices (ZR or ZED) a coordinator can have on a network?

I get the message "I (244889) ZBOSS: common/zb_address.c:1014   Achtung! Address translation table overflow!" and I'm wondering if the size is a fix value in the code (may be different accross lib version) or if it depends on some other factor (stack size, heap, fc partition size etc..)

I may have up to 250 devices in a single network, for now, can attach only ~30 devices (routers) with lib v1.1.1

Is the capacity changes if device ar routers or end-devices?  For now, I have only routers.

xieqinan, 2024-02-28:
Hello @Zeboo14 ，

The device address table has a fixed size of 32 in v1.1.1, and we plan to optimize this fixed-size table issue in the future.

> Is the capacity changes if device ar routers or end-devices? For now, I have only routers.

No, it is not. The address information of end devices will also be added to the address table as long as the end device is near the coordinator. The ESP-Zigbee-SDK will support the MAC whitelist feature, preventing unauthorized MAC addresses from associating with the current device. This feature may help mitigate the capacity constraints of the address table.

Zeboo14, 2024-02-28:
Just to make sure (it is show stopper for us), we are talking about ZB_IEEE_ADDR_TABLE_SIZE macro defined as 32 in zb_config.h, correct ?

xieqinan, 2024-04-02:
@Zeboo14 ,

The esp-zigbee-sdk v1.2.2 now supports user configuration of the network size, which could potentially help address your problem. You can find more information about this feature in the API documentation here: [link to documentation](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_core.h#L342). I suggest trying to use this API to adjust the network size and see if it resolves your issue.

xieqinan, 2024-04-15:
@Zeboo14 ，

Do you have any updates regarding this issue?

closed 2024-07-26 by chshu