espressif/esp-zigbee-sdk #268: Update devices from other manufacturers over OTA

pieterjanbuntinx, 2024-03-01:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master/v5.1.3

### esp-zigbee-lib version.

1.1.2

### esp-zboss-lib version.

1.1.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

That I can embed a .zigbee/.ota file for a specific Zigbee device to the ESP firmware and update that device.

### What is the actual behavior?

The OTA seems to start but does not complete. Using Zigbee2MQTT I am able to update the client device.

### Steps to reproduce.

Use the esp_ota_server example and replace the ota_file.bin with the corresponding update.ota/update.zigbee file for your client device. Put the client device in pairing mode and make sure it joins the network of the ESP running the esp_ota_server example.

### More Information.

ESP OTA server log:

```
I (199448) ESP_OTA_SERVER: ZDO signal: ZDO Leave Indication (0x13), status: ESP_OK
I (199878) ESP_OTA_SERVER: ZDO signal: ZDO Leave Indication (0x13), status: ESP_OK
I (201608) ESP_OTA_SERVER: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (202088) ESP_OTA_SERVER: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (202238) ESP_OTA_SERVER: New device commissioned or rejoined (short: 0xd3fc)
I (202238) ESP_OTA_SERVER: OTA Server notify
I (202488) ESP_OTA_SERVER: OTA upgrade server query image
I (202488) ESP_OTA_SERVER: OTA client address: 0xd3fc
I (202488) ESP_OTA_SERVER: OTA version: 0x1, image type: 0xfffe, manufacturer code: fffe, 
I (202498) ESP_OTA_SERVER: OTA table index: 0x0
I (202758) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0xd3fc: progress [8/390864]
I (202758) ESP_OTA_SERVER: OTA client address: 0xd3fc
I (202758) ESP_OTA_SERVER: OTA version: 0x3, image type: 0xfffe, server status: 0
I (202958) ESP_OTA_SERVER: OTA client address: 0xd3fc
I (202958) ESP_OTA_SERVER: OTA version: 0x3, image type: 0xfffe, server status: 2
I (202958) ESP_OTA_SERVER: OTA upgrade time: 0x1234a
I (203118) ESP_OTA_SERVER: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (203218) ESP_OTA_SERVER: Network(0x8cf0) is open for 180 seconds
I (242258) ESP_OTA_SERVER: OTA upgrade server query image
I (242258) ESP_OTA_SERVER: OTA client address: 0xd3fc
I (242258) ESP_OTA_SERVER: OTA version: 0x1, image type: 0xfffe, manufacturer code: fffe, 
I (242268) ESP_OTA_SERVER: OTA table index: 0x0
I (242468) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0xd3fc: progress [8/390864]
I (242468) ESP_OTA_SERVER: OTA client address: 0xd3fc
I (242468) ESP_OTA_SERVER: OTA version: 0x3, image type: 0xfffe, server status: 0
I (242508) ESP_OTA_SERVER: OTA client address: 0xd3fc
I (242508) ESP_OTA_SERVER: OTA version: 0x3, image type: 0xfffe, server status: 2
I (242508) ESP_OTA_SERVER: OTA upgrade time: 0x1234a
I (302828) ESP_OTA_SERVER: OTA upgrade server query image
I (302828) ESP_OTA_SERVER: OTA client address: 0xd3fc
I (302828) ESP_OTA_SERVER: OTA version: 0x1, image type: 0xfffe, manufacturer code: fffe, 
[... 96 lines cut ...]
D: Received new Zigbee event: 0x02
D: Function: zb_trans_get_next_packet
D: Received new Zigbee event: 0x02
D: Function: zb_trans_get_next_packet
I: Mark OTA image as downloaded.
D: Function: zb_trans_transmit, channel: 16
D: Function: zb_trans_buffer_free
D: Received new Zigbee event: 0x01
I: Current address: f4ce3627a39e9407

I: Current address: f4ce3627a39e9407

D: Function: zb_trans_transmit, channel: 16
D: Received new Zigbee event: 0x01
I: Current address: f4ce3627a39e9407

I: Current address: f4ce3627a39e9407

I: Current address: f4ce3627a39e9407

I: Mark OTA image as ready to be installed.
E: Unable to schedule the update
I: ABORT Zigbee DFU
I: Current address: f4ce3627a39e9407
```

pieterjanbuntinx, 2024-03-18:
Hi @xieqinan

Can you or somebody else please check on this?

xieqinan, 2024-03-19:
@pieterjanbuntinx ,

I will now investigate this issue, and I will inform you of any updates I find.

xieqinan, 2024-03-19:
Hello @pieterjanbuntinx ,

I just attempt to produce this issue, however the example works fine. my steps is as followed.
- Bulid ota_client and ota_server example
- Remove  `~/ota_server/main/ot_file.bin`
- move  `~/ota_client/build/esp_ota_client.bin` to `~/ota_server/main/ot_file.bin`
- configure flash size to 4M
- idf.py -p [PORT] erase-flash flash monitor

Could you provide more detailed instructions for reproducing this issue? This will be helpful for us in resolving the problem you're experiencing. Forthmore, the sniffer packet also is helpful.

> I (202758) ESP_OTA_SERVER: OTA version: 0x3, image type: 0xfffe, server status: 0
> I (202958) ESP_OTA_SERVER: OTA client address: 0xd3fc
> I (202958) ESP_OTA_SERVER: OTA version: 0x3, image type: 0xfffe, server status: 2
> I (202958) ESP_OTA_SERVER: OTA upgrade time: 0x1234a

From the log, it appears that the client halts the OTA upgrade process after the initial transmission.

pieterjanbuntinx, 2024-04-19:
Hi @xieqinan ,

I'm sorry for not getting back to you sooner. I have had some time off from work.
What you tried was sending an OTA update from an Espressif device to another Espressif device. Is this the same protocol that is generally used for Zigbee devices? Can you do the same but with an external device from another brand? This is what I have not been able to get working.

I've attached below my sniffer log (No. > 1477). As you said, the client halts the OTA by sending an OTA upgrade End request. 
[sniffer_ota_ESP.zip](https://github.com/espressif/esp-zigbee-sdk/files/15041427/sniffer_ota_ESP.zip)

I've also attached the sniffer log I have gotten from Zigbee2MQTT when using that to do an OTA.
[sniffer_ota_zigbee2mqtt.zip](https://github.com/espressif/esp-zigbee-sdk/files/15041650/sniffer_ota_zigbee2mqtt.zip)

pieterjanbuntinx, 2024-06-14:
I found my problem. The Zigbee header was already included with the binary generated by the NRF board. Deleting the header (the first 56 bytes) from the binary file makes the OTA work correctly. 

Between some amount of image block transfers I get a NLME status indication with status `0x09: ZB_NWK_COMMAND_STATUS_PARENT_LINK_FAILURE`. Sometimes it takes quite a long time for the OTA to resume after I get this. Please look at timestamps 261255 and 394295 below. Do you have any idea what could be the cause of this?

```
I (257255) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [14664/423860]
NLME Status Indication, status: 0x9
NLME Status Indication, status: 0x9
I (257675) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [14728/423860]
I (258065) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [14792/423860]
I (258475) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [14856/423860]
I (258855) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [14920/423860]
I (259255) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [14984/423860]
NLME Status Indication, status: 0x9
I (259665) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15048/423860]
I (260055) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15112/423860]
I (260455) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15176/423860]
I (260845) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15240/423860]
I (261255) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15304/423860]
NLME Status Indication, status: 0x9
NLME Status Indication, status: 0x9
NLME Status Indication, status: 0x9
NLME Status Indication, status: 0x9
W (289615) ESP_OTA_SERVER: Network(0xce77) closed, devices joining not allowed.
I (306685) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15368/423860]
I (307085) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15432/423860]
I (307475) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15496/423860]
I (307885) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15560/423860]
NLME Status Indication, status: 0x9
I (308365) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15624/423860]
I (308765) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15688/423860]
I (309165) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15752/423860]
I (309565) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15816/423860]
I (309955) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15880/423860]
I (310365) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [15944/423860]
I (310775) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16008/423860]
I (311185) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16072/423860]
I (311595) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16136/423860]
I (311985) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16200/423860]
I (312405) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16264/423860]
I (312805) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16328/423860]
I (313215) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16392/423860]
I (313625) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16456/423860]
I (314005) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16520/423860]
I (314405) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16584/423860]
I (314815) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16648/423860]
I (315305) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16712/423860]
I (315765) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16776/423860]
I (316175) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16840/423860]
I (316585) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16904/423860]
I (316975) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [16968/423860]
I (317385) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [17032/423860]
I (317775) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [17096/423860]
I (318175) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [17160/423860]
I (318575) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [17224/423860]
I (318975) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [17288/423860]
I (319365) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [17352/423860]
I (319765) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [17416/423860]
I (320155) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [17480/423860]
I (320555) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [17544/423860]
[... cut ...]
I (449185) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [22600/423860]
I (449595) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [22664/423860]
I (450005) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [22728/423860]
I (450405) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [22792/423860]
I (450855) ESP_OTA_SERVER: -- OTA Server transmits data from 0x0 to 0x66cf: progress [22856/423860]
```