espressif/esp-zigbee-sdk #269: Provide a documented way to specify an endpoint's device version.

nrw505, 2024-03-02:
### Is your feature request related to a problem?

Trying to implement a colour-control light which can be added to a Philips Hue network. The Hue Hub does not consider a device with Profile ID 0x0104 (Home Automation), Device ID 0x0102 (Color Dimmable Light), Device Version 0x0000 (the value set by esp-zigbee-sdk) to be a valid light.

The ZigBee Lighting & Occupancy Device Specification Version 1.0, Section 6 specifies that for each described device type,

> The application device version field of the corresponding simple descriptor SHALL be set to 0x1.

As far as I can tell, there is currently no documented way to set the device version of an endpoint descriptor (it is not an argument to `esp_zb_ep_list_add_ep`

### Describe the solution you'd like.

add an argument `device_version` to `esp_zb_ep_list_add_ep` (breaking API change)
or create a function `esp_zb_ep_list_add_ep_with_version` (or similar) which allows setting the endpoint's device version field.

### Describe alternatives you've considered.

Functions to modify an endpoint in an `esp_zb_ep_list_t *` after it has been registered with `esp_zb_ep_list_add_ep` would be an alternative, but this is not in line with the more declarative approach you've chosen for setting up a device.

e.g. my current workaround

```
esp_zb_af_simple_desc_1_1_t * find_ep_descriptor(esp_zb_ep_list_t *ep_list, uint8_t ep_id)
{
    if (ep_list == NULL) {
        return NULL;
    }

    if (ep_list->endpoint.ep_id == ep_id) {
        return ep_list->endpoint.simple_desc;
    }

    return find_ep_descriptor(ep_list->next, ep_id);
}
```

```
    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    esp_zb_ep_list_add_ep(esp_zb_ep_list, cluster_list, ZLO_COLOR_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_COLOR_DIMMABLE_LIGHT_DEVICE_ID);

    esp_zb_af_simple_desc_1_1_t *light_ep_desc = find_ep_descriptor(esp_zb_ep_list, ZLO_COLOR_LIGHT_ENDPOINT);
    if (light_ep_desc != NULL) {
        light_ep_desc->app_device_version = 1;
    } else {
        ESP_LOGW(TAG, "Could not find descriptor for EP %d", ZLO_COLOR_LIGHT_ENDPOINT);
    }
```

xieqinan, 2024-03-04:
@nrw505 ，

Thank you for your suggestion. It's true that the esp-zigbee-sdk currently lacks an API for setting the endpoint device version. We will take this into consideration in next released version.

xieqinan, 2024-03-08:
@nrw505 ,

The esp-zigbee-sdk v1.2.0 has already added support for setting the device version of the endpoint. You can refer to https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_endpoint.h#L35

closed 2024-04-01 by espressif-bot