espressif/esp-zigbee-sdk #271: Invalid (positive) RSSI readings for the neighbour devices

nomis, 2024-03-03:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-422-ga7fbf452fa

### esp-zigbee-lib version.

1.1.2

### esp-zboss-lib version.

1.1.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Valid RSSI readings for neighbours are provided.

This worked in 1.0.5.

### What is the actual behavior?

The RSSI reading for the parent neighbour is positive, which is not a valid value.

### Steps to reproduce.

https://github.com/nomis/candle-dribbler/blob/0.6.0/src/zigbee.cpp#L670

I'm using `zb_nwk_nbr_iterator_next()` to get the list of neighbours, and this used to work in 0.9.4 and 1.0.5:
```
2023-09-16 15:44:34.868004 I (360345) nutt.Device: Neighbours (12):
2023-09-16 15:44:34.878733 I (360345) nutt.Device: 00:12:4b:00:24:c5:95:a7/eb07 R 1 S LQI 0 RSSI 127
2023-09-16 15:44:34.884622 I (360345) nutt.Device: 00:12:4b:00:24:c5:a2:8e/ec29 R 3 S LQI 0 RSSI 127
2023-09-16 15:44:34.890105 I (360355) nutt.Device: 00:12:4b:00:26:b8:5e:e7/7992 R 2 S LQI 0 RSSI 127
2023-09-16 15:44:34.901599 I (360365) nutt.Device: 00:12:4b:00:26:b8:63:d5/7882 R 1 S LQI 32 RSSI -84
2023-09-16 15:44:34.906695 I (360365) nutt.Device: 00:12:4b:00:24:c8:8f:1c/5ce9 R 2 S LQI 0 RSSI 127
2023-09-16 15:44:34.912064 I (360375) nutt.Device: 00:12:4b:00:24:c5:9c:bc/8599 R 1 S LQI 0 RSSI 127
2023-09-16 15:44:34.923303 I (360385) nutt.Device: 00:12:4b:00:26:b8:3f:19/684c R 1 S LQI 0 RSSI 127
2023-09-16 15:44:34.928724 I (360395) nutt.Device: 00:12:4b:00:26:b7:56:0d/81b4 R 2 S LQI 0 RSSI 127
2023-09-16 15:44:34.934675 I (360395) nutt.Device: 40:4c:ca:ff:fe:42:70:ac/e1a6 R 1 P LQI 32 RSSI -62
2023-09-16 15:44:34.945442 I (360405) nutt.Device: 40:4c:ca:ff:fe:41:a4:e4/5317 R 2 S LQI 0 RSSI 127
2023-09-16 15:44:34.951039 I (360415) nutt.Device: e0:79:8d:ff:fe:b3:dc:82/0000 C 0 S LQI 0 RSSI 127
2023-09-16 15:44:34.958610 I (360425) nutt.Device: 40:4c:ca:ff:fe:42:30:78/aac6 E 2 C LQI 0 RSSI 127
```

Now in 1.1.2 it is returning an invalid positive value for the RSSI in the range 7..11, when this should always be negative:
```
2024-03-03 11:42:06.565105 I (2884745) nutt.Device: Neighbours (10):
2024-03-03 11:42:06.565176 I (2884745) nutt.Device: 00:12:4b:00:24:cb:f9:1a/860f R 1 P LQI 0 RSSI 127
2024-03-03 11:42:06.565192 I (2884745) nutt.Device: 00:12:4b:00:26:b8:3f:19/684c _ 0 _ LQI 0 RSSI 127
2024-03-03 11:42:06.587151 I (2884755) nutt.Device: 00:12:4b:00:24:c5:a2:8e/ec29 _ 0 _ LQI 0 RSSI 127
2024-03-03 11:42:06.587199 I (2884765) nutt.Device: 00:12:4b:00:24:cb:f9:03/97f6 _ 1 _ LQI 0 RSSI 127
2024-03-03 11:42:06.587214 I (2884765) nutt.Device: 00:12:4b:00:24:cb:f9:1e/0f43 _ 2 _ LQI 0 RSSI 127
2024-03-03 11:42:06.609020 I (2884775) nutt.Device: 00:12:4b:00:26:b8:63:d5/7882 R 0 P LQI 0 RSSI 127
2024-03-03 11:42:06.609061 I (2884785) nutt.Device: 00:12:4b:00:24:c5:95:a7/eb07 _ 0 _ LQI 0 RSSI 127
2024-03-03 11:42:06.609078 I (2884795) nutt.Device: 00:12:4b:00:26:b7:56:0d/81b4 _ 0 _ LQI 0 RSSI 127
2024-03-03 11:42:06.622412 I (2884795) nutt.Device: 00:12:4b:00:26:b8:5e:e7/7992 _ 2 _ LQI 0 RSSI 127
2024-03-03 11:42:06.622456 I (2884805) nutt.Device: e0:79:8d:ff:fe:b3:dc:82/0000 C 0 P LQI 255 RSSI 10 <---
```

chshu, 2024-03-04:
@nomis The issue is caused by a structure definition mismatch since [(commit)](https://github.com/espressif/esp-idf/commit/82478c87b70a0fa911b861036b4f175f20609386#diff-9df277e673901ca309449955191c6859f9b02d970ca1bccb7067d6cbcb47ae37). Could you try again with the IDF v5.1.3 release.

nomis, 2024-03-04:
No, because I want `CONFIG_ESP_COREDUMP_FLASH_NO_OVERWRITE` and fixes to the monitor command which are only on the v5.3 branch. (I tried 5f4249357372f209fdd57288265741aaba21a2b1 which is more recent but the device restarts/hangs several times a day.)

This implies that this library is not compatible with certain versions of the IDF. A version check won't work because 5.3 and 5.2 are both newer than 5.1.3. You'd need to add a check in the headers that `struct esp_ieee802154_frame_info_t` is the expected size and cause a compile error.

How am I supposed to tell which versions are compatible with the IDF that I'm using? The most recent release of this library is 4 days before that IDF version was released and there's nothing in the commit messages to indicate this change.

nomis, 2024-03-04:
If it keeps crashing on a newer v5.3 build I won't even be able to use `git bisect` because of this change in structure size which will make it incompatible.

Depending on which IDF that they're built with, the old library will ignore `bool process` while the new library will misinterpret the channel as `bool process`. Both libraries will misread the other values in the struct.

chshu, 2024-03-05:
@nomis The disadvantage of library is that it's built based with particular commit, the incompatible issue may happen when there is any mismatch change in other souce code. So we recommended using the release/v5.1 branch ([Setting up the Repositories](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#setting-up-the-repositories)) where the zigbee libraries are built from.

If you want to use v5.3, please fetch the latest master branch, we don't see any incompatible issue recently.

In the future release, we will add the corresponding IDF version that the libraries are built from.

nomis, 2024-03-09:
> The disadvantage of library is that it's built based with particular commit, the incompatible issue may happen when there is any mismatch change in other souce code.

No, this is a problem with the way changes have been made to the IDF in a way that will be both incompatible and hidden.

If an incompatible change is being made you should [version](https://developers.redhat.com/blog/2019/08/01/how-the-gnu-c-library-handles-backward-compatibility) or rename the functions so that it won't link.

chshu, 2024-03-10:
> No, this is a problem with the way changes have been made to the IDF in a way that will be both incompatible and hidden.
>
> If an incompatible change is being made you should [...]

Agree that we should avoid the incompatible change, and invode version control for any break changes. We are very close to a stable Zigbee SDK release, thanks for your suggestions and patience along the way.

xieqinan, 2024-05-07:
@BaWe155，

> Hello. I have the same, neigbors LQI always 255. When is the fix and new release?

What is the ESP-IDF version you are currently using? Please note that the ESP-IDF version 5.1.3 is required for the ESP-Zigbee-SDK.

BaWe155, 2024-05-08:
Yes I use 5.1.3
esp-zigbee-lib version.
1.1.2
esp-zboss-lib version.
1.1.2

LQI in the neighbour tabels is always 255.
see (TZ-466) #167 this problem is back.

xieqinan, 2024-05-08:
@BaWe155 ，

You maybe need to use the esp-zigbee-lib and esp-zboss-lib v1.2.0 for a try, it is compiled based on esp-idf v5.1.3.