espressif/esp-zigbee-sdk #272: ESP32-H2 core panic in zigbee code after some time running

Caian, 2024-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-espidf @ 3.50102.0 (5.1.2) 

### esp-zigbee-lib version.

1.1.2

### esp-zboss-lib version.

1.1.2

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

To Zigbee to run indefinitely without issues, even after reboots.

### What is the actual behavior?

After some days and some reboots, the following errors start to appear after the device connects to the network:

```
␛[0;32mI (10442) ZBOSS: common/zb_address.c:1014   Achtung! Address translation table overflow!␛[0m
␛[0;32mI (10442) ZBOSS: common/zb_debug.c:64   Assertion failed /home/xieqinan/ESP/esp-zboss/components/zboss_stack/zboss/nwk/nwk_join.c:671␛[0m
```

```
I (121610) ZBOSS: zcl/zcl_main.c:357   zb_zcl_get_cluster_check_value: cluster_id 2820 cluster_role 1
I (121620) ZBOSS: zcl/zcl_main.c:382   zb_zcl_get_cluster_write_attr_hook: cluster_id 2820 cluster_role 1
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42061c2c  RA      : 0x42061c2c  SP      : 0x4081a4d0  GP      : 0x4080c790
TP      : 0x407e7584  T0      : 0x408089de  T1      : 0x408093ca  T2      : 0xffffffff
S0/FP   : 0x4081cafc  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00000100
A2      : 0x000000ff  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x00000000
A6      : 0x0000abe0  A7      : 0x00000000  S2      : 0x0000000d  S3      : 0x4081ca6c
S4      : 0x00000000  S5      : 0x0000001a  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000004  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00024201  T5      : 0xffffffff  T6      : 0xffffffff
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000003
MHARTID : 0x00000000

Stack memory:
4081a4d0: 0x00000000 0x4081c648 0x4081ca6c 0x4080fa42 0x0000004f 0x4081c648 0x4081ca6c 0x42019e64
4081a4f0: 0x00000000 0x00000000 0x00000098 0x4202bf88 0x4081a5cf 0x00000004 0x00000000 0x00000000
4081a510: 0x00000000 0x00000000 0x00000000 0x00000000 0x4081a558 0x4081ca6c 0x0000000a 0x4201a08e
4081a530: 0x0000000a 0x00000b04 0x00000000 0x00000000 0x0000000a 0x00000b04 0x00000001 0x42002792
4081a550: 0x00040102 0x00e0ab00 0x00000000 0x00000000 0x00000a00 0x00000000 0x00010b04 0x00000304
4081a570: 0x00000304 0x00000001 0x00000004 0x00000020 0x0000abe0 0x00000023 0x00000000 0x00000000
4081a590: 0x00000000 0x00024201 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081a5b0: 0x00000000 0x00000000 0x420a0000 0x420012b2 0x00001800 0x00000000 0x00000000 0x00040102
4081a5d0: 0x00e0ab00 0x00000000 0x00000000 0x0000003f 0x3fced480 0x3fced443 0x00000000 0x00000078
4081a5f0: 0x35453030 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081a610: 0x00009088 0x0ffa1770 0x013657e4 0x000a005a 0x000692d0 0x00000000 0x00000050 0x0000047e
4081a630: 0x00000027 0x45343130 0x31323135 0x30363036 0x30313732 0x00000000 0x00000000 0x00000000
4081a650: 0x00000400 0x00000000 0x00000002 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081a670: 0x00000000 0x00000000 0x420a8000 0x40808b86 0x00000000 0x00001388 0x00000001 0x00000000
4081a690: 0x00000000 0x00000000 0x00000000 0x408088d8 0x00000000 0x00000000 0x00000000 0x00000000
4081a6b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081a6d0: 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x40819cb0 0x00002f3e 0x4080d5e8 0x4080d5e8 0x4081a6dc
4081a6f0: 0x4080d5e0 0x00000018 0x4084e3e4 0x4084e3e4 0x4081a6dc 0x00000000 0x00000001 0x408188d8
4081a710: 0x6e69616d 0x63fea800 0x9aae5645 0x00a55372 0x00000000 0x4081a6d0 0x00000001 0x00000001
4081a730: 0x00000000 0x00000000 0x00000000 0x408172e4 0x4081734c 0x408173b4 0x00000000 0x00000000
4081a750: 0x00000001 0x00000000 0x00000000 0x00000000 0x4208f730 0x00000000 0x00000000 0x00000000
4081a770: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081a790: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081a7b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081a7d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081a7f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081a810: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40000000
4081a830: 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081a850: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081a870: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081a890: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081a8b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
```

Sometimes the zboss assertion does not appear, it goes straight to crash.

### Steps to reproduce.

Connect to a zigbee network and wait.

### More Information.

The device is connected to a solar plant that cuts power to it every night.

To restore the device I need to erase its entire flash and re-flash it. So far I the zigbee factory reset routines did not work.

chshu, 2024-03-06:
@Caian Is it a coordinator? How many devices in the network?

Caian, 2024-03-06:
Hi @chshu, It's an Endpoint. The network has 9 other devices + coordinator

Caian, 2024-03-07:
I also have a fair number of attributes exposed on the endpoint (43 in 4 clusters), is this an issue?

Caian, 2024-03-08:
Also this is my partition table:

```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,        data, nvs,      0x9000,  0x6000,
phy_init,   data, phy,      0xf000,  0x1000,
factory,    app,  factory,  0x10000, 900K,
zb_storage, data, fat,      0xf1000, 16K,
zb_fct,     data, fat,      0xf5000, 1K,
```

xieqinan, 2024-03-08:
@Caian ,

It appears to be an issue related to insufficient space for the Zigbee address table. We will address the issue of the address table not having enough space in the next release. Before that, I would appreciate it if you could provide details about your test environment, including the network devices, the role of esp32h2 in the network(coordinator, router or end_device), and the steps to reproduce the issue in detail. This information will be helpful for us to identify the root cause of the crashing.

> I also have a fair number of attributes exposed on the endpoint (43 in 4 clusters), and most of them are set for reporting by the coordinator, is this an issue?

It is not an issue.

Caian, 2024-03-10:
> @Caian ,
>
> It appears to be an issue related to insufficient space for the Zigbee address table. We will address the issue of the address table not having enough space in the next release. Before [...]

I'm using the H2 as an endpoint device. It is the only endpoint in the entire network. I have 11 devices in my network other than H2, all of them are Tuya smart switches with router capabilities. The coordinator is a Sonoff ZBDongle-E running Zigbee2Mqtt.

The H2 is part of the ESP Thread Board Router / Zigbee Gateway Board development board.

Caian, 2024-05-01:
The issue appear to be fixed after the latest sdk update.

closed 2024-05-01 by Caian