espressif/esp-zigbee-sdk #273: WiFi-SoftAP with Zigbee coexistence not working

BogatovT, 2024-03-07:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5,2-release

### esp-zigbee-lib version.

1.1.2

### esp-zboss-lib version.

1.1.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Soft AP and Zigbee were expected to work in coexistence

### What is the actual behavior?

When the Zigbee task is active AP is not searching in other devices

### Steps to reproduce.

1. init wifi soft ap
2. `esp_wifi_set_ps(WIFI_PS_MIN_MODEM);`
3. `esp_coex_wifi_i154_enable();`
4. zigbee coordinator init and start

### More Information.

**If I just skip zigbee initialization in the code - AP works fine.**

Log:
```
I (652) coexist: coex firmware version: 77cd7f8
I (658) coexist: coexist rom version 5b8dcfa
I (903) wifi:wifi firmware version: cc1dd81
I (910) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N
I (1037) wifi:mode : softAP
I (1071) wifi:Set ps type: 1, coexist: 0
I (1076) wifi:11ax coex: WDEVAX_PTI0(0x55777555), WDEVAX_PTI1(0x00003377).
```
`esp_coex_version_get()` returns 2.0.0
`esp_coex_wifi_i154_enable()` returns ESP_OK
menuconfig: Zigbee configured to native radio type. Software controls Wifi coexistence checked.

kelin6, 2024-03-19:
@BogatovT Currently, we support WiFi station mode + Zigbee coexistence, while WiFi AP + Zigbee coexistence is not yet supported.

chxlaity, 2024-04-17:
ESP 板现在支持三个模块：蓝牙 （BT & BLE）、IEEE802.15.4 和 Wi-Fi。每种类型的电路板只有一个 2.4 GHz ISM 频段射频模块，由两个或三个模块共享。因此，当一个模块进行数据传输或接收时，一个模块无法接收或传输数据。在这种情况下，ESP32-C6 采用时分复用方法来管理数据包的接收和发送。

xieqinan, 2026-05-18:
> Is this still the case in v1.6.8?

Yes, it is actually a platform issue, so I think it still exists in both v1.0 and v2.0.

By the way, using a single ESP32-C6 as a gateway is not recommended. Since there is only one RF path in the ESP32-C6, Wi-Fi and Zigbee cannot receive data simultaneously, which significantly impacts performance. Therefore, a two-SoC solution is recommended.