espressif/esp-zigbee-sdk #274: Changing PTA for RF coexistence in Zigbee gateway example

djorr5, 2024-03-08:
### Question

I am not sure if I should ask it in here or in the OpenThread repository. So I'll start here. I am designing a board that will use H2 and S3 similar to the Thread Border Router board. To help with routing I need to change the 3 pins on the S3 that are set for PTA for RF coexistence. Currently these are set as GPIO4, 5 and 6. I've looked through the Gateway code but can't see where I could change these? I can change the UART and RESET and BOOT pins no problem. It is just the PTA ones.

gytxxsy, 2024-03-08:
You can change the pins of external coexistence via menuconfig: `ESP Thread Border Router Example → External coexist wire type and pin config` (The configurations depends on the `EXTERNAL_COEX_ENABLE` option).

djorr5, 2024-03-12:
Thanks @gytxxsy . I could only see that option on the RCP example which would only change the pins for the H2 chip. I couldn't see that option on the Gateway example meaning it must be coded somewhere else. I did a string search for "esp_external_coex_gpio_set_t" and it only pops up in the ot_rcp example so I assume again that it isn't configurable on the S3?

Also noticed in menuconfig on the esp_zigbee_gateway example that external_coex isn't enabled anyway. So this leads me to believe that it isn't being used. Am I correct in this and I can leave these pins as not needing to be connected?

chshu, 2024-03-12:
@djorr5 Oh, yes, you are right. The external coex feature is not enabled in Zigbee Gateway example yet, it's only enabled in esp-thread-br example.

We will update the gateway example. If you want to try out in advance, please refer to the code here:
https://github.com/espressif/esp-thread-br/blob/main/examples/basic_thread_border_router/main/esp_ot_br.c#L55

djorr5, 2024-03-12:
It seems to run just fine without it :D 

But are you saying that this should be implemented in the OT_RCP and ZIGBEE_Gateway examples? If it is important then I will need to include it my hardware design.

chshu, 2024-03-13:
As we documented here: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#rf-external-coexistence, external coex is not a mandatory feature, especially when the application could ensure Wi-Fi and 802.15.4 will not operate on close frequence.

djorr5, 2024-03-13:
Understood.

closed 2024-03-13 by djorr5