espressif/esp-zigbee-sdk #275: Provide a non-looping version of `esp_zb_main_loop_iteration()`

nomis, 2024-03-08:
### Is your feature request related to a problem?

The latest version of the library documents a requirement to call `esp_zb_lock_acquire()` before calling library functions from a different task. I need to be able to schedule a function to happen at some time in the near future, while the zigbee main loop is idle.

I've been using `esp_zb_scheduler_alarm()` as a way to run functions (albeit without any useful arguments) from another task because they execute as part of `esp_zb_main_loop_iteration()`. It's also necessary for some cases even while running in that task because changing attributes from callbacks doesn't work (especially the callback for a change of the same attribute). It's safer to schedule something so that it happens outside of any callback.

Now I'm expected to use `esp_zb_lock_acquire()` before calling `esp_zb_scheduler_alarm()`? This is an inconvenient way to achieve synchronisation across tasks because it can block until one or more operations in the zigbee main loop complete. I can't use this from my task because I always need to be able to respond immediately to GPIO activity, which means I'd have to create an additional task just to acquire the lock. I can't have a light take longer to turn on/off just because something in the zboss library happens to take 100ms.

### Describe the solution you'd like.

Provide a non-looping version of `esp_zb_main_loop_iteration()` that can be called in a loop by user code (preferably with no built-in sleep/yield/delay).

That way I can schedule my own functions to be called whenever the zigbee main loop is idle, using simpler locking that doesn't block while zigbee functions are executing.

### Describe alternatives you've considered.

Make `esp_zb_scheduler_alarm()` and `esp_zb_scheduler_alarm_cancel()` be thread-safe by locking only the structures that maintain the scheduler alarm list, and not holding locks on those structures while executing the functions being scheduled. This would make the scheduler alarm functions safe to call from any task but without waiting on locks for an indefinite period of time.

nomis, 2024-03-08:
All the function is doing is calling `zboss_main_loop_iteration()` while holding the lock so I could just replicate that myself, but it wouldn't be safe if something changes in the future:
```
4202075e <esp_zb_main_loop_iteration>:
4202075e:       1141                    add     sp,sp,-16
42020760:       c606                    sw      ra,12(sp)
42020762:       a029                    j       4202076c <esp_zb_main_loop_iteration+0xe>
42020764:       7d02e0ef                jal     4204ef34 <zboss_main_loop_iteration>
42020768:       fe9ff0ef                jal     42020750 <esp_zb_lock_release>
4202076c:       557d                    li      a0,-1
4202076e:       fd5ff0ef                jal     42020742 <esp_zb_lock_acquire>
42020772:       f96d                    bnez    a0,42020764 <esp_zb_main_loop_iteration+0x6>
42020774:       bfd5                    j       42020768 <esp_zb_main_loop_iteration+0xa>
```

nomis, 2024-03-08:
It's not clear what `zboss_main_loop_iteration()` is doing that causes it to not be a busy loop. Whatever that is, it's holding the lock while doing it. This is a frustrating way to interact with the main loop and makes `esp_zb_lock_acquire()` unsuitable for all time-sensitive use cases.

nomis, 2024-03-09:
After 32 minutes uptime there have been 8412 executions of `zboss_main_loop_iteration()` with execution times:
* Minimum: 167us
* Average: 231ms
* Maximum: 1.86s

With a maximum observed runtime of nearly 2 seconds, the use of `esp_zb_lock_acquire()` is just not safe to use in any application.

chshu, 2024-03-10:
@nomis Thanks for making the analysis. I assuem the long execution time from your test result is caused by task scheduling, especially when there are some other higher priority tasks.

There is no block code in `zboss_main_loop_iteration()`, it traverses the task/timer queues, parses incoming packets, triggers application callbacks, and then enters idle mode to wait for platform events. The platform events are monitored by a `seclect` function, the select timeout is the minimal time in the Zigbee timer queue. The Zigbee task is under the idle mode for most of the time.

`esp_zb_lock_acquire()` is not execpted to be used in the time sensitive case, just like Mutex should not be used in interrupts, since the application can't guarantee how much time it has to wait for the lock.

Back to your use case, it seems you want to schedule a function to be executed in near future from the GPIO interrupt, how about using a IDF timer in the GPIO interrupt, and in the execution function, call `esp_zb_lock_acquire()` before accessing Zigbee APIs?

nomis, 2024-03-10:
> I assuem the long execution time from your test result is caused by task scheduling, especially when there are some other higher priority tasks.

No, I think the zboss function is waiting for about 1.5s somewhere. The maximum time actually went up several times:
```
2024-03-09 10:28:05.067651 I (21975) nutt.ZigbeeDevice: Stats: count=614 min=167us avg=34671us max=1661017us
2024-03-09 10:28:11.584626 I (28485) nutt.ZigbeeDevice: Stats: count=636 min=167us avg=43726us max=1853898us
2024-03-09 10:28:23.440322 I (40345) nutt.ZigbeeDevice: Stats: count=678 min=167us avg=57846us max=1858771us
2024-03-09 10:28:24.953538 I (41855) nutt.ZigbeeDevice: Stats: count=679 min=167us avg=60493us max=1858771us
2024-03-09 10:28:25.606903 I (42505) nutt.ZigbeeDevice: Stats: count=679 min=167us avg=60493us max=1858771us
2024-03-09 10:28:26.517149 I (43415) nutt.ZigbeeDevice: Stats: count=687 min=167us avg=61564us max=1858771us
2024-03-09 10:28:26.936863 I (43835) nutt.ZigbeeDevice: Stats: count=689 min=167us avg=62236us max=1858771us
2024-03-09 10:28:52.858677 I (69765) nutt.ZigbeeDevice: Stats: count=809 min=167us avg=84642us max=1858771us
2024-03-09 10:36:15.815972 I (512715) nutt.ZigbeeDevice: Stats: count=2607 min=167us avg=196337us max=1859985us
2024-03-09 10:57:11.236983 I (1768125) nutt.ZigbeeDevice: Stats: count=7660 min=167us avg=230724us max=1860130us
2024-03-09 11:00:09.713951 I (1946605) nutt.ZigbeeDevice: Stats: count=8412 min=167us avg=231197us max=1860130us
```

After 21 hours the average is very high at 484ms:
```
2024-03-10 09:20:23.608367 I (77747655) nutt.ZigbeeDevice: Stats: count=160366 min=126us avg=484786us max=1860156us
```

Here's the implementation used to time it: https://github.com/nomis/candle-dribbler/blob/0.7.0/src/zigbee.cpp#L162-L201

I don't have any tasks that run for that length of time:
```
2024-03-10 09:20:24.995121 I (77749035) nutt.UI: Tasks:
2024-03-10 09:20:24.995172 Name           	State	Prio	Stack	ID
# Waiting to read the UART
2024-03-10 09:20:24.995189 ui_uart        	X	1	3220	8
2024-03-10 09:20:24.995203 IDLE           	R	0	1304	3
# Runs `zboss_main_loop_iteration()` in a loop
2024-03-10 09:20:24.995217 zigbee_main    	B	5	4956	5
# Waiting on a semaphore from a timer to blink the LED
2024-03-10 09:20:24.995231 ui_main        	B	2	3460	7
2024-03-10 09:20:24.995247 esp_timer      	S	22	3696	1
# Waiting on a semaphore from a GPIO interrupt handler and a timer used to debounce
2024-03-10 09:20:24.995266 device_main    	B	19	1088	6
2024-03-10 09:20:25.017220 Tmr Svc        	B	1	1764	4
2024-03-10 09:20:25.017271 
2024-03-10 09:20:25.017288 I (77749045) nutt.UI: Stats:
2024-03-10 09:20:25.017304 Name           	Running		CPU%
2024-03-10 09:20:25.017323 ui_uart        	408		<1%
2024-03-10 09:20:25.017337 IDLE           	322730893	73%
2024-03-10 09:20:25.017350 zigbee_main    	96473324	21%
2024-03-10 09:20:25.017365 esp_timer      	4985779		1%
2024-03-10 09:20:25.017379 device_main    	4026516		<1%
2024-03-10 09:20:25.022443 Tmr Svc        	9		<1%
2024-03-10 09:20:25.022491 ui_main        	4033159		<1%
```

> There is no block code in `zboss_main_loop_iteration()`, it traverses the task/timer queues, parses incoming packets, triggers application callbacks, and then enters idle mode to wait for platform events. The platform events are monitored by a `select` function, the select timeout is the minimal time in the Zigbee timer queue. The Zigbee task is under the idle mode for most of the time.

The problem here is that while the Zigbee task is idle waiting on a timeout, it's holding a lock. For your documented way of interacting with the library this is unusable because the lock shouldn't be held while waiting for a timeout.

Even with my current strategy I get an increase in latency, because I have to wait for the timeout to expire before I can make changes to Zigbee attributes from another task.

As well as a one-shot wrapper for `zboss_main_loop_iteration()` what I would also need is a way to wake it up from another task/interrupt. Then I can alternate between calling `zboss_main_loop_iteration()` and the other `esp_zb_*()` functions, which means I can quickly wake up the `zboss_main_loop_iteration()` from other tasks to make changes to Zigbee attributes without waiting on a lock.

I'd expect this to be implemented as an atomic flag that causes `zboss_main_loop_iteration()` to clear the flag and return immediately instead of waiting for a timeout. It doesn't need to return if it's still busy, only when it would be idle. The wake-up function would set the flag and then wake up the `select`.

> `esp_zb_lock_acquire()` is not execpted to be used in the time sensitive case, just like Mutex should not be used in interrupts, since the application can't guarantee how much time it has to wait for the lock.

With a maximum time of 1.8s it should not be used anywhere.

> Back to your use case, it seems you want to schedule a function to be executed in near future from the GPIO interrupt, how about using a IDF timer in the GPIO interrupt, and in the execution function, call `esp_zb_lock_acquire()` before accessing Zigbee APIs?

I know [how to wake up and respond to interrupts](https://github.com/nomis/candle-dribbler/blob/0.7.0/src/thread.cpp). What you're not considering is that while I'd be waiting 1.8s in a task for `esp_zb_lock_acquire()` I'd be unable to respond to another interrupt (there are 4 light switches) or perform any debouncing. To use that function I'd need a separate task that has a list of functions to call and then waits to acquire the lock, but that's just the exact opposite execution time as the main zigbee task making such a task a waste of memory for another stack.

chshu, 2024-03-11:
> The problem here is that while the Zigbee task is idle waiting on a timeout, it's holding a lock

It's not true, the lock is released before idle. The whole flow looks like:

```
lock_acquire();

zboss_main_loop_iteration() {
    // process task/timer queues, trigger application callbacks

    lock_release();
    select(); // idle to wait for platform events
    lock_acquire();
}

lock_release();
```
Please refer to the Thread task mainloop: https://github.com/espressif/esp-idf/blob/master/components/openthread/src/esp_openthread.cpp#L161, the design is similar.

The 1.8s is the time between the start and end of `zboss_main_loop_iteration()`, but the lock is released during the idle mode.

nomis, 2024-03-11:
The only way to do this safely with low latency needs another task that can never run at the same time as the Zigbee main task so it's a waste of memory.

Could you make `esp_zb_scheduler_alarm()` and `esp_zb_scheduler_alarm_cancel()` be thread-safe without using the same lock (because it's still going to be held for long periods of time)?

chshu, 2024-03-11:
@nomis Ok, we will improve the thread safety of `esp_zb_scheduler_alarm()` and `esp_zb_scheduler_alarm_cancel()` by implementing atomic operations for processing the scheduler queue.

nomis, 2024-03-11:
Something to watch out for is to make sure that `esp_zb_scheduler_alarm()` can interrupt that `select()` call, otherwise there'll still be a latency problem if it's waiting for a long period of time.

chshu, 2024-03-12:
@nomis We just confirmed that the `esp_zb_scheduler_alarm()` and `esp_zb_scheduler_alarm_cancel()` are already thread safe, and an event will be written to wake up Zigbee task from `select()`. So the current SDK (1.2.0) should already meet your requirement.

We will update the docs to exclude `esp_zb_scheduler_alarm()` and `esp_zb_scheduler_alarm_cancel()` from API lock requirement.

closed 2024-03-12 by nomis