espressif/esp-zigbee-sdk #276: Invalid Data Type error when write attribute in On/Off Switch Configuration Cluster

grebenyuk51, 2024-03-09:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-89-g5c61c89308

### esp-zigbee-lib version.

1.2.0

### esp-zboss-lib version.

1.2.0

### Espressif SoC revision.

ESP32-C6 (chip revision: v0.0)

### What is the expected behavior?

When receive Write Attributes (either Switch Type or Switch Action) Request, send Write Attributes Response with status Success

### What is the actual behavior?

When receive Write Attributes (either Switch Type or Switch Action) Request, send Write Attributes Response with status Invalid Data Type

### Steps to reproduce.

I wrote some test code on end device that sends write request to ESP32-C6 router 

```
void button_single_click_cb(void *arg,void *usr_data)
{
    uint8_t switch_type = ESP_ZB_ZCL_ON_OFF_SWITCH_CONFIGURATION_SWITCH_TYPE_MOMENTARY;
    uint8_t switch_action = ESP_ZB_ZCL_ON_OFF_SWITCH_CONFIGURATION_SWITCH_ACTIONS_TOGGLE;
    esp_zb_zcl_attribute_t attrs[] = 
    {
        {
            .id = 0x0000,  //there is no enum in esp_zigbee_zcl_on_off_switch_config.h that represents attribute id
                    //0x0000 == ATTR_ID_SWITCH_TYPE
            .data.size = 1,
            .data.type = ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
            .data.value = &switch_type
        },
        // {
        //     .id = 0x0010,  //there is no enum in esp_zigbee_zcl_on_off_switch_config.h that represents attribute id
        //             //0x0010 == ATTR_ID_SWITCH_ACTIONS
        //     .data.size = 1,
        //     .data.type = ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
        //     .data.value = &switch_action
        // }
    };
    
    esp_zb_zcl_write_attr_cmd_t req; 
    req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    req.attr_field = attrs;
    req.attr_number = sizeof(attrs) / sizeof(esp_zb_zcl_attribute_t);
    req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF_SWITCH_CONFIG;
    req.zcl_basic_cmd.dst_addr_u.addr_short = 0xD05C;  //ESP32-C6 router's address
    req.zcl_basic_cmd.dst_endpoint = 1;
    req.zcl_basic_cmd.src_endpoint = 1;
    esp_zb_zcl_write_attr_cmd_req(&req);
}
```

On the router simple cluster/attributes initialization

```
esp_zb_zcl_on_off_switch_configuration_switch_type_t on_off_cfg_switchType = ESP_ZB_ZCL_ON_OFF_SWITCH_CONFIGURATION_SWITCH_TYPE_TOGGLE;
esp_zb_zcl_on_off_switch_configuration_switch_actions_t on_off_cfg_actionType = ESP_ZB_ZCL_ON_OFF_SWITCH_CONFIGURATION_SWITCH_ACTIONS_TYPE2;
esp_zb_attribute_list_t *on_off_config = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_ON_OFF_SWITCH_CONFIG);
esp_zb_on_off_switch_config_cluster_add_attr(on_off_config, 0x00, &on_off_cfg_switchType);
esp_zb_on_off_switch_config_cluster_add_attr(on_off_config, 0x10, &on_off_cfg_actionType);

```

In ZigBee Sniffer i can see as request, as response. And response contains Invalid Data Type error
[image]
[image]

In ZCL documentation i can see data type is enum8
[image]
[image]

But SDK responds with Invalid Data Type error

grebenyuk51, 2024-03-09:
by the way, there are no callbacks in action handler `esp_zb_core_action_handler_register(zb_action_handler);`
on/off cluster receives `ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID` but on/off switch config not

xieqinan, 2024-03-12:
@grebenyuk51 ,

I checked the source code of the esp-zigbee-sdk and found that the data type of `ESP_ZB_ZCL_ON_OFF_SWITCH_CONFIGURATION_SWITCH_TYPE_TOGGLE` and `ESP_ZB_ZCL_ON_OFF_SWITCH_CONFIGURATION_SWITCH_ACTIONS_TYPE2` is registered as `ESP_ZB_ZCL_ATTR_TYPE_8BITMAP`. We will fix it in the next version. There are two workarounds for this issue:

- Temporarily change the data type of `ESP_ZB_ZCL_ON_OFF_SWITCH_CONFIGURATION_SWITCH_TYPE_TOGGLE` to `ESP_ZB_ZCL_ATTR_TYPE_8BITMAP` in the application.

- Use the `esp_zb_cluster_add_attr()` function to add attributes to the cluster.

closed 2024-04-22 by grebenyuk51