espressif/esp-zigbee-sdk #277: Use of non implemented cluster ID on ESP32C6

Danixu, 2024-03-09:
### Question

Hello,

I am trying to create a Zigbee sensor for my swimming pool with the following clusters:
- Temperature
- PH
- Chlorine

The temperature cluster is easy because there are functions to do it, for example:

 ```
  int tempValue = 2300;
  int tempMin = 0;
  int tempMax = 32000;

  esp_zb_attribute_list_t *esp_zb_temperature_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT);
  esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &tempValue);
  esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MIN_VALUE_ID, &tempMin);
  esp_zb_temperature_meas_cluster_add_attr(esp_zb_temperature_cluster, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MAX_VALUE_ID, &tempMax);

  esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
  esp_zb_cluster_list_add_temperature_meas_cluster(esp_zb_cluster_list, esp_zb_temperature_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
```

The problem comes when I try to create the other two which doesn't have their corresponding functions. I have tried several ways like esp_zb_zcl_attr_list_create + esp_zb_cluster_add_attr, esp_zb_custom_cluster_add_custom_attr... but all has failed.

The PH cluster has it's own ID, which is 0x0409, but cannot be added as custom cluster because the ID doesn't matches, and I am not able to find another function to add it.
[image]

The code I have is:

```
  uint16_t phMin = 0;
  uint16_t phValue = 750;
  uint16_t phMax = 900;

  esp_zb_attribute_list_t *esp_zb_ph_cluster = esp_zb_zcl_attr_list_create(0x0409U);
  esp_zb_cluster_add_attr(esp_zb_ph_cluster, 0x0409U, 0x0000, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &phValue);
  esp_zb_cluster_add_attr(esp_zb_ph_cluster, 0x0409U, 0x0001, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &phMax);
  esp_zb_cluster_add_attr(esp_zb_ph_cluster, 0x0409U, 0x0002, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &phMin);

  esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_analog_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
```

I suppose that the esp_zb_cluster_list_add_custom_cluster must be used for another clusters ID range, but I was not able to find another function to add a esp_zb_attribute_list_t to the clusters list.

Maybe I am doing something wrong because I'm new using Zigbee. The easy way for me will be to create three "temperature" sensors clusters and use the client APP to change the name of every cluster, but I wanted to do it correctly. I have not searched the cluster ID for the Chlorine cluster, so maybe even doesn't exists.

Is there any way to create such clusters?

Thanks!

xieqinan, 2024-03-12:
@Danixu ,

The esp-zigbee-sdk temporarily does not provide support for the pH-measurement cluster. Even though `esp_zb_cluster_list_add_cluster()` exists, the stack will not manage or handle the pH-measurement cluster. In this scenario, the pH-measurement cluster behaves similarly to a custom cluster, with the exception of having a different cluster ID. Would you consider using the custom cluster to implement the pH-measurement functionalities? We have plan to support the pH-measurement cluster in the future.

Danixu, 2024-03-16:
Hello,

I have tried this code, and I am unable to use my Cluster ID because it says that is incorrect in both methods:

```
#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "esp_zigbee_core.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

/* Zigbee configuration */
#define INSTALLCODE_POLICY_ENABLE false /* enable the install code policy for security */
#define ED_AGING_TIMEOUT ESP_ZB_ED_AGING_TIMEOUT_64MIN
#define ED_KEEP_ALIVE 3000                                               /* 3000 millisecond */
#define ESP_ZB_PRIMARY_CHANNEL_MASK ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK /* Zigbee primary channel mask use in the example */

/* Default End Device config */
#define ESP_ZB_ZED_CONFIG() \
  { \
    .esp_zb_role = ESP_ZB_DEVICE_TYPE_ED, \
    .install_code_policy = INSTALLCODE_POLICY_ENABLE, \
    .nwk_cfg = { \
      .zed_cfg = { \
        .ed_timeout = ED_AGING_TIMEOUT, \
        .keep_alive = ED_KEEP_ALIVE, \
      }, \
    }, \
  }

#define ESP_ZB_DEFAULT_RADIO_CONFIG() \
  { \
    .radio_mode = RADIO_MODE_NATIVE, \
  }

#define ESP_ZB_DEFAULT_HOST_CONFIG() \
  { \
    .host_connection_mode = HOST_CONNECTION_MODE_NONE, \
  }

#define HA_ESP_SENSOR_ENDPOINT 1

/********************* Zigbee functions **************************/
static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask) {
  ESP_ERROR_CHECK(esp_zb_bdb_start_top_level_commissioning(mode_mask));
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct) {
  uint32_t *p_sg_p = signal_struct->p_app_signal;
  esp_err_t err_status = signal_struct->esp_err_status;
  esp_zb_app_signal_type_t sig_type = (esp_zb_app_signal_type_t)*p_sg_p;
  switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
      Serial.println("Zigbee stack initialized");
      esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
      break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
      if (err_status == ESP_OK) {
        Serial.println("Start network steering");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
      } else {
        /* commissioning failed */
        char strBuf[100];
        sprintf(strBuf, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
        Serial.println(strBuf);
      }
      break;
[... 138 lines cut ...]
  xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

void loop() {
}
```

The ESP32 just send me a warning messages saying that are invalid:

```
(299) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
(300) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
(304) ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!
(312) ESP_ZIGBEE_CLUSTER: Wrong 0x409 cluster to update to cluster list
(320) ESP_ZIGBEE_CLUSTER: Custom cluster id is not within specified range!
```

I have also tried this method:
https://github.com/espressif/esp-zigbee-sdk/issues/214#issuecomment-1895341262

Without sucess. Maybe I am doing something wrong, but I am not able to find out what.

I forgot to say that I am using the Arduino IDE for programming the ESP32C6 and the module version is the latest available with support for ESP32C6 (3.0.0-alpha3).

Best regards.

xieqinan, 2024-03-19:
Hi @Danixu 

> . Right now I don't have the code because it has suffered a lot of modifications trying to make it work, but the message is just that the cluster ID is not allowed. That is why I was asking if there is any way to do it.

The custom cluster ID is constrained by the ZCL specification, preventing the effect for the standard cluster ID. In the esp-zigbee-sdk, custom cluster IDs are required to be greater than 0x8000 to accommodate this limitation.

[image]

Danixu, 2024-03-20:
> Hi @Danixu
>
>
> The custom cluster ID is constrained by the ZCL specification, preventing the effect for the standard cluster ID. In the esp-zigbee-sdk, custom cluster IDs are required to be greater [...]

I see, custom means "non standard range". I though that maybe would be possible to implement non existant clusters types with it, including a standard one. So the solution for now is to create a non standard cluster and use the client application to set the real type, and wait until it is implemented into the library, right?.

Thanks!!

InnerBushman, 2025-04-09:
Is there somewhere a list of already implemented clusters? If not, how can I tell if a cluster is or isn't implemented yet?

xieqinan, 2025-04-09:
@InnerBushman ,

Please refer to the [document](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/introduction.html#zcl-clusters).

InnerBushman, 2025-04-09:
I must not understand the document correctly cause both the cluster ID the Op mentioned for pH_measurement 0x0409 and the one I'm trying to use, electrical_measurement 0x0b04 are on the list 1.2.1 of ZCL Clusters.
Unfortunately, even tho the code compiles, when I run it I get the error: ```E (270) ESP_ZIGBEE_CLUSTER: The requested cluster ID (0x b04) is not found```
Should I open a separate issue to discuss this? Or could it be that I simply am using it wrong?

xieqinan, 2025-04-09:
> Should I open a separate issue to discuss this? 

Yes, please separate to discuss it due to this issue has been closed.