espressif/esp-zigbee-sdk #279: esp_zb_scheduler_alarm with param that can take pointers

qcrist, 2024-03-10:
### Is your feature request related to a problem?

its difficult to use esp_zb_scheduler_alarm with param being a uint8_t

### Describe the solution you'd like.

uint8_t param => void* param

### Describe alternatives you've considered.

pointer lookup table indexed by param

nomis, 2024-04-02:
https://github.com/espressif/esp-zigbee-sdk/issues/188#issuecomment-1871135380 states that the scheduler queue has a limited size.

It would be useful to document what the limit is, what happens when it's exceeded, and how many entries are safe to use.

xieqinan, 2024-04-03:
@nomis ,

Thanks for your suggestion, we will consider adding document for scheduler mechanism.

chshu, 2024-04-28:
@qcrist The new [esp_zb_scheduler_user_alarm](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_core.h#L786) API is instroduced since esp-zigbee-lib 1.3.0, which can take a pointer param.

@nomis The default configuration for scheduler queue is 16 for ED and 32 for router:
https://github.com/espressif/esp-zboss-lib/blob/645d14096d35e2e688ea484f26467992bf85f455/include/zb_config.h#L834, it will return fail when the queue is full.

closed 2024-04-28 by chshu