espressif/esp-zigbee-sdk #280: Can't send uint64 with attribute reporting

makkssim, 2024-03-10:
I'm making my own project based on HA_on_off_light/switch, which will send a uint64 number in addition to the on/off command. To do this, I added an additional cluster with the corresponding attribute to the endpoint.
[image]
After that I created a function to update the value and report of the attribute. The general appearance of the function was borrowed from another project on Github, where it performed similar functions, except that the attribute type was uint16.
[image]
When the function is called, the following output appears.
[image]
Please tell me what I'm doing wrong

xieqinan, 2024-03-18:
Hello,

The esp-zigbee-sdk now includes an example demonstrating the usage of ZCL reporting, which is helpful to address this issue. You can refer to the [temperature and thermostat example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor).

makkssim, 2024-03-18:
Thanks, this example helped

closed 2024-03-18 by makkssim